pbatard/rufus #616: Help, My USB is not Detected

Thuruv, 2015-10-13:
Hello dev,
Thanks for the tool but unfortunately I Ran te following problem. Please enlighten us.

Found USB 2.0 device 'JetFlash Transcend 4GB USB Device' (8564:1000)
Could not open '\\?\usbstor#disk&ven_jetflash&prod_transcend_4gb&rev_1100#004bxsmdal7pgixe&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found

pbatard, 2015-10-13:
Could you please provide the __FULL__ log?
By truncating the log, you removed very important information, such as the Windows version and architecture you are using.

Also, if you are using Rufus 2.4, can you try issuing <kbd>Alt</kbd>-<kbd>.</kbd> in Rufus (that's the Alt and dot keys). You should see something like `USB Debug enabled` at the bottom, and you should get a bit more information in the log.

Now, access denied usually means that __another application__ is keeping exclusive access to the drive. It can be one of the applications [listed here](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with) (non exhaustive list). Or it could be that you are using a drive with additional security that prevents low level access. Or, if you are using a Windows 10 insider build, it could even be due to a Windows bug (#605)...

Thuruv, 2015-10-13:
Hope This one helps. .!

Rufus version: 2.4.757
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'JetFlash Transcend 4GB USB Device' (8564:1000)
Could not open '\\?\usbstor#disk&ven_jetflash&prod_transcend_4gb&rev_1100#004bxsmdal7pgixe&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub20#4&1a9c75f7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[182]: USB\VID_8087&PID_0024\5&2D9CFD7D&0&1
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub20#4&2d86aeae&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[016]: USB\VID_8087&PID_0024\5&5F446B8&0&1
Processing Hub 3:
  Hub[3] = '\\?\usb#vid_05e3&pid_0610#6&374bad7a&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[187]: USB\VID_152D&PID_2339\1ACBA2349FFF
Processing Hub 4:
  Hub[4] = '\\?\usb#vid_8087&pid_0024#5&2d9cfd7d&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[160]: USB\VID_05E3&PID_0610\6&374BAD7A&0&1
Processing Hub 5:
  Hub[5] = '\\?\usb#vid_8087&pid_0024#5&5f446b8&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[096]: USB\VID_046D&PID_C31C\6&29132496&0&1
  Found ID[024]: USB\VID_046D&PID_C05A\6&29132496&0&2
  Found ID[131]: USB\VID_8564&PID_1000\004BXSMDAL7PGIXE
Processing IDs belonging to USBSTOR:
  USB\VID_152D&PID_2339\1ACBA2349FFF
  USB\VID_8564&PID_1000\004BXSMDAL7PGIXE
Processing Device: 'USBSTOR\DiskJetFlashTranscend_4GB___1100'
  Matched with ID[131]: USB\VID_8564&PID_1000\004BXSMDAL7PGIXE
  Matched with Hub[5]: '\\?\usb#vid_8087&pid_0024#5&5f446b8&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'JetFlash Transcend 4GB USB Device' (8564:1000)
Could not open '\\?\usbstor#disk&ven_jetflash&prod_transcend_4gb&rev_1100#004bxsmdal7pgixe&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub20#4&1a9c75f7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[182]: USB\VID_8087&PID_0024\5&2D9CFD7D&0&1
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub20#4&2d86aeae&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[016]: USB\VID_8087&PID_0024\5&5F446B8&0&1
Processing Hub 3:
  Hub[3] = '\\?\usb#vid_05e3&pid_0610#6&374bad7a&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[187]: USB\VID_152D&PID_2339\1ACBA2349FFF
Processing Hub 4:
  Hub[4] = '\\?\usb#vid_8087&pid_0024#5&2d9cfd7d&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[160]: USB\VID_05E3&PID_0610\6&374BAD7A&0&1
Processing Hub 5:
  Hub[5] = '\\?\usb#vid_8087&pid_0024#5&5f446b8&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[096]: USB\VID_046D&PID_C31C\6&29132496&0&1
  Found ID[024]: USB\VID_046D&PID_C05A\6&29132496&0&2
  Found ID[131]: USB\VID_8564&PID_1000\004BXSMDAL7PGIXE
Processing IDs belonging to USBSTOR:
  USB\VID_152D&PID_2339\1ACBA2349FFF
  USB\VID_8564&PID_1000\004BXSMDAL7PGIXE
Processing Device: 'USBSTOR\DiskJetFlashTranscend_4GB___1100'
  Matched with ID[131]: USB\VID_8564&PID_1000\004BXSMDAL7PGIXE
  Matched with Hub[5]: '\\?\usb#vid_8087&pid_0024#5&5f446b8&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'JetFlash Transcend 4GB USB Device' (8564:1000)
Could not open '\\?\usbstor#disk&ven_jetflash&prod_transcend_4gb&rev_1100#004bxsmdal7pgixe&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.05/20140113 (from '/isolinux/isolinux.bin')
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
  Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 14.04.3 LTS amd64'
  Size: 1054867456 bytes
  Has a >64 chars filename: No
  Has Symlinks: Yes
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (4.05)
    With an old menu.c32: No
    With an old vesamenu.c32: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: ubuntu-14.04.3-desktop-amd64.iso

pbatard, 2015-10-13:
Okay, you're using Windows 7.

As I said, the message indicates that there is something on your system preventing access to the USB.
I'm afraid that, besides what I pointed earlier, I don't have more ideas as to what it could be, so you will have to find it yourself. Have you verified that you don't use any of the software listed as incompatible in the FAQ?

pbatard, 2015-10-24:
No update &rarr; Closing.

closed 2015-10-24 by pbatard