pbatard/rufus #617: [Rufus 2.4.757] Could not get exclusive access to device

JonathanMis, 2015-10-14:
Here's the log

Rufus version: 2.4.757
Windows version: Windows 10 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x040C
Found USB 2.0 device 'WD Elements 1078 USB Device' (1058:1078)
NOTE: This device is an USB 3.0 device operating at lower speed...
Could not get device number for device \\.\Z:: [0x00000001] Fonction incorrecte.
Could not get device number for device \\.\Z:: [0x00000001] Fonction incorrecte.
Device eliminated because it was detected as an USB Hard Drive (score 23 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0930:6545)
NOTE: This device is an USB 3.0 device operating at lower speed...
Could not get device number for device \\.\Z:: [0x00000001] Fonction incorrecte.
Could not get device number for device \\.\Z:: [0x00000001] Fonction incorrecte.
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1898, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x30D8DBFD
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15610675200 bytes)
  Start Sector: 8064, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'ESD-ISO'
  Size: 6241648640 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Windows.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Could not get exclusive access to device \\.\PHYSICALDRIVE3: [0x00000005] Accès refusé.

Can you help me ? 

Thanks,
Jonathan

JonathanMis, 2015-10-14:
Well, I just read #605 

My bad !

pbatard, 2015-10-14:
Yes, it's a duplicate of #605. I will close this issue then.

closed 2015-10-14 by pbatard