pbatard/rufus #618: Rufus stops suddenly and now my USB is damaged

slyfoxcl, 2015-10-15:
Hi there,
I have an issue when I format my USB with Rufus 2.4.
I am using Rufus from ver. 2.0 and works fine but when I updated to 2.4 two of my flash drives are damaged, Rufus 2.4 starts fine the format but when it takes 2% show a error and stops suddenly causing that Windows doesn't recognize the USB Drive, I tried it formating again by Windows tool and setting manually the device letter but Windows still does not recognize the device.
The last one I used is new and it seems weird that has been damaged.

I hope you can help me.

Syncerely,

Rodolfo Aravena Collipal

blackcrack, 2015-10-15:
make you computzer off, make the stream off and disconnect it completely from the stream,
wait 2 minutes and use it again, should work... Rufus/the Software should not (do not and can't) damage your hardware damage.

best
Blacky

pbatard, 2015-10-15:
1. If you see an error, __always__ provide the log from Rufus. It can provide valuable information as to what is happening. Without the log, it's very difficult to help you.
2. Your drives probably have a hardware issue. And, __NO__, [Rufus can not damage USB drives](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive). If your drives are failed, they would have failed regardless of the application you were using to copy data. And yes, it is possible for two drives to fail in short succession, especially if you are using cheap drives.

slyfoxcl, 2015-10-15:
Thanks for your fast reply!
Well yes, I know the software can't damage the hardware and that why it seems weird.
I will try to recreate the problem using another USB and get the log file.
Thanks again.

pbatard, 2015-10-24:
No update from OP &rarr; Closing.

closed 2015-10-24 by pbatard