pbatard/rufus #620: Update dialog doesn't display correctly in v2.4.757 for some languages.

ziaa, 2015-10-17:
Although in v2.3.709 update dialog is displayed correctly, in v2.4.757 and v2.5.768 (Beta) it doesn't display properly for me in some languages (Persian, Czech, Greek, Indonesian, Italian and …)

For example In Windows 7 SP1 and for Greek language it is displayed in this way for me:
[image]
and most of the dialog is invisible at the bottom of the screen.

In Windows 8.1 it is displayed better:
[image]

And here is the screen shot for v2.3 in Win7 which is displayed properly:
[image] 

For Persian language the situation get worse and the update dialog is displayed completely out of the screen (in both win7 & 8). Similar behavior happens when I use Arabic and move the main dialog to the right side of the screen and then open the update dialog. Here is the screen shot in Win 8.1:
[image]

Resolution on win7: 1366 * 768
Resolution on win8: 1280 * 768

pbatard, 2015-10-17:
Thanks for the report. I have a fairly good idea what's causing this behaviour (it's commit 585469b8d98fff6438e3d023499b0a121810167c which was introduced for 2.4), but short of reverting these changes altogether, which I don't want to do, it might be a bit tricky to fix. I'll see what I can do.

pbatard, 2015-10-18:
OK, I think I should have a fix for this with BETA 2, which is available [here](https://rufus.akeo.ie/downloads/). If that is not the case, please reopen this issue.

closed 2015-10-18 by pbatard

ziaa, 2015-10-19:
There is another issue for this dialog and it's only happen for Persian and Arabic translation. In these languages the dialog doesn't display **at the center** of the parent dialog (i.e. about dialog) and like the previous issue it's happen with the introduction of 2.4. The last screen shot is an example of the issue.

I'm not sure but If you calculate the center of the dialog the fact that, as you're probably well aware,  in RTL mode, unlike LTR, calculation is relative to top-right corner may help fix the problem.

pbatard, 2015-10-19:
Okay. I added that last issue into the list of RTL issues, which I'm planning to address in #621.

commit 916eaa41 (2015-10-18): [ui] fix an issue with automatic resizing of the update policy

* Closes #620