pbatard/rufus #626: Windows To Go option not showing.

Degru, 2015-10-24:
I'm trying to create a Windows To Go drive using a Windows 10 ISO obtained from the Media Creation Tool. I'm doing this from Windows 10. 

The Windows To Go option is simply not showing up in the latest version (2.4). I have tried previous versions as well. 2.0-2.2 have a Windows To Go option when I select the ISO, but they get an image extraction error when I try to actually create it. Anything newer than 2.2 (2.3, 2.4, 2.5 Beta) simply don't show the Windows To Go option.

It's kinda disappointing, because Rufus used to be the only tool that was able to create a working Windows To Go installation on my flash drive; all other methods that I tried would simply hang at boot.

pbatard, 2015-10-24:
1. When you find that Rufus doesn't work as you want, you should __ALWAYS__ check the log. If you do that, you might find some __very relevant__ information regarding your issue such as:
```
  Uses Install.wim: Yes (version 0.14.0)
  Note: This WIM version is NOT compatible with Windows To Go
```

2. You didn't actually try to create a WTG drive using that ISO in Rufus 2.2. Because if you did that, you would have found out that it simply __does not work__, and that is the reason it was removed.

3. _"using a Windows 10 ISO obtained from the Media Creation Tool"_
This is the problem. As you may not be aware, Microsoft managed to screw up the Windows 10 ISOs they produce using the Media Creation Tool, so that they are incompatible with their own Windows APIs for image extraction (and yes, they are even incompatible with those APIs on Windows 10). So I had to disable WTG for those images, since it wasn't working anyway!

4. If you want to create a WTG version of Windows 10, you should ditch the Media Creation Tool and instead download one of the __retail__ ISOs, which you can freely and legally obtain from [Microsoft's Tech Bench](https://www.microsoft.com/en-us/software-download/techbench). You should find that these work much better.

closed 2015-10-24 by pbatard

Degru, 2015-10-24:
Oh, I didn't realize that the media creation tool created broken ISO's. I'll try the TechBench ISO. Thanks.