pbatard/rufus #628: Screenshot v 2.4 b800 for Russian page and question about FreeBSD ISO (used BTX 1.0.2)

VictorVG, 2015-10-29:
I looked that there is an English screenshot, but other people need to see with their own eyes the image in the native language or do not believe. Made true to ISO images of FreeBSD 10.1 did not work - Rufus says that the ISO format with BTX 1.0.2 (a traditional boot BSD UNIX, and personally I like it more than a fashionable Grub) it does not support. But the idea would not be bad. How to proceed - we decide everything in this or create a separate record for FreeBSD ISO?

[image]

pbatard, 2015-10-30:
If I understand what you are saying, you indicate that the current screenshot for Russian on the Rufus homepage is tool old, and should be updated for v2.5.

I have now just done that:
[image]

Note however that:

1. I simply can __NOT__ update the localized screenshots for each release. There are more than 30 of them, and it just takes __way too much time__ to generate and upload each screenshot. So I will continue to only update them on very occasional basis.
2. If you are running at more than 150% magnification (high DPI display), and are using a modern browser, the screenshot you will get will __always__ be the English one. That's because, to make images look good, one needs to provide 1x and 2x versions of the same image, and there again, it would take way too much time to update all these. Even if I am not the one doing the actual translations, I don't believe people understand just how time consuming the localization process really is. To give you an idea, from one release to another, I need to spend about a week just dealing with localization. Therefore, anything that can reduce that process, such as not updating or multiplying localized screenshots, is welcome.
3. Also, I try not to use a screenshot from a build that is higher than the one from the latest release version (2.5.799 vs 2.5.800), as it will invariably leads to users asking _"I downloaded Rufus 2.5.799, but the screenshot on your site has 2.5.800! Where I can I download 2.5.800?"_...

With regards to FreeBSD:

1. __Please__ don't try to merge 2 completely separate issues into one. There's no restrictions to opening multiple issues, so keep topics that should really be separate, separate.
2. If you can point me to a guide that details exactly how to produce a working version of the FreeBSD installer, from Windows, that can be booted from USB __and that uses FAT32 as the file system__, then I might consider __revisiting__ the [attempts I've __already__ done trying to add FreeBSD support to Rufus](https://github.com/pbatard/rufus/issues/244#issuecomment-31155565). Otherwise, I'll stand by my previous findings that, unlike Grub or Syslinux, the FreeBSD loing process is simply not friendly enough to non BSD OSes to be usable there.  But I sure wouldn't mind being proved wrong... :smile: 
3. Keep in mind that that trying to extract ISO content to anything but a filesystem native to Windows, such as BSD's `ufs`/`ffs` will require the provision of a __custom__ BSD file system driver for Windows, which is something that would require A LOT of time to produce (months).

pbatard, 2015-10-30:
Oh, and I should point out that, if you really want to install FreeBSD using Rufus, you just have to download the `dd` images from http://ftp.freebsd.org/pub/FreeBSD/snapshots/VM-IMAGES/10.2-STABLE/, as they should be compatible with the image mode of Rufus...

VictorVG, 2015-10-30:
Images can be downloaded directly from the FTP project FreeBSD - ftp://anonymous:@ftp.freebsd.org:21/pub/FreeBSD/releases/ISO-IMAGES/ - in this catalog are final STABLE images collected at the time of release of a specific version, CURRENT (corresponds BETA) and RELEASE (corresponds to Release Candidate) under development just get in / snapshots which can be downloaded as a final assembly and adjusting. I usually do for themselves from /release and then take the necessary updating pekety (7.0 out of 600 small packages that have been updated in the initial installation of the ports by a script nearly 430 :)).

With regard to the registration of the two questions - just wanted to consult with you because they do not know how best to proceed, and the habit to put the OS with a strong piece of optics. On KDE have K3B and car have DVD-RAM drive, and often more than one. I burn a disc and put on the shelf. It is true that after the bad old drives occupy shelf, well, next to a dump and debris to carry long. :)

As for the download from BSD UNIX optics, there is traditionally not used Grub, a bunch of EI-Torito and BTX 1.0.2, and if something is under Win as I said, I personally prefer the optics, although there is an interesting project Frenzy - Live! CD/USB image is quite compact size, able to read / write NTFS, restore damaged operating system and troubleshoot hardware. I have version 1.0 of its written 512 MB flash drive and it lacks - the boot slice occupies 250 Mb FreeBSD 6.2.4 on UFS2, and 200 MB when you install it with FAT32 format to exchange data with Windows. Comfortable and most importantly with this memory stick works on any old cars since i586 with only 16 MB of RAM! I'm with her once on the old laptops Mitsubishi repaired and stick to them FreeBSD to put - these machines have a unique shock-resistant construction from the HDD is in a special gel amortization chamber and durable magnesium alloy body (can not imagine how it can be broken - well, except that under the heavy tank to shove, but I stand :)).

pbatard, 2015-10-30:
OK.

As I pointed out, if you download the `memstick.img` version instead of the `.iso`, you should be able to run/install FreeBSD using Rufus (even the compressed `memstick.img.xz` should work), so I don't think there's anything for me to do for FreeBSD support, as BTX is NO_GO.

And I fixed the screenshot, so I will close this issue now.

closed 2015-10-30 by pbatard

VictorVG, 2015-10-30:
Ok! 

I personally think that there is no problem, I just need to consult with you and thank you for your help. I'm going to deal with the fact where my LAN fails (I suspect that refused router - not just because there are failures of demons, including the X's fail under the message "Video Driver nvlddmkm stopped responding and has successfully recovered.", Although it is possible that the consequences of errors clink v0 .44 - v 0.45 which had to be removed due to the fact that it did not disclose environment variables leading to errors throughout the console OS subsystem. These errors are not yet resolved, at least a new build of the program is formally no more, and where there is no bash me such errors more than happy. :))