pbatard/rufus #629: Not recognizing my usb thumb drive

lesonwon, 2015-10-29:
Here is the full log that was requested, my apologies on the first post being incomplete

Rufus version: 2.5.799
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
Could not open '\\?\usbstor#disk&ven_sandisk&prod_cruzer_glide&rev_1.27#20044323000f11208b2c&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found

pbatard, 2015-10-29:
Okay. This doesn't have the USB Debug information, but as I said earlier `[0x00000005] Access is denied` usually means that __another__ application (could be a security solution, something else) has exclusive access to the device. So you may have to figure which one. For instance, are you using any of the applications listed [at the very end of the Rufus FAQ](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with)? (NB: This is list is NOT exhaustive). There's really nothing I can do in Rufus if __another__ application has taken exclusive access to the USB.

lesonwon, 2015-10-29:
Ok, thank you for the quick reply, i am not sure how to access the USB debug info but i do not have any of the listed programs. I will check if another program has exclusive access to the drive

closed 2015-10-29 by lesonwon