pbatard/rufus #630: ERROR: Could not set path: [14] Not Found

artailicous, 2015-11-01:
[image]

I use Ubuntu, and followed this guide: http://onetransistor.blogspot.se/2015/09/uefi-ntfs-bootable-windows-usb-linux.html

Any ideas what can be the issue here? Some files are missing in the NTFS portion I guess, but what files?

pbatard, 2015-11-01:
This most likely indicates that you are trying to boot a __64 bit__ EFI system (e.g. Windows x64) on a __32 bit__ UEFI firmware, which is not possible. In that case, because it was booted in 32bit mode, UEFI:NTFS will be looking for a `/efi/boot/bootia32.efi' file on the USB, which it will not be able to find, hence the error.

Please note that the "bitness" of the UEFI firmware is independent of the "bitness" of the CPU, so you may very well have a system with a 64bit compatible CPU but a 32bit UEFI firmware, which forces you to install a 32bit version of Windows. That's because, in order to save a few bucks, it is common for cheap systems to have a 32 bit UEFI firmware. For more on this, please see [this](http://blogs.intel.com/evangelists/2015/07/22/why-cheap-systems-run-32-bit-uefi-on-x64-systems).

If you are indeed trying to install a 64 bit version of Windows, please try to switch to a 32bit version, and see if that works better.

pbatard, 2015-11-04:
I believe this issue has been answered so I will close it.

closed 2015-11-04 by pbatard