pbatard/rufus #631: Windows To Go problems

MarkosMark, 2015-11-01:
I'm trying to make a bootable Windows To Go drive (GPT), but I'm facing the following problems:
- I can only choose **Windows To Go** when I use Rufus 2.2 (2.2 and 2.2p) I also tried newer versions. Normal and portable, but still only 2.2 gives me the option to make Windows To Go. I also tried different ISO's of Windows 8.1 and 10. 32bit, 64bit, Pro, English, Dutch. etc. All are official ISO's from Microsoft.
- When I try to make this Windows To Go with Rufus 2.2, I'm getting the error **Error: ISO image extraction failure**. I found this could be cause by an antivirus program, but even when I disable it I still get this error. See logs in attachment

I need this because I have a *Surface Pro 3 (i7 8GB)* with a defective SSD. I can run Ubuntu from a MicroSD card but I'm not able to install Windows to the MicroSD or USB flash drive. So I'm hoping I can boot *Windows To Go* from SD or USB flash drive.

pbatard, 2015-11-01:
Yes, the Windows To Go option, __when using the ISO images created by Microsoft's Media Creation Tool__ was removed starting with Rufus 2.3, because Microsoft broke API compatibility with the `install.wim` image from these ISOs, and, as you found out, it didn't work.

When you try to use any such image in Rufus 2.3 or later, you will see the following in the log:
```
  Uses Install.wim: Yes (version 0.14.0)
  Note: This WIM version is NOT compatible with Windows To Go
```

I am however surprised by your report that official Windows 8.1 & 10 didn't work. They very much should. Are you sure you used the official retail ISOs from the MSDN (or other official sources). For instance, have you tried with the official Windows 10 retail ISOs that are freely available from Microsoft's Tech Bench](https://www.microsoft.com/en-us/software-download/techbench)?

Oh, and for the record, this is a duplicate of #626.

MarkosMark, 2015-11-02:
Okay thank you! I indeed downloaded Windows 8 and 10 with the media creation tool. That's the only way I knew how to download the official Windows releases.
What is the difference with this Tech Bench? And can I download Windows 8(.1) and/or 7 there as well?
I was able to make a Windows To Go with this Tech Bech version. I will check it out this evening if it works on my SP3 :)

Sorry I didn't see topic #626 I came here through #155

pbatard, 2015-11-02:
> What is the difference with this Tech Bench?

They are the same as the retail ISOs produced by Microsoft. The Media Creation tool aren't...

> And can I download Windows 8(.1) and/or 7 there as well?

Nope. For official retail ISOs, you have to figure out other sources (I don't know of any that's provided by Microsoft). You can however (and really __should__) validate that the retail ISOs you have are the official ones by comparing their SHA-1 with the ones you can get from the MSDN. For instance, if you are interested in validating a specific Win 8.1 retail ISO, you should locate it in [MSDN](https://msdn.microsoft.com/en-us/subscriptions/downloads/#searchTerm=&ProductFamilyId=545&Languages=en&Architectures=x64&FileExtensions=.iso&PageSize=10&PageIndex=0&FileId=0), then click on 'Details' to get the SHA-1. Then in Rufus, you can press the `#` symbol at the bottom to compare that SHA-1 with the one from the ISO you selected. If they match, then you have an official retail ISO.

pbatard, 2015-11-04:
I believe this issue has been answered, so I will close it.

closed 2015-11-04 by pbatard