pbatard/rufus #632: Undetermined error while formatting

Fr4nc3sc0NL, 2015-11-04:
I got this "Undetermined error while formatting" message twice. So I tried the standard windows format of the disk management tool and that finished without problems. After that rufus just finished formatting and started to copy my iso to the pen drive. I thought this was very odd since it didn't work at first. I can send you the log if you can tell me the best way to do that . (I can't attach .txts because I don't have write permission)

pbatard, 2015-11-04:
Please copy/paste the __FULL__ log in this issue (no need to attach files).

Fr4nc3sc0NL, 2015-11-05:
this is the full log, as you can see it includes 4 tries. First two fail because of the undetermined error the third is canceled by me and the fourth passes. I formatted it with disk management between the sencond and third trie

Rufus version: 2.5.799
Windows version: Windows 8.1 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Could not get node connection information (V2) for device 'USB\VID_0011&PID_7788\0AF30DA3': [0x00000001] Incorrect function.
Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\F:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\F:' as it spans multiple disks (RAID?)
1 device found
No volume information for drive 0x83
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1975, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x195031E4
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 15.1 GB (16252926976 bytes)
  Start Sector: 2, Boot: Yes, Recognized: Yes
Checking for Rufus updates...
Checking release channel...
No new release version found.
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'Win10_10240_x64_Sep2015'
  Size: 3513176064 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Win10_10240_x64_Sep2015_6in1_en-US_murphy78.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\F:' as it spans multiple disks (RAID?)
Will use 'M:' as volume mountpoint
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume does not have an x86 Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'M:' after error

[... 284 lines cut ...]
Extracting: M:\sources\boot.wim (276.8 MB)
Extracting: M:\sources\install.esd (3.0 GB)
Extracting: M:\sources\lang.ini (78 bytes)
Extracting: M:\sources\setup.exe (258.7 KB)
Finalizing, please wait...
Created: M:autorun.inf
Created: M:autorun.ico

Could not get node connection information (V2) for device 'USB\VID_0011&PID_7788\0AF30DA3': [0x00000001] Incorrect function.
Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\F:' as it spans multiple disks (RAID?)
Using autorun.inf label for drive M: 'Win10_10240_x64_Sep2015'
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\F:' as it spans multiple disks (RAID?)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1975, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00401E8D
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 15.1 GB (16251879424 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2015-11-05:
Thanks.

The one interesting thing I see is that your disk's partition was original set to start on sector 2:
```
Start Sector: 2
```

There's been a longstanding requirement (dating back from DOS) that the first partition should at least start at a cylinder/track limit (63 in your case), and most recent partitioning utilities, like `fdisk`, `diskpart` or Rufus will add a little bit more to that, to make the first partition start at 2048. So whatever utility was used to partition your drive before you started using it with Rufus, it seems to be doing something that it's not supposed to do if the first partition starts at sector 2.
Do you know what utility was used to partition your disk before you used Rufus?

Now, ideally, repartitioning the drive should do the trick, but there may be more needed to hadle a drive that started at sector 2, even if you try to do everything by the book. Rufus very much tells the system to refresh its partition table after it recreates the partition, which is exactly what Microsoft says you should do. And Rufus is really using the exact same API call Windows is using when doing the actual formatting (same DLL, and so on). So if that doesn't doesn't, then I suspect that Microsoft is doing something extra in their own applications, which they didn't document.

I'm trying to see if I can create a drive with a partition that starts at sector 2 (which is very tricky in itself - none of the regular partitioning utilities will let you do that!), to see if I can reproduce your issue, but even if I can, without insight from Microsoft, it may be very difficult to fix, and I don't expect many people to be affected by this, since you first need to have partitioned a drive with a partitioning utility that does something it's not supposed to do.

pbatard, 2015-11-05:
OK, I finally managed to create a 16GB flash drive that reports the same parameters as the ones Rufus indicated on your first try... but I saw no issue when creating a Windows 10 bootable drive from it. Here's my output:
```
Rufus version: 2.5.799
Windows version: Windows 8.1 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0809
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive E: 'J_CCSNA_X64FRE_EN-US_DV5'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x002FB09E
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 14.9 GB (16013941760 bytes)
  Start Sector: 2, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'J_CCSA_X64FRE_EN-US_DV5'
  Size: 4083853312 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
  Uses Install.wim: Yes (version 0.13.1)
Using image: en_windows_10_multiple_editions_x64_dvd_6846432.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume does not have an x86 Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume GUID \\?\Volume{d25bbe87-287c-11e5-82ba-acfdce2112d0}\
Opened drive \\?\Volume{d25bbe87-287c-11e5-82ba-acfdce2112d0} for write access
Writing partition boot record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted Volume{d25bbe87-287c-11e5-82ba-acfdce2112d0}\ on E:\
Copying ISO files...
Extracting files...
Image is an UDF image
Extracting: E:\autorun.inf (128 bytes)
Extracting: E:\boot\bcd (16 KB)
Extracting: E:\boot\boot.sdi (3 MB)
Extracting: E:\boot\bootfix.bin (1 KB)
(...)
Extracting: E:\support\logging\sysprepetw.dll (22.7 KB)
Extracting: E:\support\logging\windeployetw.dll (19.7 KB)
Extracting: E:\support\logging\winsetupetw.dll (22.7 KB)
Finalizing, please wait...
E:autorun.inf already exists - keeping it
NTFS Fixup (Checkdisk)...
Volume label is J_CCSA_X64FRE_EN-US_DV5.

Stage 1: Examining basic file system structure ...
  1280 file records processed.
File verification completed.
  0 large file records processed.
  0 bad file records processed.

Stage 2: Examining file name linkage ...
  1500 index entries processed.
Index verification completed.

Stage 3: Examining security descriptors ...
Security descriptor verification completed.
  110 data files processed.

Windows has scanned the file system and found no problems.
No further action is required.

  15637591 KB total disk space.
   4015548 KB in 964 files.
       304 KB in 112 indexes.
     25311 KB in use by the system.
     23136 KB occupied by the log file.
  11596428 KB available on disk.
      4096 bytes in each allocation unit.
   3909397 total allocation units on disk.
   2899107 allocation units available on disk.
NTFS Fixup completed.

Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0001B65A
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.9 GB (16012894208 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
```

This is about as close as I can get to your settings (same OS, same Rufus version, same drive configuration). The only differences I can see:
* Not the exact same flash drive
* Using USB 3.0 instead of USB 2.0. However, I see no possible scenario where this can matter for formatting: The USB speed is completely transparent for format operations.
* The fact that your system was not able to disable I/O boundary checks (`I/O boundary checks disabled` message is absent on your system). But this should only matter for partitioning however, and about all of the systems I know don't really need this for Rufus to work.

None of the differences above should matter, so at this stage, the only relevant data that I see that could explain the issue is:
* The fact that your FAT32 partition was set as type `0x0B` (__non LBA__ FAT32, ) whereas it should probably have been `0x0C` (LBA FAT32).
* The fact that you had a non standard Master Boot Record (MBR).
* The fact that the partition has a non standard Boot Record as well.

In fact, I really had to go out of my way to manually modify my flash drive to replicate these settings, and, combined with your partition starting at an odd location, everything seems to point to the tool that was used to partition and format the drive before you used Rufus.

Can you please provide any indication as to what was used to partition and format your drive prior to using Rufus? I'm pretty sure this really is the key to replicating this issue, if it is replicable, and addressing it.

pbatard, 2015-11-10:
Any update on this? If not, I will have to close this issue.

Fr4nc3sc0NL, 2015-11-11:
I'm sorry for the late response, it's been a very busy week for me. 

I've been messing around with this flash drive a lot. Formatted it on OS X (with the built in disk manager "disk utility") and multiple tools on windows. The tool I used before Rufus was one of the two tools provided by microsoft itself to make a bootable windows USB drive, but unfortunately I don't remember which of the two. (if there's more than two microsoft tools: I'm referring to the "media creation tool" and the "Windows USB/DVD Download Tool" http://windows.microsoft.com/en-US/windows-8/create-reset-refresh-media and https://www.microsoft.com/en-us/download/windows-usb-dvd-download-tool). Obviously this tool failed to do it's job otherwise I wouldn't have used Rufus. 

On a side note: I do notice that OS X is creating an extra partition when formatting sometimes, but I didn't figure out yet when it does and when it doesn't.. My 256 external SSD for instance got a protected 200MB EFI partition when I formatted it with my mac, while I never asked for it..

pbatard, 2015-11-15:
Okay. I think the OS X tool is probably the one that created the weird partitioning (AFAIK, the MS tool will simply reformat but not repartition the drive) so I'll see if I can play with USB creation on Mac.

Don't expect anything fast though...

mike02x, 2015-12-22:
Hi, I have the same issue ,Im trying to setup a USB for Windows 10, Ive tried all the ways even right click and "Run as admin" but still it fails to format it the USB I know it's good because I was able to extract and copy the ISO into it but I'm not able to boot with it and that's why a friend recommended rufus to me.

pbatard, 2015-12-22:
> I have the same issue

Can you please copy/paste your __full__ log? Without the log, it's hard to tell if it's really the same issue. Also, did you previously format your drive using OS-X (which seems to the the cause of the issue for OP).

One thing you may try is download the latest BETA [here](https://rufus.akeo.ie/downloads/) and then use <kbd>Alt</kbd>-<kbd>Z</kbd> to zero the device. Then you can interrupt the zeroing (though it's probably better to let it through - it may take some time to complete however) and try formatting your drive again.

mike02x, 2015-12-22:
Hi, 

I'm not using OSX though, I'm using Windows 10.

I'm at work at the moment, but ill definitely paste my full log file here once I get home.

Thanks!

mike02x, 2015-12-23:
Here's my logs:
Rufus version: 2.5.799
Windows version: Windows 10 32 bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x080A
Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
1 device found
Could not open GUID volume '\\?\Volume{eb170ef4-b79a-11e4-afab-d05099037e42}': [0x00000005] Acceso denegado.
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1021, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT16 (0x06)
  Size: 7.8 GB (8398045184 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'ESD-ISO'
  Size: 2592669696 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
  Uses Install.wim: Yes (version 0.14.0)
  Note: This WIM version is NOT compatible with Windows To Go
Using image: Windows.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Could not open GUID volume '\\?\Volume{eb170ef4-b79a-11e4-afab-d05099037e42}': [0x00000005] Acceso denegado.
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found!
Formatting (NTFS)...
Could not read volume name
Format error: Undetermined error while formatting.
Could not open GUID volume '\\?\Volume{eb170ef4-b79a-11e4-afab-d05099037e42}': [0x00000005] Acceso denegado.

Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
1 device found
Could not open GUID volume '\\?\Volume{eb170ef4-b79a-11e4-afab-d05099037e42}': [0x00000005] Acceso denegado.
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1021, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0011CBF8
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 7.8 GB (8398045184 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Could not open GUID volume '\\?\Volume{eb170ef4-b79a-11e4-afab-d05099037e42}': [0x00000005] Acceso denegado.
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Deleting partitions...
Bad Blocks: Checking from block 0 to 16404479
Bad Blocks: Using offset 0 for fake device check
Bad Blocks: Writing test pattern 0xAA
Cancelling
Bad Blocks: Interrupting at block 116096
Bad blocks: Check failed.
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Could not open GUID volume '\\?\Volume{eb170ef4-b79a-11e4-afab-d05099037e42}': [0x00000005] Acceso denegado.

Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
1 device found
Could not open GUID volume '\\?\Volume{eb170ef4-b79a-11e4-afab-d05099037e42}': [0x00000005] Acceso denegado.
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1021, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 7.8 GB (8399093760 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Could not open GUID volume '\\?\Volume{eb170ef4-b79a-11e4-afab-d05099037e42}': [0x00000005] Acceso denegado.
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found!
Formatting (NTFS)...
Could not read volume name
Format error: Undetermined error while formatting.
Could not open GUID volume '\\?\Volume{eb170ef4-b79a-11e4-afab-d05099037e42}': [0x00000005] Acceso denegado.

Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
1 device found
Could not open GUID volume '\\?\Volume{eb170ef4-b79a-11e4-afab-d05099037e42}': [0x00000005] Acceso denegado.
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1021, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001346A1
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 7.8 GB (8398045184 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2015-12-23:
Thanks. Because you are getting `Access Denied` it looks like something __external to Rufus__ is preventing proper access to the drive, so you will have to find out what it is (and short of [pointing you to the __non exhaustive__ list of software that I know can interfere with Rufus](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with) there's really not much I cqn do to help).

You can try <kbd>Alt</kbd>-<kbd>Z</kbd> or <kbd>Alt</kbd>-<kbd>,</kbd> and see if that helps, but if it doesn't, it'll be up to you to figure out what application is preventing access.

acer475, 2015-12-29:
Hi Pbatard

Can you help me check what happen to my USB drive, i have the same problem and here is my log:

    Rufus version: 2.6.818
    Windows version: Windows 7 SP1 64-bit
    Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
    Grub versions: 0.4.6a, 2.02~beta2
    Locale ID: 0x0409
    Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
    1 device found
    No volume information for drive 0x81
    Disk type: Removable, Sector Size: 512 bytes
    Cylinders: 1019, TracksPerCylinder: 255, SectorsPerTrack: 63
    Partition type: MBR, NB Partitions: 1
    Disk ID: 0x0046C208
    Drive has a Zeroed Master Boot Record
    Partition 1:
      Type: NTFS/exFAT/UDF (0x07)
      Size: 7.8 GB (8387559424 bytes)
      Start Sector: 2048, Boot: Yes, Recognized: Yes
    Scanning image...
    ISO analysis:
      Image is an UDF image
    Disk image analysis:
      Image does not have an x86 Master Boot Record
    ISO label: 'J_CCSA_X64FRE_EN-US_DV5'
      Size: 4017000448 bytes
      Has a >64 chars filename: No
      Has Symlinks: No
      Has a >4GB file: No
      Uses Bootmgr: Yes
      Uses EFI: Yes
      Uses Grub 2: No
      Uses Grub4DOS: No
      Uses isolinux: No
      Uses KolibriOS: No
      Uses ReactOS: No
      Uses WinPE: No
      Uses Install.wim: Yes (version 0.13.1)
    Using image: en_windows_10_multiple_editions_version_1511_x64_dvd_7223712_2.iso

    Format operation started
    Requesting disk access...
    Opened drive \\.\PHYSICALDRIVE1 for write access
    Will use 'F:' as volume mountpoint
    Analyzing existing boot records...
    Drive has a Zeroed Master Boot Record
    Volume does not have an x86 Partition Boot Record
    Deleting partitions...
    Clearing MBR/PBR/GPT structures...
    Erasing 128 sectors
    Partitioning (MBR)...
    Closing existing volume...
    Waiting for logical drive to reappear...
    Formatting (NTFS)...
    Using cluster size: 4096 bytes
    Quick format was selected
    Creating file system...
    Error while formatting
    Format error: Undetermined error while formatting.
    Re-mounted volume as 'F:' after error

    Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
    1 device found
    No volume information for drive 0x81
    Disk type: Removable, Sector Size: 512 bytes
    Cylinders: 1019, TracksPerCylinder: 255, SectorsPerTrack: 63
    Partition type: MBR, NB Partitions: 1
    Disk ID: 0x004E4FE5
    Drive has a Zeroed Master Boot Record
    Partition 1:
      Type: NTFS/exFAT/UDF (0x07)
      Size: 7.8 GB (8387559424 bytes)
      Start Sector: 2048, Boot: Yes, Recognized: Yes
    Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
    1 device found
    No volume information for drive 0x81
    Disk type: Removable, Sector Size: 512 bytes
    Cylinders: 1019, TracksPerCylinder: 255, SectorsPerTrack: 63
    Partition type: MBR, NB Partitions: 1
    Disk ID: 0x004E4FE5
    Drive has a Zeroed Master Boot Record
    Partition 1:
      Type: NTFS/exFAT/UDF (0x07)
      Size: 7.8 GB (8387559424 bytes)
      Start Sector: 2048, Boot: Yes, Recognized: Yes
    Closed Windows format prompt

Thanks

acer475, 2016-01-01:
Hi there, 

It's work now after I change to use another USB, mean my previous USB have problem.

murdockbaracus, 2016-03-23:
So I was having the same issue and it was as simple as switching from my USB hub to a direct connection to the PC. After I switched it went straight through the process.  I hope this helps.

AshMcConnell, 2016-05-17:
Thanks murdock, that worked for me!

iman110, 2016-05-27:
Hi pbatard. Please help me. I have the same error and This is my full log

iman110, 2016-05-27:
Rufus version: 1.4.9.506
Syslinux versions: 4.07, 5.10
Windows version: Windows 7 SP1 32 bit
Locale ID: 0x0409
Found USB 1.0 device 'Generic UDISK USB Device' (CD12:EF18)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive has an unknown Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 7.5 GB (8074035200 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
Scanning image...
Disc image is an UDF image
ISO label: 'Win10 Enterprise'
  Size: 4584173568 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses ReactOS: No
  Uses KolibriOS: No
  Uses EFI: Yes
  Uses Bootmgr: Yes
  Uses WinPE: No
  Uses isolinux: No
Using ISO: ISO Image.iso

Format operation started
Requesting disk access...
Caution: Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Closing existing volume...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system: Task 1/12 completed
Creating file system: Task 2/12 completed
Creating file system: Task 3/12 completed
Creating file system: Task 4/12 completed
Creating file system: Task 5/12 completed
Creating file system: Task 6/12 completed
Creating file system: Task 7/12 completed
Creating file system: Task 8/12 completed
Creating file system: Task 9/12 completed
Creating file system: Task 10/12 completed
Creating file system: Task 11/12 completed
Creating file system: Task 12/12 completed
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume GUID \\?\Volume{e39c4ed2-8f22-11e5-87bc-806e6f6e6963}\
\\?\Volume{e39c4ed2-8f22-11e5-87bc-806e6f6e6963}\ was already mounted as G:\
Caution: Opened drive \\?\Volume{e39c4ed2-8f22-11e5-87bc-806e6f6e6963} for write access
Writing partition boot record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
[... 3305 lines cut ...]
File record segment 681 is an orphan.
File record segment 682 is an orphan.
File record segment 683 is an orphan.
File record segment 686 is an orphan.
File record segment 687 is an orphan.
File record segment 689 is an orphan.
File record segment 691 is an orphan.
File record segment 692 is an orphan.
File record segment 694 is an orphan.
File record segment 697 is an orphan.
File record segment 702 is an orphan.
File record segment 706 is an orphan.
File record segment 707 is an orphan.
File record segment 709 is an orphan.
File record segment 710 is an orphan.
File record segment 714 is an orphan.
File record segment 715 is an orphan.
File record segment 716 is an orphan.
File record segment 718 is an orphan.
File record segment 719 is an orphan.
File record segment 722 is an orphan.
File record segment 723 is an orphan.
File record segment 724 is an orphan.
File record segment 725 is an orphan.
File record segment 727 is an

pbatard, 2016-05-27:
> I have the same error

__NO__, you don't!
Do you see `Format error: Undetermined error while formatting` anywhere in your log???

It's very bad practice to hijack an existing issue for something that is completely different, because it wastes the time of people who look for a solution for the original issue, and it also wastes the one for those who have the same issue as you, as they will have more trouble finding relevant entries.

Besides, your log makes it pretty explicit that the issue is 100% with your USB drive, which is failing and needs to be replaced, so there's really nothing I can do here.

Since I have answered your question, and your query was irrelevant to this issue, I will delete both your message and this one in 24 hours.

iman110, 2016-05-27:
First I thank you for your reply but I was surprised by your quick
judgement!! I got that message as separate error box. I didn't know that
message must be in the log. I was using Rufus to copy Windows on my USB
drive and in the end of copying the Windows I got that message in a single
box. I repeated several times but in each time I got that message. I don't
like to waste the time of people at all. If I didn't got that message I
wouldn't write my problem here. I know that the time of people is valuable.
You mean My USB drive is crashed? I can copy and paste different files to
and from it.

pbatard, 2016-05-27:
Your USB is defective. The flash memory on your drive has probably failed as identified by the following Checkdisk messages:
```
The first NTFS boot sector is unreadable or corrupt.
(...)
Critical master file table (MFT) files are corrupt.
(...)
Attribute record (128, "") from file record segment 1 is corrupt.
```
Please be aware that flash memory has a very limited number of read/write cycles before it fails (which is usually much shorted than people think, especially for cheap drives).

Your USB flash drive is almost dead and this is not something that can be fixed. Please buy a new one.

pbatard, 2016-08-31:
I'm going to close this issue.

closed 2016-08-31 by pbatard

Akimu89, 2017-12-17:
Hi All,
   Rufus keeps preventing me to burn any windows on my flash drive thought ive changed it with a 32gb, still the same issue.
Any help is appreciated 
PS I'm using Windows 10 x64bit 
Here is the log...

Rufus version: 2.18.1213
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Found USB 2.0 device 'Generic Flash Disk USB Device' (0000:7777)
1 device found
Disk type: Removable, Disk size: 8GB, Sector size: 512 bytes
Cylinders: 1019, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0027FBB7
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.8 GB (8387559424 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'DVD'
  Size: 3.7 GB (Projected)
  Uses: EFI
  Uses: Bootmgr
Using image: win7.iso (3.6 GB)

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'H:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'Generic Flash Disk USB Device' (0000:7777)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 8GB, Sector size: 512 bytes
Cylinders: 1019, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001D862A
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 7.8 GB (8387559424 bytes)
  Start Sector: 2048, Boot: Yes
Closed Windows format prompt

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'H:' as volume mountpoint
Requesting lock...
Zeroing drive...

huolu, 2018-04-15:
Yes it seems related with USB ports and loading.
My Dell M2800 laptop has 3 USB ports. The usb disk is on port B while the mouse is on port C.
I tried a couple of times to burn a FreeDOS on this usb disk but consistently got "Undetermined error while formatting". Meanwhile I tried another u-disk on the same USB port B and it passed.
Inspired by murdockbaracus's finding, I then tried these:
1. Tried the first u-disk on port A, got "Read error".
2. Unplug the mouse from port C and tried the first u-disk on port C, got "Read error".
3. Tried the first u-disk on port B again but w/o mouse plugged in, this time it passed.

I've attached log 1/2/3 below and hopefully it's helpful. Previous logs were flushed out.

huolu, 2018-04-15:
Warning: Conflicting process search failed to complete due to timeout

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
No drive letter was assigned...
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 63 sectors
Deleting partitions...
Partitioning (MBR)...
Adding extra partition
Reserved 1 tracks (31.5 KB) for extra partition
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 1024 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
read_sectors: Read error [0x00000002] The system cannot find the file specified.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have an x86 Master Boot Record
read_sectors: Read error [0x00000002] The system cannot find the file specified.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not read MBR
Could not get Disk Extents: [0x00000002] The system cannot find the file specified.

0 devices found
0 devices found
Found USB 2.0 device 'SMI USB DISK USB Device' (090C:1000)
1 device found
Disk type: Removable, Disk size: 131MB, Sector size: 512 bytes
Cylinders: 15, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00CC1155
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 124.9 MB (130991616 bytes)
  Start Sector: 63, Boot: Yes
Partition 2:
  Type: Rufus Extra (0xea)
  Size: 31.5 KB (32256 bytes)
  Start Sector: 255906, Boot: No

huolu, 2018-04-15:
Warning: Conflicting process search failed to complete due to timeout

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
No drive letter was assigned...
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 63 sectors
Deleting partitions...
Partitioning (MBR)...
Adding extra partition
Reserved 1 tracks (31.5 KB) for extra partition
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 1024 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
read_sectors: Read error [0x00000002] The system cannot find the file specified.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have an x86 Master Boot Record
read_sectors: Read error [0x00000037] The specified network resource or device is no longer available.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not read MBR
Could not get Disk Extents: [0x00000002] The system cannot find the file specified.

0 devices found
0 devices found
Found USB 2.0 device 'SMI USB DISK USB Device' (090C:1000)
1 device found
Disk type: Removable, Disk size: 131MB, Sector size: 512 bytes
Cylinders: 15, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00CCC9E7
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 124.9 MB (130991616 bytes)
  Start Sector: 63, Boot: Yes
Partition 2:
  Type: Rufus Extra (0xea)
  Size: 31.5 KB (32256 bytes)
  Start Sector: 255906, Boot: No

huolu, 2018-04-15:
Warning: Conflicting process search failed to complete due to timeout

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
No drive letter was assigned...
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 63 sectors
Deleting partitions...
Partitioning (MBR)...
Adding extra partition
Reserved 1 tracks (31.5 KB) for extra partition
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 1024 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume GUID \\?\Volume{916c6b0b-a1c7-11e5-838f-acfdce08dba7}\
Waiting for access on \\?\Volume{916c6b0b-a1c7-11e5-838f-acfdce08dba7} [\Device\HarddiskVolume36]...
Opened \\?\Volume{916c6b0b-a1c7-11e5-838f-acfdce08dba7} for exclusive write access
Writing partition boot record...
Using FreeDOS FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Successfully remounted \\?\Volume{916c6b0b-a1c7-11e5-838f-acfdce08dba7}\ on G:
Copying DOS files...
Successfully wrote 'G:\COMMAND.COM' (66945 bytes)
Successfully wrote 'G:\KERNEL.SYS' (46685 bytes)
Successfully wrote 'G:\LOCALE\DISPLAY.EXE' (3651 bytes)
Successfully wrote 'G:\LOCALE\KEYB.EXE' (11446 bytes)
Successfully wrote 'G:\LOCALE\MODE.COM' (16244 bytes)
Successfully wrote 'G:\LOCALE\KEYBOARD.SYS' (40360 bytes)
Successfully wrote 'G:\LOCALE\KEYBRD2.SYS' (29750 bytes)
Successfully wrote 'G:\LOCALE\KEYBRD3.SYS' (32178 bytes)
Successfully wrote 'G:\LOCALE\KEYBRD4.SYS' (13105 bytes)
Successfully wrote 'G:\LOCALE\EGA.CPX' (6464 bytes)
Successfully wrote 'G:\LOCALE\EGA2.CPX' (7059 bytes)
Successfully wrote 'G:\LOCALE\EGA3.CPX' (5469 bytes)
Successfully wrote 'G:\LOCALE\EGA4.CPX' (4431 bytes)
Successfully wrote 'G:\LOCALE\EGA5.CPX' (7217 bytes)
Successfully wrote 'G:\LOCALE\EGA6.CPX' (7409 bytes)
Successfully wrote 'G:\LOCALE\EGA7.CPX' (5387 bytes)
Successfully wrote 'G:\LOCALE\EGA8.CPX' (6973 bytes)
Successfully wrote 'G:\LOCALE\EGA9.CPX' (5785 bytes)
Successfully wrote 'G:\LOCALE\EGA10.CPX' (5543 bytes)
Successfully wrote 'G:\LOCALE\EGA11.CPX' (7228 bytes)
Successfully wrote 'G:\LOCALE\EGA12.CPX' (8119 bytes)
Successfully wrote 'G:\LOCALE\EGA13.CPX' (6281 bytes)
Successfully wrote 'G:\LOCALE\EGA14.CPX' (7758 bytes)
Successfully wrote 'G:\LOCALE\EGA15.CPX' (6458 bytes)
Successfully wrote 'G:\LOCALE\EGA16.CPX' (7793 bytes)
Successfully wrote 'G:\LOCALE\EGA17.CPX' (8929 bytes)
Successfully wrote 'G:\LOCALE\EGA18.CPX' (5158 bytes)
Windows KBID 0x00000409
Will use DOS keyboard 'us' [US-English]
Unable to find an EGA file with codepage 936 [Chinese (Simplified)]
Will use codepage 437 [US-English]
Successfully wrote 'AUTOEXEC.BAT'
Finalizing, please wait...
Created: G:\autorun.inf
Created: G:\autorun.ico
Successfully remounted \\?\Volume{916c6b0b-a1c7-11e5-838f-acfdce08dba7}\ on G:

Found USB 2.0 device 'SMI USB DISK USB Device' (090C:1000)
Using autorun.inf label for drive G: '131MB'
1 device found
Disk type: Removable, Disk size: 131MB, Sector size: 512 bytes
Cylinders: 15, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00CDB8DB
Drive has a Rufus Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 124.9 MB (130991616 bytes)
  Start Sector: 63, Boot: Yes
Partition 2:
  Type: Rufus Extra (0xea)
  Size: 31.5 KB (32256 bytes)
  Start Sector: 255906, Boot: No

pbatard, 2018-04-15:
If you are getting read errors at any stage, then this is a hardware issue, and it has nothing to do with Rufus.

For instance, you could have a Power Supply issue to your USB ports (too many devices plugged, and some will have trouble drawing power), or flaky ports/cables, or many of the other issues that can cause a read failure.

Posting logs to this issue tracker won't help, because, clearly, you would get the same issues regardless of the application being used, and there is nothing that can be fixed, software-wise, to help you.