pbatard/rufus #633: Help, Rufus damaged my flash drive.

PamPam-42, 2015-11-05:
Yeah, I've read the FAQ.

Anyway, I was trying to create a bootable Windows 10 USB on my Kingston DataTraveler SE9, and I let check for bad blocks in 4 passes first. It was a very slow operation so I just let my PC work. When I later checked on it, I found that the operation failed and that my USB was now not formatted. And Windows 7 couldn't format it. Nor could a number of other USB/HDD formatting tools, including diskpart in Win7.

Here's what Rufus is saying now when I try to format the USB. Any help would be appreciated.
```
Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'J:' as volume mountpoint
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Volume does not have an x86 Partition Boot Record
Deleting partitions...
Could not delete drive layout: [0x0000045D] The request could not be performed because of an I/O device error.
Could not reset partitions
Re-mounted volume as 'J:' after error

Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1881, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 14.4 GB (15479597056 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1881, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 14.4 GB (15479597056 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
```

pbatard, 2015-11-05:
Well, if the bad block check failed, it seems fair to assume that your flash has an issue. It only takes trying to access a block of flash data that has a failure to turn a drive that was about to fail into a drive that failed completely. If you value your data, you should realize that it's much better to have a flash drive fail during a bad blocks check than when you try to read valuable data back from it.

If you expect me to say "Oh yeah, despite what I went great length to explain [in the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive), your drive can totally be saved and restored as good as new", prepare to be disappointed. Once flash memory, which is __NOT__ everlasting, starts to fail, you should try to use another drive, period.

closed 2015-11-05 by pbatard

PamPam-42, 2015-11-05:
The drive was working properly before the check. In fact, the bad blocks check completed through, and the operation failed afterwards, during the ISO extraction portion, I just didn't save those log files. I hoped you'd be able to provide some insight, seeing as you've got enough expertise to create a program that formats USB's. I guess I'll keep looking.

Thanks for taking the time to at least read my post.

pbatard, 2015-11-05:
> In fact, the bad blocks check completed through

That's new information. Why didn't you say so? Your original comment definitely gave the impression that the drive failed during the bad blocks check.

> operation failed afterwards, during the ISO extraction portion

It's still possible that your flash went bad during this 5th writing phase. For the record, Rufus gets downloaded about 2 million times each month, so if either its bad blocks check operation or its ISO write operation could damage the drive in any way (outside of regular flash failures, which can happen at any time, and are independent of the application used), the issue would be very apparent.

Still `I/O device error` tends to point to a hardware failure for your drive. You may want to try removing any extension cable (if you use one), and plugging in to another port and another computer, but if the problem persists, it basically means that your drive decided to fail as Rufus was writing ISO content to it, and the same would have happened if you had been manually copying files.