pbatard/rufus #637: v2.6 b806 - MS VC++ 2015 compiled time SDK warnings (Release, x86 and AMD64 build's).

VictorVG, 2015-11-09:
When assembling v2.6 b806, I received a number of compiler warnings MS VC++ 2015, but it seems in addition to the severity of syntax checking them unfounded, but I still think it's better to say about their availability - [**build list v2.6 b806**](http://pastebin.com/zdPR63zJ).

**P.S.**

Sorry, the machine on which the VS 2015 a stranger, so the path to the directory specified by the relative - or were not allowed to publish logs.

closed 2015-11-09 by pbatard

VictorVG, 2015-11-09:
Big thanks!

pbatard, 2015-11-09:
You're welcome. Thanks for reporting the issue.

VictorVG, 2015-11-09:
Well, we will Warnig destroyed, and my account to a company with forum.ru-board.com guys crash claimed the 06.11.2015. We look forward to restore or not ....

commit 60e90f7c (2015-11-09): [misc] fix VS2015 warnings when compiling ms-sys for x64

* Closes #637