pbatard/rufus #639: Not recognising USB drive

wielder91, 2015-11-12:
I have inserted the USB drive I wish to burn to, but it doesn't appear in the 'Drive' section at the top, apparently "because it didn't report itself as a disk".

Log, the important part is at the bottom:

Rufus version: 2.5.799
Windows version: Windows Vista SP2 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB device 'Generic USB CF Reader USB Device' (????:????)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_cf_reader&rev_1.01#058f312d81b1&1#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
Found USB device 'Generic USB MS Reader USB Device' (????:????)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_ms_reader&rev_1.03#058f312d81b1&3#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
Found USB 2.0 device 'Generic USB SD Reader USB Device' (058F:6362)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_sd_reader&rev_1.00#058f312d81b1&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
Found USB device 'Generic USB SM Reader USB Device' (????:????)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_sm_reader&rev_1.02#058f312d81b1&2#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB device 'Generic USB CF Reader USB Device' (????:????)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_cf_reader&rev_1.01#058f312d81b1&1#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
Found USB device 'Generic USB MS Reader USB Device' (????:????)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_ms_reader&rev_1.03#058f312d81b1&3#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
Found USB 2.0 device 'Generic USB SD Reader USB Device' (058F:6362)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_sd_reader&rev_1.00#058f312d81b1&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
Found USB device 'Generic USB SM Reader USB Device' (????:????)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_sm_reader&rev_1.02#058f312d81b1&2#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB device 'Generic USB CF Reader USB Device' (????:????)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_cf_reader&rev_1.01#058f312d81b1&1#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
Found USB device 'Generic USB MS Reader USB Device' (????:????)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_ms_reader&rev_1.03#058f312d81b1&3#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
Found USB 2.0 device 'Generic USB SD Reader USB Device' (058F:6362)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_sd_reader&rev_1.00#058f312d81b1&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
Found USB device 'Generic USB SM Reader USB Device' (????:????)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_sm_reader&rev_1.02#058f312d81b1&2#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
Found USB 2.0 device 'SanDisk Cruzer Facet USB Device' (0781:5576)
Could not open '\\?\usbstor#disk&ven_sandisk&prod_cruzer_facet&rev_1.00#4c530202831005118042&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found

pbatard, 2015-11-15:
I don't like this `[0x00000005] Access is denied.` that you get when trying to access the SanDisk Cruzer. The ones pertaining with the SD/CF/MS card can be understood, if there is no car in the drive, but the one about the Cruzer seems to indicate that there is something on your system (Security solution? Windows policy?) that is preventing access to the USB drive. This can happen, __for instance__ (I'm not saying this is your case, just using this as a possible example) on corporate PCs from companies that is paranoid about pluggable USB drives being used to exfiltrate data.

If that is the case, there isn't much I can do: if your OS is being restricted in any way in terms of accessing devices, I cannot guarantee the proper behaviour of Rufus.

So I guess my questions are:
* Have you tried on a different PC?
* Do you know if there are restrictive policies in place on that PC?
* Is your user account on that PC being restricted in any way?
* Have you checked whether you are using any of the __incompatible__ software listed [at the end of the Rufus FAQ](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with)?

You may also want to run with USB Debug enabled (<kbd>Alt</kbd>-<kbd>.</kbd>), which will provide more verbose info on device detection, but I doubt it will help much....

pbatard, 2015-11-30:
No update from OP &rarr; Closed.

closed 2015-11-30 by pbatard