pbatard/rufus #640: Error Message

mhmayyan, 2015-11-14:
Why I get error message "YO BNLA #2"

pbatard, 2015-11-14:
> Why I get error message "YO BNLA #2"

So that you report this error! :smile:
I deliberately made a [couple of error messages](https://github.com/pbatard/rufus/blob/master/src/dos_locale.c#L423-L424) very nonsensical so that they would get a higher chance to get reported.

If I was only saying something like `Unable to find an EGA file with codepage 720` (the actual error) I would probably never get these issues reported...

I'll make sure this is fixed in the next version. It would also help if you could provide your __FULL__ log rather than just an excerpt.

mhmayyan, 2015-11-14:
here is the full log

Rufus version: 2.5.799
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'SanDisk USB Flash Drive USB Device' (0781:557E)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001641FF
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16008577536 bytes)
  Start Sector: 63, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{74915251-53ec-11e5-b7b8-806e6f6e6963}\
Opened drive \\?\Volume{74915251-53ec-11e5-b7b8-806e6f6e6963} for write access
Writing partition boot record...
Using FreeDOS FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Successfully remounted Volume{74915251-53ec-11e5-b7b8-806e6f6e6963}\ on E:\
Copying DOS files...
Successfully wrote 'E:\COMMAND.COM' (65814 bytes)
Successfully wrote 'E:\KERNEL.SYS' (45494 bytes)
Successfully wrote 'E:\LOCALE\DISPLAY.EXE' (3648 bytes)
Successfully wrote 'E:\LOCALE\KEYB.EXE' (11446 bytes)
Successfully wrote 'E:\LOCALE\MODE.COM' (16254 bytes)
Successfully wrote 'E:\LOCALE\KEYBOARD.SYS' (40360 bytes)
Successfully wrote 'E:\LOCALE\KEYBRD2.SYS' (29750 bytes)
Successfully wrote 'E:\LOCALE\KEYBRD3.SYS' (32178 bytes)
Successfully wrote 'E:\LOCALE\KEYBRD4.SYS' (13105 bytes)
Successfully wrote 'E:\LOCALE\ega.cpx' (6464 bytes)
Successfully wrote 'E:\LOCALE\ega2.cpx' (7059 bytes)
Successfully wrote 'E:\LOCALE\ega3.cpx' (5469 bytes)
Successfully wrote 'E:\LOCALE\ega4.cpx' (4431 bytes)
Successfully wrote 'E:\LOCALE\ega5.cpx' (7217 bytes)
Successfully wrote 'E:\LOCALE\ega6.cpx' (7409 bytes)
Successfully wrote 'E:\LOCALE\ega7.cpx' (5387 bytes)
Successfully wrote 'E:\LOCALE\ega8.cpx' (6973 bytes)
Successfully wrote 'E:\LOCALE\ega9.cpx' (5785 bytes)
Successfully wrote 'E:\LOCALE\ega10.cpx' (5543 bytes)
Successfully wrote 'E:\LOCALE\ega11.cpx' (7228 bytes)
Successfully wrote 'E:\LOCALE\ega12.cpx' (8119 bytes)
Successfully wrote 'E:\LOCALE\ega13.cpx' (6281 bytes)
Successfully wrote 'E:\LOCALE\ega14.cpx' (7758 bytes)
Successfully wrote 'E:\LOCALE\ega15.cpx' (6458 bytes)
Successfully wrote 'E:\LOCALE\ega16.cpx' (7793 bytes)
Successfully wrote 'E:\LOCALE\ega17.cpx' (8929 bytes)
Successfully wrote 'E:\LOCALE\ega18.cpx' (5158 bytes)
Windows KBID 0x00000409
Will use DOS keyboard 'us' [US-English]
Unable to find an EGA file with codepage 720 [Someone missed a codepage!]
Will use codepage 437 [US-English]
Successfully wrote 'AUTOEXEC.BAT'
Finalizing, please wait...
Successfully remounted Volume{74915251-53ec-11e5-b7b8-806e6f6e6963}\ on E:\

Found USB 2.0 device 'SanDisk USB Flash Drive USB Device' (0781:557E)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0040B174
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16007561216 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2015-11-15:
Thanks. I will have a fix in next version.

commit f5813e9c (2015-11-15): [dos] add missing locale name for CP-720

* Closes #640
closed 2015-11-15 by pbatard