pbatard/rufus #641: Create windows install media same way as Microsoft Mediacreationtool

Faguy, 2015-11-18:
If I create a Windows 10 USB install media using MediaCreationTool.exe this USB drive can be used to boot on both a legacy bios (or UEFI-CSM) and on a proper UEFI system. This drive have an MBR and an active FAT32 primary partition. I can't create a similar drive with rufus using any of the 3 options... the only one that boots on a legacy system is the first but only if I create the partition in NTFS that can't work on a UEFI. If I create the partition in FAT32 the legacy system stop with "Invalid system disk".

pbatard, 2015-11-18:
> but only if I create the partition in NTFS that can't work on a UEFI

Oh, [__how wrong you are!__](https://github.com/pbatard/uefi-ntfs).

Let it be repeated __LOUD AND CLEAR__: There's nothing in the UEFI specs that says you can only boot from FAT32. As a matter of fact, UEFI systems have no issue booting from NTFS, __as long as your provide an UEFI NTFS file system driver__ (which Rufus conveniently provides for you).

How I wish the myth of UEFI being only compatible with FAT32 was dispelled once and for all.

As to your other question:

> (a) drive (that could) be used to boot on both a legacy bios (or UEFI-CSM) and on a proper UEFI system

You can... if you [read the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_both_BIOS_and_UEFI_mode).

Just use <kbd>Alt</kbd>-<kbd>E</kbd> to enable dual BIOS+UEFI mode for Windows installations. And yes, you __can__ (and should) use NTFS as the target file system, as Rufus will install `UEFI:NTFS` to allow an UEFI system to boot from NTFS.

Note however that there is a small bug in version 2.5, that prevents disabling <kbd>Alt</kbd>-<kbd>E</kbd> once enabled. This will be fixed in 2.6.

If this doesn't answer your question, feel free to reopen the issue.

closed 2015-11-18 by pbatard

Faguy, 2015-11-19:
Thank you for the support but I've not been able to create an USB disk MBR+FAT32 (like that created with Mediacreationtool) that can boot on my computer on legacy bios, even after using the "cheat code" Alt+E. The USB drive boot fine on my UEFI pc, even if I don't use the cheat code. If I've understand the link you provide if Rufus is set to create a NTFS partition it actually create two partitions? One NTFS and one FAT32? I'm a little confused...

pbatard, 2015-11-19:
> but I've not been able to create an USB disk MBR+FAT32

That's because you should create an MBR+NTFS disk (and yes, that NTFS disk will boot on UEFI).
You can ignore the extra FAT32 partition (which is set to install the UEFI NTFS driver). All you need to know is that this extra FAT32 partition is there to allow boot from the first NTFS partition.

Please make sure you use the following options:
*  <kbd>Alt</kbd>-<kbd>E</kbd>
* MBR partition scheme for BIOS or UEFI
* NTFS file system

If it still doesn't work, please provide a __COMPLETE__ log from Rufus from when you created your drive.

Faguy, 2015-11-19:
It worked :)
It's a little different from the USB disk created by the microsoft tool though. On the legacy system it prints a message (press a key to boot from usb...) that the microsoft usb don't. On the UEFI system prints some text (probably the loading of the NTFS driver).
I take that's impossible to create a usb key similar to that created by the microsoft tool but created beginning from an ISO image?

pbatard, 2015-11-19:
>  On the legacy system it prints a message (press a key to boot from usb...) that the microsoft usb don't.

... which you can remove by unchecking `Use Rufus MBR with BIOS ID...` in Rufus under Advanced Options (click on the white triangle). But if you realize that it is there to avoid having to be in front of the computer on the first reboot, so that you can perform an __unattended__ install, which the MS tool won't allow you to do, you should also realize that you are better off keeping that option.

> I take that's impossible to create a usb key similar to that created by the microsoft tool

Different applications do what they think is best for the __user__. I disagree with some of Microsoft choices, and, until proven otherwise (it worked, didn't it?), I will kindly ask that you write your own tool, with your own set of preferences, if you are unhappy with both what the MS tool or Rufus are doing.

Faguy, 2015-11-19:
I've successfully created an *universal* usb boot MBR+FAT32 drive using Rufus (no need of the cheat code) and after that using the *bootsect* command under administrative windows command prompt. Is that possible to add to Rufus?

I think that the Rufus interface can be made more clear for the user... perhaps indicate that when you select NTFS it creates a NTFS+FAT32 boot partition? I don't know why you say that's possible to boot a NTFS partition under uefi while Rufus creates a FAT32 partition to boot from...

pbatard, 2015-11-19:
> perhaps indicate that when you select NTFS it creates a NTFS+FAT32 boot partition?

Except, with your comments you are providing the perfect example as to why this is __not__ a good idea. It only took one mention of me mentioning that there was a FAT32 partition created for you to latch on to it and get confused. had I mentioned that it was simply a small area to load the NTFS driver, you'd have been fine with it.

Oh, and some systems (some Intel and ASUS computers) will boot from NTFS even without needing to load a driver (so, just from an NTFS partition), so, __yes__ it is possible to boot straight from an NTFS partition under UEFI - it just takes an update from your computer/motherboard manufacturer to do so, or you injecting an NTFS driver into your firmware if you don't already have one.

Plus, you do realize that there are some ISOs out there (including official ones from Microsoft) that contain an `install.wim` file larger than 4 GB out there, and that are therefore incompatible with FAT32. To me that's more than reason enough not to encourage FAT32 usage, because then I suddenly would have to explain to non tech-savvy people, who simply want to install Windows ASAP that some file systems have limitations and that the UEFI committee was really short-sighted in picking FAT32 as their go-to file system.

As you indicated, the BIOS+UEFI NTFS drive created by Rufus __does work__ (and is about as _universal_ as the FAT32 one). You tested it yourself. But now, for trifling differences, such as 4 extra lines of inconsequential text at boot, and a prompt, which can be disabled if you really want to, and which is really there to do a much better job on unattended installs than what the MS tool does, you are asking me to add yet another way to perform the same working operation, and as a result, have to educate users on why there are 2 different ways to achieve one job, when only one is really needed?

Sorry, but that's just not gonna happen.

Faguy, 2015-11-20:
Ok, now I understand more... just want to add that I opened this issue because I found Rufus a very useful tool and wanted to use it for this task. The options are there to create such a disk.

Now you've linked me to the FAQ that explain the options but I still think that there's something wrong.

On the FAQ you say: *"Especially, the Windows 8 installation ISOs, that support both UEFI and BIOS boot, will be converted to either one or the other mode, depending on the option you selected under Partition scheme and target type: If you select the first option (MBR partition scheme for BIOS or UEFI computer), the USB will be bootable in BIOS-mode only (even on UEFI systems), and if you select any of the other options, the USB will be bootable in UEFI mode only (and not bootable on a BIOS system at all). "*

But if I select "MBR partition scheme for BIOS or UEFI computer" with FAT32 the USB will *not* be bootable in BIOS-mode but instead *is* bootable in UEFI-mode. The exact opposite...

I think that the FAT32 drive is missing a step to make it bootable in BIOS-mode? That's there if I select NTFS?

The other two options are there for the more advanced configurations that are possible with UEFI, so for them is not better to explain what Rufus is doing?