pbatard/rufus #642: Compile Error (VS2015)

Buzzroid, 2015-11-20:
I've got lots of the same errors as like,

Error   C2001   newline in constant (compiling source file ..\icon.c)   rufus   d:\ripper-projects\rufus\src\rufus.h    45

Thanks.

James

pbatard, 2015-11-20:
How did you obtain the source? Did you clone from git? Somewhere else?

You are the only person reporting this issue, and I am using VS2015 for all Rufus development (and clone it from scratch on regular basis), so this is most likely an issue with your end.

pbatard, 2015-11-30:
No update from OP &rarr; Closed.

closed 2015-11-30 by pbatard

ljporljp, 2016-09-05:
My page code is 936(Chinese Simplified),i really cloned from git, and this problem confuses me long time!
Is the file-encoding problem?

pbatard, 2016-09-05:
You probably misconfigured your git newline options. At any rate this is NOT a Rufus issue, and you will have to figure it out by yourself.

ljporljp, 2016-09-05:
thinks a lot ! is that to say, the git config influences the file-encoding or not ?

pbatard, 2016-09-05:
Yes it does. Please become familiar with git, Visual Studio, and newline handling. Google is your friend!

ljporljp, 2016-09-05:
ok,i will search to learn all the things you said, but because of some policy limit, my country can not use google, -.-..thank you so much !

zhengfish, 2017-07-13:
change System language locale to 'English'