pbatard/rufus #644: UEFI W10 ISO won't boot on DELL XPS 9550

cbaldan, 2015-11-22:
Hello.

I could not get a Rufus prepared UEFI USB stick to boot as UEFI in the new DELL XPS 9550 laptop.

ISO was downloaded from Microsoft using the Media Creation Tool.
Rufus 2.5 portable
Laptop firmware, 1.0.7 (latest)

I tried using two different options for the partition scheme dropdown:
1) The auto-selected value after the ISO file was selected - MBR partition scheme for BIOS or UEFI-CSM
2) Second time - manually changed the scheme to "GPT partition scheme for UEFI"

The stick was recognized for boot only if Legacy Bios Boot option was enabled.
I was only able to boot an UEFI W10 installation using the Microsoft Media Creation Tool to create the USB stick.

Preivously, I have used Rufus to create a W7 UEFI USB stick, and it worked, but it was a desktop PC with an ASUS motherboard. Seems this laptop UEFI is more strict in some way.

pbatard, 2015-11-23:
> ISO was downloaded from Microsoft using the Media Creation Tool.

Avoid that. The ISOs generated by the Media Creation Tool have issues (for instance, the `install.wim` on them is incompatible with Microsoft's own WIM extraction APIs, even on Windows 10).

Always try to get the retail Windows 10 ISOs, which you can download freely straight from Microsoft [here](https://www.microsoft.com/en-us/software-download/techbench).

Also, a __complete log__ from Rufus would have been nice, though I suspect this is a pure issue with your BISO options:

1) If you are using _MBR partition scheme for BIOS or UEFI-CSM_, then your ISO will ONLY be bootable in legacy BIOS. "UEFI-CSM" actually means just that (as opposed to pure UEFI).
2) If you want to boot in pure UEFI mode, then you should indeed use _GPT partition scheme for UEFI_, but you may also have to disable Secure Boot.

cbaldan, 2015-11-23:
Hello pbatard.

Thanks for the tip, I did not know that Media Creation Tool had such issue. Nice to know about that URL.

I will give a try using that URL you provided, just for the sake of it.
I'll grab the log and provide to you, not sure it will contain anything useful, since I think the issue is related to laptop UEFI, as I mentioned. Rufus did work very well.

My bad, I forgot to say in the ticket, but I did try all combinations of 'Secure Boot' on and off.
The stick was only listed under Legacy boot devices on the boot menu.

I'm not sure how the HDD partition scheme will affect UEFI recognizing the USB stick as a UEFI enabled. Perhaps some specific configs vendors might have that issue, but was not the case in this laptop.
Once I created the stick using 'Media Creation Tool' it worked: the HD had MBR partition, and the setup made me delete the partition and change to GPT, so I could get my W10 running with Fast Boot mode enabled.

pbatard, 2015-11-24:
> The HDD partition scheme should not affect UEFI recognizing the USB stick as UEFI enabled

You are assuming wrong here. In the case of _MBR for BIOS or UEFI-CSM_, and with the default settings, Rufus will create an NTFS drive which in 99.9% of cases will not be bootable in UEFI mode (since you'd need to have an NTFS driver installed in your UEFI environment to "see" the partition).

So, contrary to what you believe, the option you select in Rufus very much affects you ability to detect the USB in UEFi mode. If you want to install Windows in pure UEFI mode with Rufus (or get your device detected as an UEFI bootable device), you __must__ select either the second or third option, __not__ the first one.

The reasoning behind this is that, a large number of Rufus users are not experience enough to be able to set their boot mode to legacy or pure UEFI on their own, so, to ensure that someone doesn't end up installing their OS in legacy mode, when they wanted to install it in UEFI mode, it is much better to ensure that only one of those will be active at any one time, depending on the option you use.

As I stated before, as long as you see "UEFI-CSM" mentioned anywhere, it means that only BIOS compatibility will be active, and pure UEFI mode will be disabled.

cbaldan, 2015-11-24:
> I tried using two different options for the partition scheme dropdown:
> 1) The auto-selected value after the ISO file was selected - MBR partition scheme for BIOS or UEFI-CSM
> 2) Second time - manually changed the scheme to "GPT partition scheme for UEFI"

If you read carefully my original post, you'll see that I did try the `first` and `third` options, which I layed out the complete description - __GPT partition scheme for UEFI__ - and yet, it didn`t work.

> The HDD partition scheme should not affect UEFI recognizing the USB stick as UEFI enabled. 

In my second post, I did not say that the partition scheme in the __USB drive__ didn`t matter, I said the __HDD__ partition scheme should not affect __USB drive__ being detected as UEFI enabled.

You seem to be sure this is a user misuse case, so, let's facilitate our lives and close this ticket once and for all.

micahg, 2016-02-19:
Hi, was there a workaround for this? I seem to have run into the same problem on my XPS.

cbaldan, 2016-02-19:
I used Microsoft Media Creation Tool.
Rufus was unable to create a bootable UEFI usb media that worked with my
XPS.

micahg, 2016-02-19:
OK, ty.

pbatard, 2016-02-19:
Or you can post a complete log from Rufus (that's what the prominent Log button is meant for), so that I can take a look at how your USB was created, which may help identify the issue.

micahg, 2016-02-19:
I'll do both this evening. Thanks fellas! Happy to help.

cbaldan, 2016-02-19:
I'll work on that and report back.

pbatard, 2016-02-19:
Appreciate that. I definitely want to investigate this further.

micahg, 2016-02-19:
[rufuslog.zip](https://github.com/pbatard/rufus/files/138869/rufuslog.zip)

micahg, 2016-02-19:
Logs attached. Thanks in advance!

pbatard, 2016-02-20:
Thanks. According to your log you are partitioning in MBR mode:
```
Partitioning (MBR)...
```
Be mindful that if you use MBR partitioning, the UEFI booting will be __disabled__ by Rufus (see [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_both_BIOS_and_UEFI_mode)), so the only mode that will work is BIOS-Compatibility/CSM... which may not be enabled by your UEFI firmware.

I assume that the XPS is set to only boot pure UEFI by default so, if you want to install Windows in UEFI mode you will need to:
1. Select `GPT partition scheme for UEFI` in Rufus under _Partition scheme and target system type_
2. Disable Secure Boot if enabled on your target machine

micahg, 2016-02-20:
I swear I selected GPT... Not sure what to tell you. I'll rerun the test just to make sure its not a user malfunction.

pbatard, 2016-02-20:
Make sure you change the partition scheme __AFTER__ you select the ISO.
Picking an ISO will reset a bunch of settings (which I know can be confusing, but I'll be working to make that more explicit in Rufus 3.0 #117).

pbatard, 2016-02-22:
Any update on this?

micahg, 2016-02-22:
Hi, sorry -- I'll try to retest tonight.

pbatard, 2016-02-24:
I guess I'll close this issue. Feel free to reopen it if you have information that appears to indicate that this wasn't due to the choice of settings in Rufus.

closed 2016-02-24 by pbatard

ForcedFeed, 2016-08-07:
I think people don't notice that the scheme is set back to "MBR partition scheme for BIOS or UEFI-CSM" after selecting the ISO. If you aren't paying attention and set GPT / UEFI before selecting the ISO, it's easy to miss that the scheme has been changed back.

kodas2, 2016-12-11:
I'm having this issue of a non bootable USB stick, created with Windows 10 32-bit iso, from the media creation tool. The USB stick is 8 GB. What I did before: Partially zero'd the USB stick (around 20%). I tried using the iso with GPT for UEFI. It didn't boot. I then tried the same settings, ensuring that GPT for UEFI is selected. Here is the log: 
[rufus_notworking.txt](https://github.com/pbatard/rufus/files/644829/rufus_notworking.txt)