pbatard/rufus #646: Error extracting image (Windows to Go) [0xC03A0005] Access is denied

ziaulrehman, 2015-11-28:
Log:
Rufus version: 2.5.799
Windows version: Windows 8.1 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x2585BC2E
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8002165248 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 256 KB (262144 bytes)
  Start Sector: 15631277, Boot: No, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'Win10_10586_x64_Nov2015'
  Size: 4392968192 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
  Uses Install.wim: Yes (version 0.13.1)
Using image: Win10_10586_x64_Nov2015_5in1_en-US_murphy78.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'J:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{b20543d9-2625-11e5-8285-84349790f0a9}\
Opened drive \\?\Volume{b20543d9-2625-11e5-8285-84349790f0a9} for write access
Writing partition boot record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted Volume{b20543d9-2625-11e5-8285-84349790f0a9}\ on J:\
Applying Windows image...
Windows To Go mode selected
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Could not open ISO 'F:\downloads\Completed\10\Windows 10 5in1 10586 x64 en-US Nov2015\Win10_10586_x64_Nov2015_5in1_en-US_murphy78.iso': [0xC03A0005] Access is denied.
Could not mount ISO for Windows To Go installation

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x2588C5E3
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8002165248 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 256 KB (262144 bytes)
  Start Sector: 15631277, Boot: No, Recognized: Yes

As you can see formt he log, i am having access denied issue, while my iso is being written normally in create windows installation and is normally being mounted to windows.
I recieved same error when trying to create windows to go on my windows 8.1 x64 and target iso was of windows 10 x86.
then i searched and read that windows version must be same, so i tried creating windows 8.1 x32 from my same windows 8.1 x64 and error is same.
I don't know if this is the issue or some thing which should be addressed properly in windows to go FAQs, but i hope this will be helpful for lot of people if addressed.
Thakyou for your efforts.

pbatard, 2015-11-28:
Please try an __OFFICIAL__ Windows 10 image, such as the ones you can legally download from [Microsoft Tech Bench](https://www.microsoft.com/en-us/software-download/techbench).

I do not support custom/pirated Windows images, especially Windows To Go will NEVER be supported for unofficial images that try to include multiple versions of Windows into one ISO.

Also, it seems that the issue is with the mounting of the ISO as a virtual optical drive. But Rufus uses the Windows native API to mount such images, so I suspect that this is a problem with your environment. Did you (or you system administrator) apply any system policy that might have disabled mounting of ISOs? I am fairly confident that if you just try to mount the same ISO from Windows Explorer (usually by double clicking on it, or using right click &rarr; Open With &rarr; Windows Explorer), you will have the same issue, in which case this is clearly not a Rufus problem.

pbatard, 2015-11-28:
Just for the record, I tried that specific ISO (on Windows 10 x64) and didn't run into any problem during the Windows To Go drive creation:

```
Rufus version: 2.5.799
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0809
0 devices found
Checking for Rufus updates...
Checking release channel...
No new release version found.
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'Install Debian GNU/Linux'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00C5765A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16012894208 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'Install Debian GNU/Linux'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00C5765A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16012894208 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'Win10_10586_x64_Nov2015'
  Size: 4392968192 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
  Uses Install.wim: Yes (version 0.13.1)
Using image: Win10_10586_x64_Nov2015_5in1_en-US_murphy78.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE8 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Adding extra partition
[... 365 lines cut ...]
   7855784 KB available on disk.
      4096 bytes in each allocation unit.
   3909134 total allocation units on disk.
   1963946 allocation units available on disk.
NTFS Fixup completed.

Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'Win10_10586_x64_Nov2015'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00B7D603
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.9 GB (16011813888 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 256 KB (262144 bytes)
  Start Sector: 31275122, Boot: No, Recognized: Yes
```

I will therefore close this issue.

closed 2015-11-28 by pbatard