pbatard/rufus #648: Support Kali

Machou, 2015-12-01:
Bonjour

Peux-tu ajouter le support de Kali Linux stp ?

pbatard, 2015-12-01:
Can you please detail the issue you are encountering? Kali should be supported (unless you are trying to use their [custom and NON UEFI COMPLIANT](https://github.com/pbatard/rufus/issues/526#issuecomment-111835917) EFI loader.

If you are encountering an error during the drive creation, please provide a __FULL__ log from Rufus. This will tell me the ISO you've been using, your OS, and most of the options you used to format the drive, as well as the error, which is the only way I can look into it.

Machou, 2015-12-01:
I don't have a problem, but i don't see Kali in the list :P

pbatard, 2015-12-01:
You mean the [list on the homepage](http://rufus.akeo.ie/#ref2) that specifically says _"non exhaustive"_ (which, I should point out, means the same in English and in French...)?

Because of the UEFI issue, I'm not planning to add Kali to that list (and, to be honest, I think the list if big enough as it is so I'm not really planning to add more images, especially Linux derivatives which are expected to work).

closed 2015-12-01 by pbatard