pbatard/rufus #649: Windows 10 Education version 1511 build 10586.14 not recognized as Windows To Go

nielsvf, 2015-12-02:
Screenshot:

[image]

This is the build which was pulled from the Microsoft Update servers but I managed to grab it, so it isn't the insider build. Not sure if this is just for the Education version though, but might be for the build overall. Sorry for my bad English.

pbatard, 2015-12-02:
See the 3rd bullet point from the [Windows To Go entry from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Windows_To_Go_issues). If you look at the log (which you should always do when you see an issue), you will most likely see the line:
```
  Uses Install.wim: Yes (version 0.14.0)
  Note: This WIM version is NOT compatible with Windows To Go
```

But can you please indicate:
1. Where you downloaded the ISO or how you created it. Did you use the Windows update tool by any chance?
2. What is the __exact__ name of the ISO you are using.
3. What is the SHA-1 of the ISO you are using. You can get the SHA-1 by selecting the ISO in Rufus can clicking on the ```#```  sign on the status line.

nielsvf, 2015-12-02:
I download it through the Update Tool, so it isn't compatible indeed. I haven't been on GitHub a while and pretty tired today, and didn't see it. Feel free to delete the issue. My bad, sorry for wasting your time.

pbatard, 2015-12-02:
Nah, that's okay &mdash; I reckon that it's easy to miss. WTG should be available with Tech bench images if you need it (which are official and freely available... though of course I think Microsoft reverted the availability of the 1511 ISOs...)

closed 2015-12-02 by pbatard