pbatard/rufus #650: Requesting disk access issue - Win7HP

s6eko, 2015-12-09:
try to burn onto flash drive but stays on requesting disk access until the error occurs : Cannot open media. I previously may have burnt the file onto the disk but may be incomplete or might be ok, if it has burnt fully then i would really appreciate some feedback
Rufus version: 2.5.799
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'USB DISK 2.0 USB Device' (26BD:9917)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1885, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x005B4451
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.4 GB (15507390464 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'ESD-ISO'
  Size: 3514171392 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: pro81-64.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
Could not get exclusive access to device \\?\Volume{51683361-938c-11e5-b20d-083e8e8dc33a}: [0x00000005] Access is denied.
Could not lock volume
Re-mounted volume as 'E:' after error

Found USB 2.0 device 'USB DISK 2.0 USB Device' (26BD:9917)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1885, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x005B4451
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.4 GB (15507390464 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB 2.0 device 'USB DISK 2.0 USB Device' (26BD:9917)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1885, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x005B4451
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.4 GB (15507390464 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2015-12-09:
This error means that there is __another__ application on your computer how has taken exclusive access to the USB and won't relinquish access.

I'm afraid that you will have to find what that application might be, as Windows (and therefore Rufus) does not currently provide means to tell which application this might be (it can do that for some files, but not for whole drives, which is what would be needed).

A lot of the time, this has to with poorly designed security solutions. Or it can be the software listed [here](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with).

Now, if you are really sure that whatever application has taken a lock to the drive is unlikely to use it, you can try the <kbd>Alt</kbd>-<kbd>,</kbd> [cheat mode](https://github.com/pbatard/rufus/wiki/FAQ#Power_keysCheat_modes) in Rufus, which can disable the request for exclusive access. But if that doesn't work, or creates any kind of problem, you're on your own.

s6eko, 2015-12-09:
Ok, really appreciate it, also the usb already has a number of files on it, so it may have already fully burnt anyway, is there a way for me to check for certain?

pbatard, 2015-12-10:
The only way to be certain is if you have the log from when the USB was created. Without this, you'll just have to try and see if it works.

s6eko, 2015-12-10:
ok, thanks anyway ;)

pbatard, 2015-12-15:
Since there isn't much more I can do, I will close this issue now.

closed 2015-12-15 by pbatard