pbatard/rufus #651: Zero Drive Functionality cannot be disabled once triggrered

deadair3210, 2015-12-11:
Version = 2.6.816 Beta
If the Zero Drive cheat mode (Alt-Z) is enabled any time in a runtime of the program, the mode cannot be disabled unless the program is restarted

Procedure
1 - Enable Alt-Z
2 - Hit cancel
3 - Even though the user hit cancel, and possibly changed options in the program, once start is pressed the program will begin Zeroing the drive

pbatard, 2015-12-11:
Thanks for the report.

I'd like to fix the problem, but I'm afraid, even with the added explanation, I still can't reproduce it. I don't think I quite understand the exact steps that are needed to see the issue. Can you please tell exactly which controls one needs to press, from a clean start of Rufus, to see the issue?

pbatard, 2015-12-11:
Ah, I get you know - you need to press 'Cancel' on the first prompt after <kbd>Alt</kbd>-<kbd>Z</kbd>. Let me fix that.

pbatard, 2015-12-11:
This should be fixed now. I don't think I want to publish a new BETA, but I did put a TEST version with the fix that will go into 2.6 release [here](https://rufus.akeo.ie/testing/). I will close this issue, but don't hesitate to let me know if you still see issues related to zeroing drives in the new TEST version.

Many thanks again for testing the BETA and spotting this issue!

commit fe3241d0 (2015-12-11): [core] ensure that drive zeroing is reset on user cancel

* Closes #651
closed 2015-12-11 by pbatard