pbatard/rufus #652: Block by block copy of linux image not working

hifi25nl, 2015-12-15:
Some friend is trying to make a block by block copy of an archlinux image (EFI fat32  + ext4 OS partition + MBR) to a USB stick with Rufus with the option DD
The original image is:
-----
Disk /dev/sdc: 14,9 GiB, 16013942784 bytes, 31277232 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 524A1828-2002-4187-B469-3E7A2DE53ED0
                                                                                                                             
Dispositivo   Start     Fine  Settori  Size Tipo                                                                             
/dev/sdc1      2048  1001471   999424  488M EFI System                                                                       
/dev/sdc2   1001472 27826175 26824704 12,8G Linux filesystem 
-----

However, after the copy, if I check it with gparted there is this error
<i>dumpe2fs 1.42.13 (17-May-2015)
Journal superblock magic number invalid!</i>
and the ext4 partition cannot be mounted.

If I copy it with this dd command in linux
gunzip     -c     /mnt/audiolinux_mini.img.gz     |     dd of=/dev/sdd
all is fine.

What could be wrong?

pbatard, 2015-12-15:
I will need to know the exact image you used: What version of Arch Linux was it? Where exactly did you download it? A link to the ISO/IMG you downloaded would work wonders...

hifi25nl, 2015-12-15:
I have sent you a link by email.

pbatard, 2015-12-15:
Thanks. I'm downloading your file and will try to run a test.

But I also see that this is a `.img.gz`, which means that Rufus will uncompress it on the fly before writing it in `dd` mode.

On-the-fly uncompression in Rufus is fairly recent and hasn't been extensively tested however (because, to be fair, few people seem to be using compressed `dd` images with Rufus), so it is very possible that there are issues that need to be fixed. I do recall that I once got a somewhat similar issue with a Raspian compressed image, which of course, I tried to investigate as a possible bug with the decompression library. However, when I tried that (by writing the same image multiple times, and comparing the data), I found that everything was fine and the newly written images worked as expected, so I had to dismiss the issue I saw as a possible fluke.

I guess I will run some more tests again to see if I can get to the end of it.

What I would suggest you do in the meantime is:
* Try to write the compressed image again with Rufus, and see if that works.
* Uncompress the image before writing it with Rufus. You can do that very easily if you open it in 7-Zip to convert from `.img.gz` to `.img`. As far as I know, noone has ever had any issue with uncompressed images in Rufus.

hifi25nl, 2015-12-15:
I have made the copy another time and now it is working. See attached files.
The pen initially was a new, never used Sandisk Extreme with Fat32.
The first attempt was unsuccessful.
After I have copied the image in linux with dd and all was fine.
Now is the second attempt on the pen already written with linux and all 
is right.

My doubt is that, if I make the same with a brand new pen, the problem 
will remain...I have finished usb sticks now...

I wait for your results.

Thank you

Piero

My doubt is that if I use a new

-- 

Piero Olmeda su Facebook <https://www.facebook.com/pieroolmeda>
Blog di Piero Olmeda <http://pieroolmeda.blogspot.it>
Piero Olmeda su Google+ 
<https://plus.google.com/117349928118716099173/posts>
Piero Olmeda su Linkedin 
<https://www.linkedin.com/pub/piero-olmeda/a5/102/b66>
Profilo autore su Amazon <http://www.amazon.com/Piero-Olmeda/e/B00YEG8VO8>
Profilo autore su Goodreads 
<https://www.goodreads.com/author/show/14038972.Piero_Olmeda>

Logo <http://audio-linux.com>

The backup GPT table is corrupt, but the primary appears OK, so that will be used.
Disk /dev/sdb: 14.9 GiB, 16013942784 bytes, 31277232 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 524A1828-2002-4187-B469-3E7A2DE53ED0

Device       Start      End  Sectors  Size Type
/dev/sdb1     2048  1001471   999424  488M EFI System
/dev/sdb2  1001472 27826175 26824704 12.8G Linux filesystem

hifi25nl, 2015-12-16:
Hello, it seems that there was a problem with UEFI boot with my image.
Maybe Rufus is really working fine in this case. I will see.

Sorry for the inconvenience and thanks

Piero

-- 

Piero Olmeda su Facebook <https://www.facebook.com/pieroolmeda>
Blog di Piero Olmeda <http://pieroolmeda.blogspot.it>
Piero Olmeda su Google+ 
<https://plus.google.com/117349928118716099173/posts>
Piero Olmeda su Linkedin 
<https://www.linkedin.com/pub/piero-olmeda/a5/102/b66>
Profilo autore su Amazon <http://www.amazon.com/Piero-Olmeda/e/B00YEG8VO8>
Profilo autore su Goodreads 
<https://www.goodreads.com/author/show/14038972.Piero_Olmeda>

Logo <http://audio-linux.com>

pbatard, 2015-12-16:
Okay.

I ran a comparison test, by writing your image a few times onto a 32 USB 3.0 drive, and then comparing that data against the uncompressed `.img`, and I did not see any differences for the whole content, except for benign differences on the [GPT Partition Table Header](https://en.wikipedia.org/wiki/GUID_Partition_Table#Partition_table_header_.28LBA_1.29), for the Backup LBA and Last Usable LBA (as well as the CRC) which are due to Windows automatically updating those according to the disk size and have nothing to do with Rufus.

I will of course continue to run some tests against compressed images as they come along, but as you indicate that the issue you originally observed might have been due to UEFI booting, and the one issue I observed on my own also seems to have been a fluke, I will close this issue for now.

closed 2015-12-16 by pbatard