pbatard/rufus #654: Fedora-Live-Cinnamon-x86_64-23-10.iso

gbanks, 2015-12-17:
Rufus completes the install onto the usb. It boots into the Fedora live installer and begins installing but fails.

pbatard, 2015-12-18:
I'll try to have a look when I have a chance (might be a while), but getting more data about the actual error your observe during boot, as well as the boot mode you use (is it BIOS? UEFI?) would probably help too.

gbanks, 2015-12-18:
I think it was UEFI. Sorry I don't remember more about the error. I just decided to use regular Linux Mint Cinnamon.

pbatard, 2016-01-04:
Well, I'm not seeing any issue booting in UEFI mode, and I'm afraid that, without an __EXACT__ report of the error, I'm not going to test a full Fedora installation to see if this might possibly be related to Rufus or, __more likely__ if this is simply a compatibility/environmental issue between Fedora installer and OP's machine.

Closing.

closed 2016-01-04 by pbatard

garfieldbanks, 2016-01-04:
Is it possible to test this using vmware?

pbatard, 2016-01-05:
Yes but it takes time. Time that I don't really have (Rufus is not a commercial endeavour - it's really just one guy doing development in his limited spare time, for free, and this specific issue is one of many which I have to look into), especially as you are asking me to compensate for something that you should really have done in the first place, which was write down the __exact__ error you saw.

At this stage, and since nobody else complained about Fedora installs (Rufus gets downloaded about 2 million times each month, so it's fair to assume that a few of these downloads will be for Fedora installs) I estimate it much more likely that this is an issue with between the Fedora installer and your specific configuration, and that it has nothing to do with Rufus, even more so as, once something has booted, it's really up to the application that booted to take over.

garfieldbanks, 2016-01-05:
I understand. Sorry I didn't write that error down.