pbatard/rufus #655: FAIL: Cant get device acces

OfficialJoe, 2015-12-18:
I cannot get access to my USB, I do no unterstand where to look to here my log

'''

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Could not get exclusive access to device \\?\Volume{985c7715-a588-11e5-9e46-448a5b83c1e8}: [0x00000005] Toegang geweigerd.
Could not lock volume
Re-mounted volume as 'G:' after error

Found USB 2.0 device 'USB20 USB USB Device' (FAFA:FAFA)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3843, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive has an unknown Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 29.4 GB (31610958848 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
Found USB 2.0 device 'USB20 USB USB Device' (FAFA:FAFA)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3843, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive has an unknown Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 29.4 GB (31610958848 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
'''

pbatard, 2015-12-18:
`Could not get exclusive access` means that you have another application that has taken an exclusive lock to the drive. I'm afraid you will have to figure out what application this might be (as Windows does not make it possible to find what application may have taken a lock for a whole drive).

First, you should check if you are using any of the applications listed [at the end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with), bearing in mind that the list provided is not exhaustive and there are probably many more applications that I am not yet aware of.

Now, if you are absolutely sure that whatever application has taken an exclusive lock will not try to access the drive while Rufus is formatting it, you can try to press <kbd>Alt</kbd>-<kbd>,</kbd> in Rufus, as indicated [here](https://github.com/pbatard/rufus/wiki/FAQ#Power_keysCheat_modes), which tells Rufus not to try to get an exclusive lock during formatting. But there is no guarantee that this will work, or not interfere with other software running on your computer.

pbatard, 2016-01-04:
Closing this issue.

closed 2016-01-04 by pbatard