pbatard/rufus #656: Check for updates not detecting update

cookieboyeli, 2015-12-23:
Rufus 2.5.799 did not find any new updates on a manual check despite 2.6.818 being released. 
2.6.816 Beta was found with check for beta updates was enabled.

In addition, check for beta updates cannot be toggled on if automatic checking for updates is disabled. (Probably should be separate issue).

```
Rufus version: 2.5.799
Windows version: Windows 8.1 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Could not get node connection information (V2) for device 'USB\VID_058F&PID_6366\058F63666438': [0x00000001] Incorrect function.
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Device eliminated because it appears to contain no media
0 devices found
Checking for Rufus updates...
Checking release channel...
No new release version found.
```

pbatard, 2015-12-23:
I usually wait about a da

closed 2015-12-23 by pbatard

cookieboyeli, 2015-12-23:
I figured it was something like that.

I have it set not to check for updates automatically because I don't want to be distracted if I'm trying to work fast, and because I don't want any unmonitored network activity on computers other than my own that I use my flash drive on.

If I want a beta release, it is a bi silly to have to enable automatic updates, check, then disable them again. Perhaps it's an unusual thing to want, but "Why not?" is really the question.

pbatard, 2015-12-23:
Sorry but betas come on top of the check for release. Think of it as the threshold level used for Windows 10 updates. Someone subscribing to threshold 2 will automatically get threshold 1 as well, as it is a deeper level of experimental releases. Same goes for beta and release for Rufus.

cookieboyeli, 2015-12-23:
Yes, a release newer than a beta would automatically show up as it should instead. That's not exactly what I'm asking about though.

The problem is when a beta is newer than an official release. Scheduled automatic checks detect new betas, but manual check can not without working around it.

As it stands you can ONLY manually check for a beta if automatic update checking is enabled - which is silly to have to do if you have it disabled all the time.

pbatard, 2015-12-23:
Sorry but that's not gonna change because [the feature you request would only benefit a very small minority and therefore just isn't worth spending time on](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want_I_mean_youre_a_developer__it_shouldnt_be_that_hard).

If you are interested in updates, you really shouldn't disable the automated check, especially as its frequency can be parametrized and Rufus will not send any request to the update server more than once a month if you do set it it to one month (and the data being sent is super small). If you do disable the update check, then getting notified about updates, be them betas or releases, is your responsibility. The check for update button might or might not help you there - it was added for convenience, but I make no guarantee whatsoever about its ability to detect the type of release you are interested in.