pbatard/rufus #657: The request could not be performed because of an I/O device error.

Zuhayer, 2015-12-25:
I cannot format my usb or delete any files from it. Seems like it has gone permanently read only as i can read files on it. I also used DISKPART in cmd got this error:

>"DiskPart has encountered an error: The request could not be performed because of
>an I/O device error. 
>See the System Event Log for more information."

Below is the log from Rufus, any help will be appreciated:

>Format operation started
>Requesting disk access...
>Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
>I/O boundary checks disabled
>Analyzing existing boot records...
>Drive has an unknown Master Boot Record
>Volume has an unknown Partition Boot Record
>Deleting partitions...
>Could not delete drive layout: [0x0000045D] The request could not be performed because of an >I/O device error.
>Could not reset partitions
>Re-mounted volume as 'D:' after error
>
>Found USB 2.0 device 'Memorex Pilot B USB Device' (12F7:1926)
>1 device found
>Disk type: Removable, Sector Size: 512 bytes
>Cylinders: 250, TracksPerCylinder: 255, SectorsPerTrack: 63
>Partition type: MBR, NB Partitions: 1
>Disk ID: 0xCA35F07D
>Drive has an unknown Master Boot Record
>Partition 1:
 > Type: FAT32 (0x0b)
 > Size: 1.9 GB (2062532608 bytes)
 > Start Sector: 32, Boot: No, Recognized: Yes
>Found USB 2.0 device 'Memorex Pilot B USB Device' (12F7:1926)
>1 device found
>Disk type: Removable, Sector Size: 512 bytes
>Cylinders: 250, TracksPerCylinder: 255, SectorsPerTrack: 63
>Partition type: MBR, NB Partitions: 1
>Disk ID: 0xCA35F07D
>Drive has an unknown Master Boot Record
>Partition 1:
 > Type: FAT32 (0x0b)
>  Size: 1.9 GB (2062532608 bytes)
>  Start Sector: 32, Boot: No, Recognized: Yes

pbatard, 2015-12-25:
Looks like your drive decided to fail when Rufus was copying files (the same would have happened if you were just using Windows Explorer - unfortunately, flash memory is not everlasting).

There's really not much that can be done to fix hardware failures - see [here](https://github.com/pbatard/rufus/wiki/FAQ#Rufus_cant_detectformat_my_drive_and_reports_that_it_contains_no_media) for more.

closed 2015-12-25 by pbatard

Zuhayer, 2015-12-26:
It was showing Write protect message before i used Rufus. I had a doubt that it might be time to send off the USB with a 21 gun salute then :)

Thanks for the response.