pbatard/rufus #658: UEFI:NTFS partition could not be mounted: [14] Not Found

feidaman, 2015-12-27:
Hi, I'm trying to install W10 from USB and the files are >4GB so I'm trying to use Rufus's UEFI:NTFS mode, using the following settings:

[image]

I click Start and the process completes successfully. Then I mount my W10 installer ISO and drag the files to the NTFS partition of the USB flash drive, using Windows Explorer.

I connect the USB drive to the laptop where I'm trying to install Windows (no optical drive available) and it comes up with the following error:

'*** UEFI:NTFS ***

Loading NTFS Driver... DONE
Locating the  first NTFS partition on this device...
   ERROR: NTFS partition could not be mounted: [14] Not Found

Press any key to exit.'

In the BIOS, UEFI boot is enabled, Secured Boot is disabled. Any ideas what could be causing this?

PS: The laptop is a Clevo P751DM.

pbatard, 2015-12-27:
If you are working with an ISO, you shouldn't have to use UEFI:NTFS explicitly. Just select _Create bootable disk using_ &rarr; _ISO_, and the UEFI:NTFS partition will be created for you if you use GPT.

Can you please try that?

I do suspect that you will get the same error however, as it looks like your UEFI firmware is not implementing `ConnectController()` properly. 

When your error occurs, we know that we have successfully loaded the NTFS EFI driver as well as identified the NTFS partition on the USB drive (else we would have gotten errors earlier), so the call to the UEFI Boot Service [ConnectController()](https://github.com/pbatard/uefi-ntfs/blob/master/boot.c#L318-L319), which indicates that the NTFS partition should be serviced by the NTFS driver, should succeed. The fact that it doesn't seems to indicate that your UEFI firmware implementation is different from other UEFI firmwares. Without having direct access to your machine, it will be next to impossible to figure out what is going on. And even then, the most likely explanation I see is that the maker of the Clevo firmware decided to deviate or not implement some of the UEFI specs.

If I ever manage to get my hands on a machine that exhibits such an issue, I may try to look into it, but until then, I don' think there is much I can do...

feidaman, 2015-12-28:
Thank you for getting back so quickly. I had already tried selecting "ISO" from the list and, as you guessed, it didn't work.

In the end the only solution that worked was to use an ISO whose size was < 4GB and just use FAT32. A firmware update from Clevo may solve this, but at the moment there are non available as this is a new model. I've just sent them an email about this and pointed them to this issue. Touch wood!

pbatard, 2016-01-29:
I think that this may be linked to the same issue that was referenced [here](https://github.com/pbatard/uefi-ntfs/issues/3) and which will be addressed in the next version of Rufus. However that would mean that the Clevo P751DM contains an NTFS EFI driver in its firmware, which is possible, but which should have provided 2UEFI boot options for the same drive, one of which should have worked.

At any rate, the UEFI:NTFS driver used by Rufus will contain a handful of improvements in the next release, and since there is not much I can do, short of getting a machine that reproduces the same problem, I will close this issue for now.

closed 2016-01-29 by pbatard

lcdss, 2016-03-05:
Same here on [**Asus X44C**](https://www.asus.com/support/Download/3/360/0/5/lQoqDQGy6QyxA9Iy/29/) (BIOS 208).

pbatard, 2016-03-06:
@lcdss,

Can you please confirm that what you are seeing __EXACTLY__ as the full output from UEFI:NTFS?

Some of the output elements were changed between 2.6 and 2.7, so I would really like to have your actual output, which may look something like this because, at the very least, there should be a difference in the title:
```
*** UEFI:NTFS ***

Loading NTFS Driver... DONE
Locating the  first NTFS partition on this device...
   ERROR: NTFS partition could not be mounted: [14] Not Found

Press any key to exit.
```
The more of an exact copy from the output you get you can provide, the better.

lcdss, 2016-03-06:
The **EXACTLY** output you paste above.

I used Rufus 2.7 (855) to create a bootable USB and install Windows 10 on my laptop (HP envy 15t).

Partition Scheme: GPT for UEFI
File System: NTFS

On my laptop works perpectly but on my friend's notebook i get this error.

pbatard, 2016-03-06:
> The EXACTLY output you paste above.

That cannot be correct, because if you are using Rufus 2.7, then at the very least you will not see `*** UEFI:NTFS ***` but `*** UEFI:NTFS (64-bit) ***` or `*** UEFI:NTFS (32-bit) ***`. Which means that, if `*** UEFI:NTFS ***` is what you see, then you are not using Rufus 2.7, and if you are using Rufus 2.7, then I'd like to know if you see 32-bit or 64-bit on your friend's notebook, as well as any other differences you might see in the output.

I explained that, at the very least, there should be a difference in the title, so could you __please__ make sure that you get the exact output from your friend's computer?

lcdss, 2016-03-07:
Sorry, the exact output was:

```
*** UEFI:NTFS (64-bit) ***

Loading NTFS Driver... DONE
Locating the  first NTFS partition on this device...
   ERROR: NTFS partition could not be mounted: [14] Not Found

Press any key to exit.
```

pbatard, 2016-03-07:
Nice try, but no... :unamused: 

Please come back when you have the actual output you saw, because the one you copy/pasted is simply __IMPOSSIBLE__ to get, with any version of Rufus.

If you really can't be bothered to provide the actual output from your friend's computer, don't expect people to bother helping you out.

RobertSpir, 2016-04-24:
hi, I've had this problem on both my computers (older asus g73sw laptop and newer desktop with msi Z87-GD65 mainboard) with rufus 2.6
after upgrading to version 2.8, desktop with msi mainboard boots without problems but on asus g73sw I have this error:
`*** UEFI:NTFS (64-bit) ***

Loading NTFS driver... DONE
Locating the first NTFS partition on this device... DONE
Check if partition needs the NTFS driver service... DONE
Starting NTFS driver service...
  ERROR: Could not start NTFS service: [14] Not Found

Press any key to exit.`

pbatard, 2016-04-24:
There will be multiple improvements to UEFI:NTFS in v2.9 of Rufus, including ones that should address these kind of issues (or at the very least give a better idea of what the underlying problem actually is).

Can you please try with [this test version](https://rufus.akeo.ie/testing/rufus-2.9_TEST1.exe), which contains the UEFI:NTFS improvements and report back?

RobertSpir, 2016-04-24:
it works, thanks!

pbatard, 2016-04-24:
I'm very pleased to hear that! Thanks for testing and reporting back. :smile:

brunoel, 2017-10-19:
hello i am a new, As I saw that on this site people have the same problem as I thus I joined.Help me please this is my problem:
UEFI: NTFS(x64)
[INFO]boot disk: PciRoot(0)/Pci(0x1D,0x0)usb(0x0,0x0).
[INFO]Disconnecting possible blocking drivers.
[INFO]Starting NTFS driver.
[INFO]Started driver: efifs NTFS driver V1.0 (GRUB 2.0)
[INFO]Locating the first NTFS partition on the boot device.
[INFO]Checking if partition needs the NTFS service.
[INFO]Starting NTFS partition service
[INFO]looking for NTFS EFi loader
[FAIL]Could not locate\efi\boot\bootx64efi',[14]not found.

pbatard, 2017-10-19:
@brunoel 

Do you actually see an `\efi\boot\bootx64.efi` __on the NTFS partition__?

Be mindful that if you are trying to boot a 32-bit Windows image with 64-bit UEFI firmware (which is what you have, as indicated by `UEFI:NTFS (x64)`, then Windows will boot to fail, because you'll only get an `\efi\boot\bootia32.efi`.

You __MUST__ make sure that you use a 64-bit image, and you also __MUST__ check that this image provides an `\efi\boot\bootx64.efi`.

brunoel, 2017-10-20:
Thank you for answering.
Oh ok.
But we can not overwrite this boot / bootx32 and replace bootx64.

pbatard, 2017-10-20:
> But we can not overwrite this boot / bootx32 and replace bootx64.

It doesn't work like that I'm afraid. Each `boot###.efi' is specifically written to boot a specific ISO, so you can't take a `bootx64.efi` from a 64-bit ISO, drop it onto a 32-bit ISO, and expect it to work. If you don't have a `bootx64.efi` that was designed specifically for the ISO you are booting, you will not be able to boot it in UEFI mode.

brunoel, 2017-10-20:
Okay, I want to try anyway to see.