pbatard/rufus #659: Crashbug in GetInstallWimVersion from src\iso.c

DaPa, 2015-12-27:
It's on line 1027, where the check should be for the null pointer is done. It should be:
	for (p = wim_path; *p != 0; p++)
	{
		if (*p == '\\') *p = '/';
	}

pbatard, 2015-12-27:
Good catch - this is a nasty bug indeed!

Can I ask you however exactly how you observed the crash? Did you get an actual crash of the Rufus 2.6 release? If so, on which platform? And what was the SHA-1 of the ISO you were using?

My issue is that, while I would expect a crash from this bug, I am not experiencing any be it with the 2.6 executable (compiled with MinGW) or when running in Visual Studio, and I also haven't received any other reports from a crash. I'm basically trying to ascertain whether this warrants an immediate bugfix release or, if recent compilers appear to be smart enough to mitigate the expected crash, whether this can wait.

Any information you can provide as to how exactly one can produce a crash due to this bug would be really helpful.

DaPa, 2015-12-28:
I had my own VS2010 build from 2.5 sources and it crashed with Win10_1511_N_EnglishInternational_x64.iso and the other variants. Then got and build from latest sources and saw the same... 
Could be due VS2010 compiler as I was wondering the same: why nobody reported yet. Maybe 2.5/2.6 was not yet tried with Win10.
Anyway forgot to say: nice application, thanks for making it open.

pbatard, 2015-12-28:
> Could be due VS2010 compiler

Most likely. I'm not seeing it in VS2015, and MinGW seems to also alleviate the issue. I suspect this kind of bug is common enough to have modern compilers take action against it.

> Maybe 2.5/2.6 was not yet tried with Win10.

Rufus gets downloaded close to 2 million times a month, so I suspect million of users have been using it with Windows 10 since the July 29th release. I can assure you it does get tested extensively with Windows 10, especially as I am developing on this platform too.

I have now patched the issue for the next release, and I think I will wait to see if other users report that crash with the latest release, to decide if I need to produce an early bugfix release. At any rate, thank you very much for reporting this potentially very nasty bug!

commit 961ca93a (2015-12-28): [iso] fix a possible crash when scanning the install.wim version

* Closes #659
* Issue and fix reported by DaPa
closed 2015-12-28 by pbatard