pbatard/rufus #662: Undetermined error while formatting

mike02x, 2016-01-01:
I´ve been trying to use Rugus to convert an ISO of Windows x64 to my USB but I kept getting the same error ¨Undetermined error while formatting¨ however my current OS is Windows 10 x32 could that affect the way it operates?

pbatard, 2016-01-02:
[Most likely, this is an issue with your USB drive](https://github.com/pbatard/rufus/issues/632#issuecomment-168302420). Have you tried a different USB?

mike02x, 2016-01-02:
What's strange is that that USB works fine, as I can copy and paste anything files and they don't get corrupted, after many tries I just ended manually doing it... I disabled my AV and ran as Admin as it's an home PC not a workstation and gave the same error, so what I did at last was to extract the ISO into an USB.

acer475, 2016-01-02:
I tried 2 USB with 8GB and 16GB those are different brand but still cannot fix this issue, after no any idea I trying to use camera micro-SD card with card reader... It's work like charm!!! (use Rufus to burn an image)

lot of error I think those come from USB drive maybe you can try your SD card instead :)

acer475, 2016-01-02:
One more thing! Don't forgot to check whether your chosen drive to installed Windows is "MBR" or "GPT" before choose burn option in Rufus or any application used to create USB bootable drive, otherwise your PC or laptop cannot run again

>
>
>
if you choose as wrong type between MBR or GPT partition it's doesn't matter don't worry, you can use another PC or laptop to make a USB/DVD bootable instead.

pbatard, 2016-01-04:
Another thing you can also try is the <kbd>Alt</kbd>-<kbd>Z</kbd> cheat mode to completely zero the device before attempting to format it in Rufus, to see if that will help. Be mindful however that completely filling the device with zero may take a lot of time (hours) if your device is large or if you are using a an USB 2.0 connection.

Alternatively, because the error above really comes from the native Windows formatting API, which is what Rufus uses, and not actually from the Rufus code, it would also be interesting to see what happens when you try to format your drive in Windows. Basically I expect the same error to happen, because the way Windows formats a drive and the way Rufus does it should be exactly the same.

Finally, a __FULL__ log from Rufus is __ALWAYS__ exceedingly helpful to find out if there is any problem that is being reported prior to the format operation, and that could tell us why format is failing.

pbatard, 2016-01-04:
Also, since we have #632 opened, with the exact same title, I'm going to close this duplicate.

closed 2016-01-04 by pbatard

Hassankhan1, 2016-03-23:
Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'H:' as volume mountpoint
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume does not have an x86 Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'Sony Storage Media USB Device' (054C:05BA)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1891, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0014C0DD
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 14.5 GB (15554576384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB 2.0 device 'Sony Storage Media USB Device' (054C:05BA)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1891, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0014C0DD
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 14.5 GB (15554576384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Closed Windows format prompt

pbatard, 2016-03-23:
@Hassankhan1, what happens if you try to format your device with the native Windows format tool (right clik &rarr; Format drive). Also, __PLEASE__ don't truncate the log &mdash; it is missing very important information.