pbatard/rufus #663: Ubuntu16.04~ 15.10~ instll error

oklokl, 2016-01-01:
test@test:~/문서/rufus-2.6$ ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for ar... ar
checking for ranlib... ranlib
checking for a sed that does not truncate output... /bin/sed
checking for rm... /bin/rm
checking for strip... strip
checking for windres... no
checking for inline... inline
enabling Large File Support (ISO support)
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating src/bled/Makefile
config.status: creating src/libcdio/iso9660/Makefile
config.status: creating src/libcdio/udf/Makefile
config.status: creating src/libcdio/driver/Makefile
config.status: creating res/localization/Makefile
config.status: creating src/ms-sys/Makefile
config.status: creating src/syslinux/libfat/Makefile
config.status: creating src/syslinux/libinstaller/Makefile

test@test:~/문서/rufus-2.6$ make
Making all in src
make[1]: Entering directory '/home/test/문서/rufus-2.6/src'
Making all in bled
make[2]: Entering directory '/home/test/문서/rufus-2.6/src/bled'
  CC       libbled_a-bled.o
In file included from bled.c:14:0:
libbb.h:20:2: error: #error Only Windows platforms are supported
 #error Only Windows platforms are supported
  ^
In file included from libbb.h:33:0,
                 from bled.c:14:
../msapi_utf8.h:22:21: fatal error: windows.h: 그런 파일이나 디렉터리가 없습니다
compilation terminated.
Makefile:330: recipe for target 'libbled_a-bled.o' failed
make[2]: *** [libbled_a-bled.o] Error 1
make[2]: Leaving directory '/home/test/문서/rufus-2.6/src/bled'
Makefile:462: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/test/문서/rufus-2.6/src'
Makefile:283: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1
test@test:~/문서/rufus-2.6$

pbatard, 2016-01-02:
```
#error Only Windows platforms are supported
```

Should be pretty obvious. Rufus __is__ a Windows application only. If you try to compile or run it on Linux, you're on your own.

closed 2016-01-02 by pbatard

oklokl, 2016-01-02:
I want to use in Ubuntu.
be about to install
It is difficult..
I do not know at all when it comes to installing the program.

pbatard, 2016-01-02:
> I want to use in Ubuntu.

You __CAN'T__. Rufus runs on Windows only.
Also it does NOT have an installer. What you are trying to do by running `git` and `make` is recompiling a __Windows-only__ program. To do that you would need MinGW. But even if you managed to recompile this __Windows-only__ application on Linux, you still won't be able to run it there.

Again, if you persist on compiling or using Rufus on Linux, you are 100% on your own because it is not supported.

oklokl, 2016-01-02:
존경 하는 개발자님..
감사 합니다 ^^..
전 이 프로그램 명령어가..
리눅스에서 
사용 할수 있게 도와 주는 명령어라 
생각 했습니다..

하지만 아니군요.
^^
전 당신이 천재라고 생각 합니다.
리눅스에서도 설치 가능
하게 기대 하고 있습니다.
^^..

Dear developer's.
Thank you ^^ ..
Before the program instructions.
In Linux
La instructions that can use help
I thought ...

But it's not.
^^
Before I think you're a genius.
Installation on Linux available
I look forward to.
^^ ..

pbatard, 2016-01-02:
> Installation on Linux available
> I look forward to.

This is not going to happen, as per the [FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxOS_XSome_other_OS).

oklokl, 2016-01-02:
Thanks anyway .. ^^