pbatard/rufus #664: Windows 10 UEFI boot problem

AerSnipe, 2016-01-02:
Hello guys i am having a problem booting from my USB drive
my PC specs are:
MB: ASUS Z68 DELUXE
I5 3550 / 8GB ram/128Gb SSD/1.5 TO HDD
so whene i try to boot from the fash drive i get this error (launching ntfs efi loader efi boot bootx64.efi failure 14). and whene i change the rom option to EFI my PC shut down and won't turn on until i clear the CMOS. can any one tell me whats the problem

pbatard, 2016-01-04:
Could you __please__ provide the exact formatted text for the error message you see?
`bootx64.efi failure 14` is not a complete error message from UEFI:NTFS, and because of this it makes it very difficult to figure out how to help you.

Basically, I expect that you might be seeing something like this:

```
*** UEFI:NTFS ***
Loading NTFS Driver... DONE
Locating the first NTFS partition on this device... DONE
Looking for NTFS EFI loader... 
  ???????????????????????????????

Press any key to exit.
```

The __EXACT__ `??????????????????` part is what I am interested in.

AerSnipe, 2016-01-04:
thank you for your rely , i actually solved the problem by my self , you
see the issue was in the disk i was using (i had a windows 10 allinone)
with both 32 and 64 bit and that was the problem so when i changed the disk
to windows 10 64bit only the system booted normally and i finished
installing on UEFI. and everything is perfect now, and again thank you very
much for your time.

pbatard, 2016-01-04:
Ok. I guess I'll close the issue then.

closed 2016-01-04 by pbatard