pbatard/rufus #670: Error selected fonts on locale Win-1251, ru-RU in to About dialog after b818 :)

VictorVG, 2016-01-11:
Error selected fonts on locale Win-1251, ru-RU in to About dialog after b818:

b818 screenshot:

[image]

Fonts dialog are correct and inscriptions are read.

b821 or later - the code page of the font chosen is incorrect and we see just such a funny picture:

x64 build:

[image]

x86 build:

[image]

Tested on Win7 SP1 x64, locale ru-RU, support UNICODE for non-UNICODE language is ON.

pbatard, 2016-01-11:
Hi Victor - thanks for the report!

I'm seeing the same issue (as well as for languages that have Cyrillic influences), and I know precisely where the issue was introduced (in 0f56c2d1ef90e754e17610110afeab690f13d767, aka `b821`).

The problem is, I had to switch the default dialog font from `Segoe UI` to `Segoe UI Symbol` as I found that Thai would have the exact same issue with the About dialog as the one you report now.

I guess it means I'm going to set the Segoe font to use according to the language... which is exactly what I was trying to avoid in the first place. :disappointed: 

Why oh why, even in Windows 10, does Microsoft have to make things so hard, and can't provide people with a __single__ Segoe UI font that actually works for all Unicode codepoints (even with RTF fields)?!?

Oh well, guess I'll have to figure out a way to fix this, __again__, for 2.7...

VictorVG, 2016-01-11:
Basically, I came across a similar thing when translating programs and solve problems forced prostanovkoy code page of the translated resource. Maybe there to do the same? I see that the flaw in Microsoft always gets the most is not the right moment. :) Swear to "advisers" offering "solutions" of the form:

HKLM \ Software \ CurrentContolSet \ ... \ NLS

"CP_1250" = "cp_1251.nls"
"CP_1251" = "cp_1251.nls"
"CP_1252" = "cp_1251.nls"
...
"CP_1256" = "cp_1251.nls"
...
"CP _..." = "cp_1251.nls"

that Fixed a properly localized their programs (usually in the code page of the resource they have 866 or less neutral) they spoil locale entire OS in general, I'm still twenty years ago stopped because they say it is useless - no one and listen do not want so that in such cases if I really hinders itself the same Resource Hacker correcting codes resources page - quickly made, kept, I forgot. And quiet - do not have to talk to the stubborn. :)

pbatard, 2016-01-11:
I'm using Unicode all the way through - In 2016, you DON'T want to use codepages or anything that is not 100% Unicode. 

The issue here is that Microsoft probably screwed up some character mappings when displaying Unicode on RTF fields so that the `Segoe UI Symbol` font will properply display Unicode Thai characters, but failt to display Cyrillic, and the `Segoe UI` font will properly display Cyrillic but not Thai. So this is really an issue between Microsoft's RTF engine and font mapping (as well as Microsoft deciding to multiply `Segoe UI *` fonts, when there should only be __ONE__ `Segoe UI`, with all Unicode characters needed to display all languages.

Working arount this shouldn't be too hard, but it does annoy me that it still seems impossible for companies like Microsoft to get Unicode right...

closed 2016-01-11 by pbatard

VictorVG, 2016-01-11:
It's cold and I want to bask in it:

[image]

and because the winter mood. :)

WoodCrop, 2016-04-01:
J-U-S-T when we all thought that " Segoe UI  font will properly display Cyrillic " -- NO-P, it sure DOESN'T for ME-E-E-E.  (For example, in Magento "Custom Menu / Main (top) Menu" categories and sub-categories, it WRITES "in English," but categorically REFUSES to write in ANYTHING Cyrillic, AT ALL.)  Solutions, any one?  (...to a point where I'd seriously consider becoming a "permanent client" of the potential successful SOLVER that SOLVES THIS, huh??...

VictorVG, 2016-04-01:
Use a suitable UNICODE font having the desired code page. For example Lucida Console family Sans Serif. With him such effects are not observed, and the terminal is working fine with him.

WoodCrop, 2016-04-01:
Ogromnboe SPASIBO, Viktor!  Ja i po-Russki govorju, a sajt noj -- smallcrop(dot)NET - uzhasno seriozno nuzhno lokalizirovat' -- BEZ kakogo-libo genial'nogo Mikrosofta, v slutchae...  HOTITE stat' moim sovetnikom na dolgosrotchnoe budushtchee??... (Mozhno otvetit' prjamo cherez  moj sajt, esli tak...)

pbatard, 2016-04-01:
@WoodCrop, is that an issue you see in Rufus? Or is this a more general question about displaying Cyrillic.

Please only use the Rufus issue tracker to report issues you actually see in Rufus, else it becomes very confusing for me to try to understand if there's something I need to address in Rufus, and there are user forums dedicated to general programming questions and other stuff.

VictorVG, 2016-04-01:
Write me use e-mail.

WoodCrop, 2016-04-02:
Good morning, Victor!  (Huge THANKS for your brief e-mail, here, and YES
-- it would make me EXTREMELY happy to start periodically corresponding
with you -- for example, from THIS e-mail box of mine, as it may be a
little MORE "cyrillic-type ENABLED" than some of the OTHER ones I have
come across.)
 
The SHORT of it is that, if you kindly allow me, I will gradually
"explain myself" regarding this MULTILINGUAL-expertise applied NEED
that I currently happen to have -- as I am localizing about 3-4
specific, PROTOTYPE-like Magento templates into languages like
BULGARIAN (which I have spoken since childhood), RUSSIAN (which I have
ALSO spoken since childhood), Brazilian Portuguese (which, as you may
have seen, ALSO has lots and lots of Unicode-rendition GUI problems in
MAGENTO), and also -- CANADIAN FRENCH (...with PLENTY of "special
characters" THERE, as well...).
 
I arrived in the USA in the 1980'-s, basically, TO STUDY AND TEACH
(several multilingual things, as I will gradually explain).  I ended up
living in Seattle (State of Washington, USA) -- and am currently still
THERE, but have gradually developed an extremely STRONG desire
"vernut'sja k svoim ves'ma glubokim christijanskim kornjam" -- at least
as far as personal and professional ETHICS are concerned -- of which I
have NEVER EXPERIENCED ANY AT... (incredibly INCOMPETENT) "Mikro-squish"
(...and YES -- I DID spend quite a long period of time THERE, by the
way... Don't REALLY know HOW I "survived" said "ethically MALIGNANT"
professional event, actually...
 
Toward the end of it, they startedemphatically calling me "too OVER-
QUALIFIED to HELP (them)" -- which, I thought, wa an absolutely G-E-N-I-A-
L way of putting it.
 
And so, Victor -- if you'd like, you can give me the spelling of a
"literal" (specific) EXTERNAL mail box of yours for similar purposes,
and we can potentially start CORRESPONDING -- quite valuably and
constructively, about good and helpful things -- very much INCLUDING
applied multilingual-localization, web-based mutual efforts (to be
always properly PAID (in ADVANCE) by me, if they involve PROFESSIONAL
EFFORT on your part, of course.  One excellent way of periodic
payment for project estimates is, for example, "PayPal" -- which I
happen to have.
 
I'll go and milk 2-3 domestic milk-goats now!  (QUITE a
"farming/gardening/ranching" person -- me -- by the way...)
 
 
Greetings, Cheers and Thanks, in Advance! --
 
(Olga-Elena Pjeter von Innus)
-- Olja, iz Seattla, Shtata Vashington (Amerika)

 

Links:

  1. https://github.com/pbatard/rufus/issues/670#issuecomment-204540759

commit a952221a (2016-01-11): [loc] fix Cyrillic not displaying properly in About dialog

* Issue was introduced in 0f56c2d1ef90e754e17610110afeab690f13d767
* Closes #670