pbatard/rufus #671: Rufus is detecting a required version of Syslinux that does not exist.

corruptnova, 2016-01-11:
I went to build a live usb of NixOS, however, Rufus needed Syslinux 6.04 which is a version of Syslinux that does not exist, to my knowledge at least.
[image]

Log from Rufus:

Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'NIXOS_ISO'
  Size: 1028653056 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (6.04)
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: nixos-graphical-16.03pre73316.93d8671-x86_64-linux.iso

Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/ldlinux.sys
Unable to access file: 404
Could not download the file - cancelling

pbatard, 2016-01-11:
You will need to contact the NixOS developers, who did something they shouldn't do, and created their own 6.04 version of Syslinux. Rufus only reports what it sees, and if it says an ISO uses Syslinux 6.04, it's because the developers of said ISO set their version of Syslinux to exactly that. Here is the hexdump of the `isolinux.bin` file above:

[image]

pbatard, 2016-01-11:
Okay, I suspect the NixOS people recompiled their own version from the development tree (which is a bad idea - if you release something, you should stick to officially released component, as going for unreleased code creates exactly this kind of issues!). The current Syslinux dev tree [uses 6.04 as the version](https://github.com/geneC/syslinux/blob/master/version).

I'll see if I can recompile my own version of 6.04 then, but I don't like this, because it could introduce all kind of incompatibilities, and I will have to replace it once 6.04 is officially released...

Again, unless the __original__ project has released __official__ binaries for a specific version, you should not use that version for anything but stuff you're not going to publicly release.

pbatard, 2016-01-11:
Okay, I recompiled my own 6.04 version from latest, and made it available on the Rufus website for download. I tested it against `nixos-graphical-16.03pre73316.93d8671-x86_64-linux.iso` and it seemed to install and boot okay.

However, since this is a purely __unofficial__ version of Syslinux, I make no guarantee that it will work with this or any Syslinux 6.04 based images.

I will close this issue now.

closed 2016-01-11 by pbatard

corruptnova, 2016-01-12:
Oh wow, regardless if this works or not... thank you! I will update this post with whether or not this is successful.