pbatard/rufus #672: could not open '\\?\usbstor#disk&ven_pny&prod_usb_3.0_fd&rev_1.00#596031457&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.

200294600, 2016-01-13:
Hello, I'm trying to Make my computer use tails so I can access certain parts of the internet. My computer simply doesn't allow it, so I downloaded the program and this is the error it gave me, what do I do?

pbatard, 2016-01-13:
`Access is denied` means you have __another__ application who has taken control of the drive.

Unfortunately, Windows does not offer an API that can tell which one it is, so you will have to find out what it might be on your own. It could be an antivirus solution or something else, such as one of the apps listed [at the end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with) (this list is NOT exhaustive).

Also, it always helps to post the __FULL__ log, and not just one line.

pbatard, 2016-01-29:
No update from OP &rarr; Closing.

closed 2016-01-29 by pbatard