pbatard/rufus #673: Rufus 2.6 Error "Error: Installation failure"

dan13698, 2016-01-13:
I'm trying to get a bootable ISO of "Refelx OS for PC" onto my Seagate 2tb HDD but i keep getting this error and no google search has helped me find anything about it. Here is the log:

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Closed Windows format prompt
Formatting (Large FAT32)...
Opened drive \\?\Volume{26a61f82-0000-0000-0000-100000000000} for write access
Size : 1.8 TB 3907027119 sectors
Cluster size 32768 bytes, 512 Bytes Per Sector
Volume ID is 1ee0:3352
32 Reserved Sectors, 476816 Sectors per FAT, 2 FATs
61032397 Total clusters
61032396 Free Clusters
Clearing out 953728 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 476848
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{26a61f82-0000-0000-0000-100000000000}\
\\?\Volume{26a61f82-0000-0000-0000-100000000000}\ was already mounted as F:\
Installing Syslinux 4.04...
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{26a61f82-0000-0000-0000-100000000000} for write access
Could not write Syslinux boot record: [0x00000000] The operation completed successfully.
\\?\Volume{26a61f82-0000-0000-0000-100000000000}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found USB 3.0 device 'Seagate Expansion USB Device' (0BC2:231A)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243201, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x26A61F82
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000397884928 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2016-01-13:
<Sigh>...

__PLEASE__ Don't truncate the Rufus log! I don't know the ISO you used or what OS you are running, which makes it SUPER DIFFICULT to test and help you out... 

Can you __PLEASE__ copy/paste the complete log?

dan13698, 2016-01-13:
That is the complete log, I'm running Windows 10 Pro 64 bit and the ISO i'm trying to use is "Remix_OS_for_PC_64_B2016011201_Alpha"

Also i'm not sure what you mean by truncate, sorry.

dan13698, 2016-01-13:
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0809
Found USB 3.0 device 'Seagate Expansion USB Device' (0BC2:231A)
Device eliminated because it was detected as an USB Hard Drive (score 33 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found USB 3.0 device 'Seagate Expansion USB Device' (0BC2:231A)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243201, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x26A61F82
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000397884928 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.04/0x4e86cf20 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Android-x86 LiveCD'
  Size: 2258632704 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: No
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (4.04)
    With an old menu.c32: No
    With an old vesamenu.c32: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Remix_OS_for_PC_64_B2016011201_Alpha.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Closed Windows format prompt
Formatting (Large FAT32)...
Opened drive \\?\Volume{26a908bf-0000-0000-0000-100000000000} for write access
Size : 1.8 TB 3907027119 sectors
Cluster size 32768 bytes, 512 Bytes Per Sector
Volume ID is 1ee4:255
32 Reserved Sectors, 476816 Sectors per FAT, 2 FATs
61032397 Total clusters
61032396 Free Clusters
Clearing out 953728 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 476848
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{26a908bf-0000-0000-0000-100000000000}\
\\?\Volume{26a908bf-0000-0000-0000-100000000000}\ was already mounted as F:\
Installing Syslinux 4.04...
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{26a908bf-0000-0000-0000-100000000000} for write access
Could not write Syslinux boot record: [0x00000000] The operation completed successfully.
\\?\Volume{26a908bf-0000-0000-0000-100000000000}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found USB 3.0 device 'Seagate Expansion USB Device' (0BC2:231A)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243201, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x26A908BF
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000397884928 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2016-01-13:
Still truncated. The first 2 lines of the log are something like:
```
Rufus version: 2.6.818
Windows version: Windows 10 64-bit (Build 10586)
```

So I still don't know which version of Rufus you use (I'm gonna __assume__ 2.6) or which version of Windows, which can be very important, but okay. I'll download the image and test.

dan13698, 2016-01-13:
Rufus version: 2.6.818
Windows version: Windows 10 64-bit (Build 10240)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0809
Found USB 3.0 device 'Seagate Expansion USB Device' (0BC2:231A)
Device eliminated because it was detected as an USB Hard Drive (score 33 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found USB 3.0 device 'Seagate Expansion USB Device' (0BC2:231A)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243201, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x26A908BF
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000397884928 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2016-01-13:
I tested the image creation and saw no issue on my machine. For good measure, I used the closest type of enclosure i could find compared to the one you have (but with a 1 TB HDD instead of a 2 TB - I am positive the size of the HDD has nothing to do with the issue, as in both cases Large FAT32 will be used).

Here is my log output:
```
Rufus version: 2.6.818
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0809
Found USB 3.0 device 'Seagate Expansion Desk USB Device' (0BC2:3312)
Device eliminated because it was detected as an USB Hard Drive (score 13 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found USB 3.0 device 'Seagate Expansion Desk USB Device' (0BC2:3312)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 121601, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x005CD6EF
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 931.5 GB (1000203837440 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.04/0x4e86cf20 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Android-x86 LiveCD'
  Size: 2258632704 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: No
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (4.04)
    With an old menu.c32: No
    With an old vesamenu.c32: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Remix_OS_for_PC_64_B2016011201_Alpha.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE7 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{00440cb7-0000-0000-0000-100000000000} for write access
Size : 931.5 GB 1953523119 sectors
Cluster size 32768 bytes, 512 Bytes Per Sector
Volume ID is 1809:1a35
32 Reserved Sectors, 238409 Sectors per FAT, 2 FATs
30516347 Total clusters
30516346 Free Clusters
Clearing out 476914 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 238441
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{00440cb7-0000-0000-0000-100000000000}\
\\?\Volume{00440cb7-0000-0000-0000-100000000000}\ was already mounted as F:\
Installing Syslinux 4.04...
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{00440cb7-0000-0000-0000-100000000000} for write access
Successfully wrote Syslinux boot record
Successfully remounted Volume{00440cb7-0000-0000-0000-100000000000}\ on F:\
Copying ISO files...
Extracting files...
libcdio: Found Extension: Joliet Level 3
Image is an ISO9660 image
This image will be extracted using Joliet extensions (if present)
Extracting: F:\TRANS.TBL (1.5 KB)
Extracting: F:\efi\TRANS.TBL (216 bytes)
Extracting: F:\efi\boot\TRANS.TBL (220 bytes)
Extracting: F:\efi\boot\grub.cfg (493 bytes)
Extracting: F:\initrd.img (1.3 MB)
Extracting: F:\install.img (2.8 MB)
Extracting: F:\isolinux\TRANS.TBL (1.1 KB)
Extracting: F:\isolinux\android-x86.png (31.7 KB)
Extracting: F:\isolinux\boot.cat (2 KB)
Extracting: F:\isolinux\isolinux.bin (24 KB)
Extracting: F:\isolinux\isolinux.cfg (877 bytes)
Extracting: F:\isolinux\vesamenu.c32 (155.3 KB)
Extracting: F:\kernel (4.7 MB)
Extracting: F:\ramdisk.img (1.1 MB)
Extracting: F:\system.img (2.1 GB)
Created: F:\syslinux.cfg
Finalizing, please wait...
Created: F:autorun.inf
Created: F:autorun.ico

Found USB 3.0 device 'Seagate Expansion Desk USB Device' (0BC2:3312)
Using autorun.inf label for drive F: 'Android-x86 LiveCD'
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 121601, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00440CB7
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 931.5 GB (1000203836928 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
```

I don't like that `Could not write Syslinux boot record: [0x00000000] The operation completed successfully.` line though, so I'd like to find a bit more about it. Can I please ask you to try again with `rufus-2.7_TEST2.exe`, which you can download [here](https://rufus.akeo.ie/testing/rufus-2.7_TEST2.exe) and copy/paste the log you get? It should hopefully provide a bit more information about that Syslinux boot record error.

dan13698, 2016-01-14:
Rufus version: 2.7.828 (Test2)
Windows version: Windows 10 64-bit (Build 10240)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0809
Found USB 3.0 device 'Seagate Expansion USB Device' (0BC2:231A)
Device eliminated because it was detected as an USB Hard Drive (score 33 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found USB 3.0 device 'Seagate Expansion USB Device' (0BC2:231A)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243201, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x26A908BF
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000397884928 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.04/0x4e86cf20 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Android-x86 LiveCD'
  Size: 2258632704 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: No
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (4.04)
    With an old menu.c32: No
    With an old vesamenu.c32: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Remix_OS_for_PC_64_B2016011201_Alpha.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{02b2718a-0000-0000-0000-100000000000} for write access
Size : 1.8 TB 3907027119 sectors
Cluster size 32768 bytes, 512 Bytes Per Sector
Volume ID is 13ec:1b3c
32 Reserved Sectors, 476816 Sectors per FAT, 2 FATs
61032397 Total clusters
61032396 Free Clusters
Clearing out 953728 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 476848
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{02b2718a-0000-0000-0000-100000000000}\
\\?\Volume{02b2718a-0000-0000-0000-100000000000}\ was already mounted as F:\
Installing Syslinux 4.04...
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{02b2718a-0000-0000-0000-100000000000} for write access
Partial write of Syslinux boot record: wrote 12057367 bytes but requested 0
\\?\Volume{02b2718a-0000-0000-0000-100000000000}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found USB 3.0 device 'Seagate Expansion USB Device' (0BC2:231A)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243201, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02B2718A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000397884928 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2016-01-14:
Thanks a lot for running the test version. Unfortunately, from this message:
```
Partial write of Syslinux boot record: wrote 12057367 bytes but requested 0
```
I realize two things:
1. That I screwed up the output for that message, and it should actually have displayed `Partial write of Syslinux boot record: wrote 0 bytes but requested 512`. I'll fix that.
2. As I suspected, the call to `WriteFile()`, that is issued to write the Syslinux boot record, reports that the operation completed successfully, but it also indicates that zero bytes were written (instead of the 512 bytes expected).

I guess I could add a retry for the write operation, when it says that zero bytes were written, but I'm not sure this will actually fix anything, as I am wondering if this could have anything to do with the HD possibly using a 4096-bytes sector size behind the scenes (even though the enclosure reports `Sector Size: 512 bytes`, but I know Seagate enclosures can be weird). Just a question - do you have a different disk or flash drive you could test with, just to see if you get the same issue?

I think I need to think about this a little bit more before I produce another version to test, that can attempt to fix your issue. I hope you can wait another day.

pbatard, 2016-01-14:
Okay, I now have a TEST4 that adds retry on write failure. I'm not sure if this will fix your issue, but I'd like you to test and paste the log. This version can be downloaded [here](https://rufus.akeo.ie/testing/rufus-2.7_TEST4.exe). Thanks!

dan13698, 2016-01-15:
Rufus version: 2.7.831 (Test4)
Windows version: Windows 10 64-bit (Build 10240)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0809
Found USB 3.0 device 'Seagate Expansion USB Device' (0BC2:231A)
Device eliminated because it was detected as an USB Hard Drive (score 33 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found USB 3.0 device 'Seagate Expansion USB Device' (0BC2:231A)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243201, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02B2718A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000397884928 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.04/0x4e86cf20 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Android-x86 LiveCD'
  Size: 2258632704 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: No
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (4.04)
    With an old menu.c32: No
    With an old vesamenu.c32: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Remix_OS_for_PC_64_B2016011201_Alpha.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{091e00c2-0000-0000-0000-100000000000} for write access
Size : 1.8 TB 3907027119 sectors
Cluster size 32768 bytes, 512 Bytes Per Sector
Volume ID is 19e7:2263
32 Reserved Sectors, 476816 Sectors per FAT, 2 FATs
61032397 Total clusters
61032396 Free Clusters
Clearing out 953728 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 476848
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{091e00c2-0000-0000-0000-100000000000}\
\\?\Volume{091e00c2-0000-0000-0000-100000000000}\ was already mounted as F:\
Installing Syslinux 4.04...
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{091e00c2-0000-0000-0000-100000000000} for write access
  Wrote 0 bytes but requested 512 - retrying...
  Wrote 0 bytes but requested 512 - retrying...
Could not write Syslinux boot record: [0xC00305B6] Incorrect size argument.
\\?\Volume{091e00c2-0000-0000-0000-100000000000}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found USB 3.0 device 'Seagate Expansion USB Device' (0BC2:231A)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243201, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x091E00C2
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000397884928 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2016-01-15:
Thanks. As I suspected, issuing a write retry didn't work (the `Incorrect size argument` is just the error I generate on size issue), and I'm really not sure what the underlying issue is at this stage. Considering that you are the first person to report this, I would think this is an issue with your hardware or software environment or the specific USB &rarr; SATA controller that is used in that Seagate HDD enclosure you have.

I would still be very interested to hear what happens when you try with a flash drive or a different enclosure, or with the same enclosure, but on a different PC. Can you please do that?

dan13698, 2016-01-18:
hi, sorry for the late reply. i tried with a different flash drive and it did work that time but since it was only USB 2.0 i couldn't get the OS running (since it requires USB 3.0 with 20mb/s write speed). I'll try the seagate hdd on another pc now to see if that does me any good.

pbatard, 2016-02-05:
Did you get a chance to try the same enclosure on a different PC? If so, what was the outcome?

dan13698, 2016-02-06:
I tried the seagate hdd on a laptop and the outcome was the same

pbatard, 2016-02-08:
OK, so that seems to indicate that this is an issue with this specific Seagate enclosure. Unfortunately it looks like the enclosure firmware might be preventing some operations (maybe Seagate, or whoever developed this specific USB &harr; SATA controller tried to add security features, to prevent writing outside of the active partition space, which is what we need to do for Syslinux installation), but without getting my hands on such an enclosure, this will be hard to investigate. And if it comes from the firmware, even if the cause can be determined, it may be next to impossible to fix, or at least fix with an reasonable amount of effort, when factoring the expected number of people affected by it. 

As I mentioned, I have a few official Seagate enclosures (some of which I got when buying external 2TB Seagate HDDs), using a handful of different USB &harr; SATA controllers, and I have not seen this behaviour. So my expectation right now is that this probably affects a very limited number of people.

matthewschallenkamp, 2016-05-23:
I encountered this error today while trying to install [manjaro linux](manjaro.github.io), specifically xfce 16.06 32 bit on a PNY brand micro sd card using the enclosed adapter. I then attempted the same install with the same iso on a different computer and it worked. 

Log from the computer that failed
```
Rufus version: 2.9.934
Windows version: Windows 10 64-bit (Build 10240)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found card reader device 'Generic SDU8G SD Card'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0FD7ABA5
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8009023488 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'MJRO1606'
  Size: 1416314880 bytes
  Uses: Syslinux/Isolinux v6.03
Using image: manjaro-xfce-16.06-rc1-i686.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{e5c9464e-1f2f-11e6-9c5b-e47fb2f72334}\
Installing Syslinux 6.03...
Using existing './rufus_files/syslinux-6.03/ldlinux.sys'
Using existing './rufus_files/syslinux-6.03/ldlinux.bss'
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{e5c9464e-1f2f-11e6-9c5b-e47fb2f72334} for write access
Cannot read sector 0
Syslinux FAT access error
\\?\Volume{e5c9464e-1f2f-11e6-9c5b-e47fb2f72334}\ was already mounted as D:\
Re-mounted volume as 'D:' after error

Found card reader device 'Generic SDU8G SD Card'
[... 215 lines cut ...]
Extracting: E:\manjaro\i686\mhwd-image.sqfs (114 MB)
Extracting: E:\manjaro\i686\root-image.sqfs (314.1 MB)
Extracting: E:\manjaro\i686\xfce-image.sqfs (845.4 MB)
Extracting: E:\manjaro\isomounts (270 bytes)
Created: E:\syslinux.cfg
Finalizing, please wait...
Created: E:autorun.inf
Created: E:autorun.ico

Found USB device 'SDHC Card' (????:????)
Using autorun.inf label for drive E: 'MJRO1606'
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:C75C)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0703D68E
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8009023488 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
```

I am happy to provide whatever other information is needed to help fix the problem. Looking at the logs, the ones that failed appear to all be windows 10 build 10240, where the ones succeeding are 10586, which may actually be the problem. Dan13698, I suggest trying updating windows and attempting the install again.

pbatard, 2016-06-09:
@matthewschallenkamp, I think I may now have a fix for this.
Can you please try [rufus-2.10_TEST11.exe](https://rufus.akeo.ie/testing/rufus-2.10_TEST11.exe) and let me know if that works better?

matthewschallenkamp, 2016-06-10:
That worked perfectly! Thank you very much.
Log:
```
Rufus version: 2.10.957 (Test11)
Windows version: Windows 10 64-bit (Build 10240)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found card reader device 'Generic SDU8G SD Card'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 7
Disk ID: 0x0703D68E
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 122 MB (127951872 bytes)
  Start Sector: 2048, Boot: No, Recognized: Yes
Partition 2:
  Type: Extended (0x05)
  Size: 7.3 GB (7881071616 bytes)
  Start Sector: 251954, Boot: No, Recognized: No
Partition 5:
  Type: GNU/Linux (0x83)
  Size: 32.0 MB (33553408 bytes)
  Start Sector: 1998, Boot: No, Recognized: Yes
Partition 6:
  Type: Extended (0x05)
  Size: 7.3 GB (7846495232 bytes)
  Start Sector: 67532, Boot: No, Recognized: No
Partition 9:
  Type: FAT32 LBA (0x0c)
  Size: 63 MB (66060288 bytes)
  Start Sector: 2, Boot: No, Recognized: Yes
Partition 10:
  Type: Extended (0x05)
  Size: 7.2 GB (7780433920 bytes)
  Start Sector: 196558, Boot: No, Recognized: No
Partition 13:
  Type: GNU/Linux (0x83)
  Size: 7.2 GB (7779385344 bytes)
  Start Sector: 2048, Boot: No, Recognized: Yes
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking beta channel...
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'MJRO1606'
  Size: 1416314880 bytes
  Uses: Syslinux/Isolinux v6.03
Using image: manjaro-xfce-16.06-rc1-i686.iso
No new beta version found.
Checking test channel...
No new test version found.
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
[... 119 lines cut ...]
Extracting: D:\manjaro\boot\intel_ucode.LICENSE (6.8 KB)
Extracting: D:\manjaro\boot\intel_ucode.img (676.5 KB)
Extracting: D:\manjaro\i686\live-image.sqfs (45.3 MB)
Extracting: D:\manjaro\i686\mhwd-image.sqfs (114 MB)
Extracting: D:\manjaro\i686\root-image.sqfs (314.1 MB)
Extracting: D:\manjaro\i686\xfce-image.sqfs (845.4 MB)
Extracting: D:\manjaro\isomounts (270 bytes)
Created: D:\syslinux.cfg
Finalizing, please wait...
Created: D:autorun.inf
Created: D:autorun.ico

Found card reader device 'Generic SDU8G SD Card'
Using autorun.inf label for drive D: 'MJRO1606'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x000A2B16
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8009023488 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
```

pbatard, 2016-06-10:
Great! I'll close this issue then.

closed 2016-06-10 by pbatard