pbatard/rufus #678: FreeDOS 1.1 support

sremick, 2016-01-18:
Could the quick FreeDOS support in Rufus be updated to FreeDOS 1.1? v1.1 came out in 2012 so it'd be nice to be current. I tried using the IMG and ISO options... unfortunately the FreeDOS 1.1 ISO is an installer, not a live boot, so that was no good. And all the IMG files I could find are FAT16 only (not FAT32) and don't use the whole disk.

Rufus is the ONLY USB-making utility I have found that can make a bootable FAT32 FreeDOS device that uses the full capacity. THANK YOU

pbatard, 2016-01-18:
Huh?

As per https://github.com/pbatard/rufus/tree/master/res/freedos (see the readme at the bottom), Rufus uses the very latest FreeDOS binaries currently available. For instance, there's no newer kernel than 2041, which is what Rufus uses.

Now, if you do have links to official binaries from the FreeDOS project that are more recent than what Rufus use, I'd be interested. But as far as I know, I am already using the most recent binaries I can get, as none of these have been updated since 2012.

pbatard, 2016-01-18:
Addon: I'm actually curious to hear what made you think that the FreeDOS version used by Rufus was not the latest. Where exactly did you get the (inaccurate) hint that the FreeDOS used by Rufus might be older than 1.1?

pbatard, 2016-01-20:
No update from OP &rarr; Closing.

closed 2016-01-20 by pbatard