pbatard/rufus #680: ERROR: Could not set path: [14] Not Found

MrX-at-GitHub, 2016-01-20:
There's something wrong with Windows 7.

Here's a screenshot:
[image]

It happens either on these two modes:
MBR partition scheme for BIOS or UEFI
GPT partition scheme for UEFI

- Tested on real machine, it fails.
- Tested on original untouched iso and modified iso, it fails.
- Tested on optical drive (burned DVD) and UEFI boot worked.

pbatard, 2016-01-20:
I need to see the __FULL LOG__ from Rufus of how you created your USB (when using GPT partition scheme for UEFI) before I can try to help you out. Please re-create your drive as you did, click on the log button, and then copy paste the __FULL__ content here (please do not truncate the log).

pbatard, 2016-01-20:
Also, can you provided the SHA-1 from your original untouched ISO? After selecting the ISO in Rufus, press the `#` button at the bottom. It will compute the SHA-1, which you can then paste here.

MrX-at-GitHub, 2016-01-20:
Hello,

Thanks for your quick reply. This is SHA-1 of the following iso file:
File     : es_windows_7_enterprise_with_sp1_x64_dvd_u_677652_71E51E3ACDE2A3285A62498200FAFF2FD2EF961F.iso
Size     : 2.90 GB
SHA1     : 71E51E3ACDE2A3285A62498200FAFF2FD2EF961F

I'll come back with the logfile you requested.

MrX-at-GitHub, 2016-01-21:
Also I found that if I extract the iso contents to a folder then recompile it again (modified iso), it fails.
Why? Well I found the \efiI\boot\bootx64.efi file is not written on the stick. I guess Rufus extracts that .efi file from somewhere.

pbatard, 2016-01-21:
Please provide the log as requested. I cannot help you unless I see the log. Thank you.

pbatard, 2016-01-21:
Looks like I missed the log, becausde you attached it by modifying your earlier comment (but I don't get a notification on that). For the record, if you attach something in the future, please make sure you create a new comment. Let me take a look at that and get back to you.

pbatard, 2016-01-21:
For Windows 7 x64, and unlike what is the case with later versions of Windows, Microsoft does not provide an `\efi\boot\bootx64.efi` on the ISO, so it needs to be extracted from `\sources\install.wim`, which contains a file called `Windows\Boot\EFI\bootmgfw.efi`, as `\efi\boot\bootx64.efi`.

If you look at the log, you will see the following (which is the reason why I *really* wanted to see your log):
```
Win7 EFI boot setup (this may take a while)...
Opening: E:\sources\install.wim:[1] (7-Zip)
Extracting: E:\efi\boot\bootx64.efi (From Windows\Boot\EFI\bootmgfw.efi)
  7z.exe did not extract E:\efi\boot\bootmgfw.efi
Opening: E:\sources\install.wim:[1] (API)
Extracting: E:\efi\boot\bootx64.efi (From Windows\Boot\EFI\bootmgfw.efi)
Closing: E:\sources\install.wim
```

Rufus tries 2 different methods to extract the required file from `install.wim`:
1. If you have 7-zip installed, it tries to use 7-zip. However, as you can see from above, even as you have 7-zip, it didn't manage to extract the file, which leads me to think that maybe your version of 7-zip is too old to handle WIM files. What version are you using? Can you please install the latest version from [here](https://sourceforge.net/p/sevenzip/discussion/45797/thread/09fd2931/)?
2. Using the native Windows WIM extraction API. This is usually slower than 7-zip, which is why 7-zip is called first. And aslo, this serves as a fallback, as you can see here, in case there was an issue during the 7-zip extraction attempt. What puzzles me here is that there wasn't an error message on this second attempt, which would tend to indicate that the file was properly extracted, but as per the error when you boot, as well as what you indicate, this doesn't seem to be the case. I am currently downloading `es_windows_7_enterprise_with_sp1_x64_dvd_u_677652.iso` and will try to replicate your issue. But I do suspect that, once you install a recent version of 7-zip, you should no longer see the extraction error, and find that your drive boots as expected.

pbatard, 2016-01-21:
Interesting. When using `es_windows_7_enterprise_with_sp1_x64_dvd_u_677652.iso` I got the same output from the log as you got:
```
Opening: F:\sources\install.wim:[1] (7-Zip)
Extracting: F:\efi\boot\bootx64.efi (From Windows\Boot\EFI\bootmgfw.efi)
  7z.exe did not extract F:\efi\boot\bootmgfw.efi
Opening: F:\sources\install.wim:[1] (API)
Extracting: F:\efi\boot\bootx64.efi (From Windows\Boot\EFI\bootmgfw.efi)
Closing: F:\sources\install.wim
```
However, the API extraction fallback worked, so I did get `\efi\boot\bootx64.efi` created on the USB.
So it doesn't look like upgrading 7-zip might work. Strangely, when using the 7-zip UI, I was able to open `install.wim` and extract the `Windows\Boot\EFI\bootmgfw.efi` file there, so this may be an issue with the commandline version of 7-zip.

Strangely also, when using `en_windows_7_ultimate_with_sp1_x64_dvd_618240.iso`, 7-zip worked as expected:
```
Win7 EFI boot setup (this may take a while)...
Opening: F:\sources\install.wim:[1] (7-Zip)
Extracting: F:\efi\boot\bootx64.efi (From Windows\Boot\EFI\bootmgfw.efi)
```

So there are a few things I need to figure out:
1. Why did the extraction fallback didn't seem to manage to successfully extract `\efi\boot\bootx64.efi` in your case, and expecially, why does the error log indicate that your extraction was successful. The only way it can do that is if [`WIMExtractImagePath()`](https://msdn.microsoft.com/en-us/library/windows/desktop/dd851957.aspx) returned TRUE, but according to Microsoft, that is not possible unless if the extraction was successful. Rufus is definitely set to check the return value and produce an error in the log if that function fails, so I really can't see how you can not have an error from the WIM API extraction, and yet no `\efi\boot\bootx64.efi`.
2. Why does 7-zip seem to fail when invoked from Rufus whereas the UI version extracts WIM files just fine
3. Why does the English/Ultimate version of Windows 7's `install.wim` appear to work just fine with 7-zip when invoked from Rufus, whereas the Spanish/Enterprise version doesn't?

I can explore #2 and #3 from my side, but #1 is just weird, since I can't reproduce it at all. Can you please run a test again, and confirm that there really isn't an `\efi\boot\bootx64.efi` file created once Rufus is done?

MrX-at-GitHub, 2016-01-22:
"Can you please run a test again, and confirm that there really isn't an \efi\boot\bootx64.efi file created once Rufus is done?"

Sorry for not replying earlier, duties in the first place lol
Well I'm sorry again because I didn't explain you correctly since my second post. The log show an error but USB stick has the efi file, as I said earlier in that post, the stick is working correctly.

Btw I have the latest version of 7-zip.

MrX-at-GitHub, 2016-01-22:
My problem is and I know you are going to say duh!!!! lol is I am using customized isos too! xD

In one of those isos I am compiling a dual boot and dual folder architecture Win7 AIO so Rufus can't read the default path to see the install.wim file.

pbatard, 2016-01-22:
>  I am compiling a dual boot and dual folder architecture Win7 AIO so Rufus can't read the default path to see the install.wim file

Well that explains it (also: Duh! :smile:). Of course if you create your own `install.wim` and move it in a different place, there is no way Rufus will be able to either locate or extract the file it needs to EFI boot, so your USB will not boot in UEFI mode. The only ISOs that are supported in Rufus are the official ones, because it's just too easy for someone to modify their images in a manner that renders them incompatible with a form of boot or another. So, if you don't use an official Microsoft ISO for your installation, you're on your own...

Still, your report tells me that I probably want to have a closer look at the 7-zip extraction process for the Windows 7 Enterprise images (I also found out that `en_windows_7_enterprise_with_sp1_x64_dvd_u_677651.iso` had the same issue as the spanish version).

Also, I'll improve UEFI:NTFS in the next version of Rufus, to make the message about not finding `/efi/boot/bootx64.efi` more explicit.

MrX-at-GitHub, 2016-01-22:
Alright, good to know you are working hard for improvements. Thanks a lot for that.

Now I think I could circumvent the Win7 AIO customized iso limitation by including bootx64.efi file in the modified iso. Either way Rufus creates a efi\microsoft\boot directory with the rest of the files inside (cdboot.efi, cdboot_noprompt.efi, etc but bootx64.efi) so copying that .efi file manually before compiling the iso should work for Rufus. I think.

pbatard, 2016-01-22:
That's what I would suggest. If you extract `Windows\Boot\EFI\bootmgfw.efi` from your `install.wim` and copy it as `\efi\boot\bootx64.efi` on the ISO, then your converted image will probably work. Rufus may still try to extract the file, but I don't think it will cancel the copy (and this happens at the very end, so even if it reports a failure, you should still be okay).

Of course, since you are not using an official image, what you are doing is unsupported...

MrX-at-GitHub, 2016-01-22:
"Of course, since you are not using an official image, what you are doing is unsupported..."

I understand that Pete, but myriads of people, for a myriads reasons of convenience, craft customized isos to make job easier, a lot.
I don't have a solid scientific statistical base to say this but I truly believe there are more customized isos than untouched isos to be copied to a stick whether is done with Rufus or any other program. 

Anyways, I'm going to start testing my idea and see what happens.

Thank you so much for your kind and quick support. Amazing!

MrX-at-GitHub, 2016-01-23:
Hi

Just came back to tell it worked. The bcd file and folders are correctly copied. Nice!!!

Regards.

pbatard, 2016-01-23:
Happy to hear that! Please leave this issue open, as I will be closing it when I have addressed the 7-zip/Enterprise extraction issues.

As to your earlier comment, there might be myriads of people, but I hope you do understand that, unlike what is the case for a Microsoft, Goggle or Apple, there is only one guy, working in his spare time, who would be tasked to support them. Oh, and what's more, someone who's providing their software entirely for free and who's not being paid for their development. There is NO WAY I can realistically support unofficial images. So if you use them, you will __always__ be 100% on your own. The only thing I can do is try to explain failures resulting from such use a bit more clearer.

MrX-at-GitHub, 2016-01-23:
I understand Pete and I agree. Fortunately, in my case problem was circumvented so easily that I am very glad it happened this way.
As for the 7-zip / Enterprise extraction issue I just can say thanks for such amazing work you do.

pbatard, 2016-01-26:
Okay, the issue between Enterprise and Ultimate is that when there are multiple Images within a WIM, 7z requires the image index to appear before the path (e.g. `1\Windows\Boot\EFI\bootmgfw.efi`). But when there is only one image in a WIM, 7z requires the index not be used (e.g. `Windows\Boot\EFI\bootmgfw.efi`).

I wasn't aware of this quirk until now. Unfortunately the `install.wim` for Windows 7 Ultimate (which is what we mostly tested Windows 7 with) contains 4 images, so we have always been adding the index when invoking `7z` in Rufus. The `install.wim` for Windows 7 Enterprise however contains only one, which means that `7z` was unable to process the path, and failed...

I should be able to fix this by trying to run `7z` both with and without the index.

commit 1f2b73df (2016-01-26): [vhd] fix an issue when extracting WIM with 7z

* 7z has a quirk in that one MUST specify the image index
  when a WIM has multiple images, but the same index MUST
  be omitted if there is only one.
* Because of this, Windows 7 Enterprise images reported a
  (nonfatal) extraction error when trying 7z.
* Closes #680
* Also fix a typo in registry.h
closed 2016-01-26 by pbatard