pbatard/rufus #682: ** i have good idea

hunandy14, 2016-01-23:
my english is not good but
please read my message 
this is good idea and 

this idea from "itiankong"
https://www.itiankong.net/thread-362761-1-1.html

this is Windows PE
it installer is interesting
install partition is three cutting
    |  ESP  |  USERDATA partition  | boot partition |

system install to boot partition

this is good idea

you think if you have a 64GB USB
now you make a Windows install USB
Windows only 4GB but USB have 64GB

60GB is not use
if you want use it
you can create a folder
i do it
finally windows is not install 
need recreate

so can you add this function?

down is original message
----------------------------------------------------------------------------------
我英文不是很好，但請試著閱讀，我認為這是很好的點子

這個點子來自於IT天空
https://www.itiankong.net/thread-362761-1-1.html
這是windows PE
他的安裝方法很特別會將磁區分開
這是所謂的3分區
| ESP | DATA partion  | boot partion |
他將PE安裝在第三分區
這樣可以極力避免使用者操作失誤
原本你的程序會將它安裝在同一分區
實際上使用發現這很容易出問題
一個64G的隨身碟不可能為了安裝W10
只耗費4G就不使用他，實際上我用資料夾分開
用一段時間之後W10將不能安裝
必須重新製作，但是如果用這種3分區的辦法
就不會發生這種事情了

pbatard, 2016-01-23:
Note: I cannot open your link.

> this is good idea

Not it's not. 99% of flash drives out there don't have the FIXED attribute, which means Windows can only ever mount ONE partition at a time. Therefore people would not be able to access the Boot partition.

Also trying to preserve existing data is a MAJOR headache and creates all kind of reliability issues (the 'R' in Rufus stands for 'Reliable') because you can no longer control the set of initial conditions when creating the drive. Also what happens if you want to convert from MBR to GPT?

Besides, this seems aimed at WinPE. What about GRUB or Syslinux which are widely used for Linux?

Finally, [I have no plans to support multiple partitions/keep existing partitions in Rufus](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_add_multiple_partition_support). Rufus is a formatting tool. It is meant to delete all existing data from the drive to perform its operations. If you are unhappy with that, I will kindly request that you find another tool.

closed 2016-01-23 by pbatard

hunandy14, 2016-01-23:
i know.
today i have a new discovery .
Windows PE copy file to RAM (file is 200MB)
so,it can multiple partition.

windows install USB Cutting partition.
today i try it. again and again.
finally USB boot success but don't read install file.

i get this idea. i excitement.
finally Rufus is good soft.
l very like.
thank you.

pbatard, 2016-01-23:
Not gonna happen. Sorry.

pbatard, 2016-01-23:
Also, just so you know, the whole point of Rufus is to create an USB that is as close as possible to the ISO.
If the ISO uses one partition, then Rufus will use one partition.

Deviating from the ISO content __always__ creates issues.