pbatard/rufus #683: Operation run time font color is hardcoded, and cannot obey Windows themes

cookieboyeli, 2016-01-25:
The font is hardcoded gray on all themes, breaking Windows High Contrast support.
(Sorry I know it's a PITA). It doesn't meet the required contrast values for official High Contrast support and can become completly invisible depending on background color.

 Here's some pictures to show the breakage:
[image]
[image]
[image]
Not even windows supports WIndows high contrast correctly. Switching between it and regular themes leaves a multitude of elements stuck on the old colors, some persisting through reboots and other theme changes! All these lovely bugs are in Windows 10 too. Great how they're trying to keep it familiar for the user. /s

pbatard, 2016-01-25:
I'm gonna have to pass on that one, sorry, on account that:
1. The duration of the operation is not critical at all to Rufus operations.
2. I'll try to take a look at high contrast and following the user's font choices when I redesign the UI for Rufus 3.0 anyway
3. I have more important issues to deal with, and I'm already behind where I'd like to be for the 2.7 release.

closed 2016-01-25 by pbatard

cookieboyeli, 2016-01-25:
Yup, no problem. It is a mostly trivial UI function. Just making sure it's on your radar.