pbatard/rufus #687: No Ubuntu boot on a tabletPC

Nic2012, 2016-02-05:
I am really confused, created a bootable USB Stick with Rufus Rufus 2.6.818 and one of the following images linuxmint-17.3-cinnamon-64bit, kubuntu-15.10-desktop-amd64 or ubuntu-15.10-desktop-amd64, but each doesnt boot on a tablet PC ASUS VivoTab Note 8 M80T Win10pro. Bios shows the usb stick as EFI device in boot options. Fast and secure boot is disabled.

When i try to boot from stick there is a short break of 2-3 seconds black screen and no logs etc. then system starts with win10.

But the only image that works on this tablet is from Parted Magic 2015-11-13, i can see its using Grub 2.00 as bootloader, I can select between a 32 or 64bit version and it runs very fine on the ASUS.

Whats wrong with the Ubuntu distros ?

Best regards,
Nic

pbatard, 2016-02-05:
Did you use GPT for UEFI as the partition scheme by any chance?

If so, please be aware that Debian derivatives, like Ubuntu, probably have the same issue as the one [reported here](https://github.com/pbatard/rufus/issues/531#issuecomment-111892364) (also [here](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=789600)) in that they forgot to include the GPT module in their GRUB version. This means that it won't be able to boot (but you should see the GRUB prompt then). This has been fixed in the very latest Debian (8.3) but it may have to find its way downstream.

If not, __PLEASE__ provide the __FULL__ Rufus log of how you created your USB (preferably using `ubuntu-15.10-desktop-amd64.iso` as it's probably better to work with the most popular), and also indicate if you have tried both the ISO and DD mode. I know you __will__ get prompted to choose between ISO and DD mode when using `ubuntu-15.10-desktop-amd64.iso`. Have you tried both?

Nic2012, 2016-02-05:
Tested it with **ubuntu-15.10-desktop-amd64.iso**, but it doesnt matter if DD or ISO mode, both won't boot.

**pmagic_2015_11_13.iso** is the only one that I can boot and run fine, which is a UDF image in opposite to the others (ISO9660 Image). No matter if GPT or MBR mixed mode and ISOImage mode both will run. But in case of DDImage **Parted Magic 2015-11-13** will finally stop in GRUB prompt.

Let me know if y need another run or log for further investigations.
On weekend I will test it with Debian 8.3.
Thanks for y support and best regards,
Nic

[logs.zip](https://github.com/pbatard/rufus/files/119977/logs.zip

pbatard, 2016-02-07:
Thanks.

There is definitely a hardware component to your issue, as I have no problem getting `ubuntu-15.10-desktop-amd64.iso` to boot on the platforms I tried, and also your logs look okay to me.

It may be an incompatibility between the Ubuntu kernel or Syslinux (since that's what Ubuntu uses as a boot loader, even for UEFI) and your hardware. You may also want to run a bad blocks check, or try with a different USB, just in case.

I'm afraid there's not much I can do in Rufus with regards to these kind of hardware specific issues. But one thing you may try, to see if it's a Syslinux issue, is use Rufus as follows with your USB plugged in (no need to select an ISO for this):
1. Under _Partition scheme and target system type_ make sure you select _MBR partition scheme for BIOS or UEFI_
2. Click on the white triangle besides _Format Options_ so that it displays the advanced panel
3. Now, in the _Create bootable disk using_ dropdown and select _Syslinux 6.03_. Note that this option will only be populated when the advanced panel is displayed
4. Click start and try to boot with that USB.

If Syslinux works with your system, then you should see a prompt such as this:
``` 
SYSLINUX 6.03 EDD 2014-10-06 Copyright (C) 1994-2014 H. Peter Anvin et al.
WARNING: No configuration file found
boot: _
```

If you don't see that, then either there's an incompatibility with Syslinux and your system, or your system can simply not boot in UEFI-CSM mode (but then you should probably still have seen something when trying Ubuntu in GPT).

Nic2012, 2016-02-07:
Probably I run into this problem described here: http://askubuntu.com/questions/392719/32-bit-uefi-boot-support and a workaround could be like this http://www.jfwhome.com/2014/03/07/perfect-ubuntu-or-other-linux-on-the-asus-transformer-book-t100/.

The Asus Vivitab 8 has a Win10 32bit env., so I assume it only runs on a 32bit UEFI but the CPU has x64(!), if i add the mentioned `bootia32.efi` under EFI/boot on LiveUSB (in Rufus with GPT/Fat32/Iso mode; disabled Fast+Secure boot in BIOS) then system boots stick, showing grub menu and finally i can load/run e.g. Lubuntu64 in live env. That's fine fo far.

Yes it has nothing to do with Rufus, so we can close the ticket. May I give a suggestion to y program: maybe y put sth. as a warning or hint, that user should take care what kind of UEFi Bios they have on pc.

Thanks for y support and providing us with y useful utility.
Best regards,
Nic

pbatard, 2016-02-08:
Good deductive work! 

I kind of blame myself for not having seen what the issue was before you did, as it is obvious now that you mention it, and I am well aware that some manufacturers go with a 32-bit UEFI even with a 64-bit CPU, [in order to claim some space](http://blogs.intel.com/evangelists/2015/07/22/why-cheap-systems-run-32-bit-uefi-on-x64-systems/).

One thing I will be doing in Rufus 2.7, when UEFI:NTFS is being used (which wouldn't have helped in your case, as it's mostly used for Windows ISOs) is indicate whether the platform UEFI is 32 or 64 bit, so that should help detect these kind of issues during Windows UEFI installation. On the other hand, popping a message every time someone creates an UEFI bootable USB, to indicate that they should be mindful of the actual bitness of their target system is not something I think I want to do. The reason is, there are actually plenty of warnings such as these that I could add (the 32 vs 64 bit UEFI is just one of many things that one might need to be aware of), so if I start warning about this, I would also have to warn people about that they may have to disable Secure Boot, be careful about whether CSM is enabled or not, and other things to keep everyone happy. This would rapidly become exceedingly annoying for everyone...

I will close this issue now.

closed 2016-02-08 by pbatard