pbatard/rufus #690: Impossible to create USB with Ubuntu 15.10

shineworld, 2016-02-09:
Rufus version: 2.6.818
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0410
Found USB 2.0 device 'USB Mass  Storage Device USB Device' (1A40:0201)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00EDEC91
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Checking for Rufus updates...
Checking release channel...
No new release version found.
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20150813 (from '/isolinux/isolinux.bin')
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
  Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 15.10 amd64'
  Size: 1178386432 bytes
  Has a >64 chars filename: No
  Has Symlinks: Yes
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (6.03)
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: ubuntu-15.10-desktop-amd64.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20150813/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
[... 259 lines cut ...]
Extracting: G:\casper\filesystem.manifest (62.9 KB)
Extracting: G:\casper\filesystem.manifest-remove (1.7 KB)
Extracting: G:\casper\filesystem.size (11 bytes)
Extracting: G:\casper\filesystem.squashfs (1.1 GB)
  Error writing file: [0x00000002] Impossibile trovare il file specificato.
  RETRYING...
  Error writing file: [0x00000037] La risorsa o la periferica di rete specificata non è più disponibile.
  RETRYING...
  Error writing file: [0x00000022] Dischetto errato inserito nell'unità.
Inserire %2 (numero di serie del volume: %3) nell'unità %1.
  RETRYING...
  Error writing file: [0x000003EE] Il volume corrispondente al file è stato alterato dall'esterno. Il file aperto non è più valido.
Created: G:\syslinux.cfg

Found USB 2.0 device 'USB Mass  Storage Device USB Device' (1A40:0201)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0174F24D
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2016-02-09:
Looks to me like your USB is defective.

Please run a bad blocks check (check the _Test controllo blocchi errati unità_ option).

pbatard, 2016-02-15:
Haven't heard back from OP with regards to what is most likely a hardware issue with their flash drive &rarr; Closing.

closed 2016-02-15 by pbatard