pbatard/rufus #693: ER: Add non USB card reader support

pbatard, 2016-02-12:
And also improve USB card reader support, for devices that don't use `USBSTOR` as the card reader driver.

This is all made possible thanks to the good work of @Herst in issue #334.

pbatard, 2016-02-16:
@Herst,

Please find [a new TEST version](https://rufus.akeo.ie/testing/rufus-2.7_TEST12.exe) (TEST 12) that I expect to work with your `PCISTOR` card reader on device No.1. And of course, it should also work with your other devices...

If you feel like testing and reporting back, so that I can push the code changes, I'd appreciate it.

Herst, 2016-02-16:
Device № 1:
```
Rufus version: 2.7.858 (Test12)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
System locale ID: 0x0409
Will use default UI locale 0x0409
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_8564&pid_4000#5&11cbd2d0&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[246]: USB\VID_0403&PID_6001\FTYO5SYU
Processing Hub 2:
Processing Hub 3:
  Hub[2] = '\\?\usb#root_hub30#4&1a0eaf57&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[208]: USB\VID_8086&PID_0A66\019150104604
  Found ID[165]: USB\VID_046D&PID_C07F\0A8034533934
  Found ID[093]: USB\VID_8564&PID_4000\5&11CBD2D0&0&1
  Found ID[019]: USB\VID_8564&PID_4000\5&11CBD2D0&0&13
Processing 'PCISTOR' device:
  Hardware ID: 'RSPCIESTOR\GenDisk'
Found non-USB removable device 'SDHC Card'
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk_Samsun_SSD_850_EVO_250GEMT0'
  Non-USB or non-removable => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk_Samsun_SSD_850_EVO_250GEMT0'
  Non-USB or non-removable => Eliminated
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3918, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xA6202AF7
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 56 MB (58720256 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 3.0 GB (3213885440 bytes)
  Start Sector: 122880, Boot: No, Recognized: Yes
```
Apparently works (did not try to flashing though). Great work!

Herst, 2016-02-16:
Device № 3:
```
Rufus version: 2.7.858 (Test12)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
System locale ID: 0x0409
Will use default UI locale 0x0409
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub30#4&48210f7&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[023]: USB\VID_1286&PID_204C\0000000000000000
  Found ID[132]: USB\VID_045E&PID_07DE\5&26F3984C&0&3
Processing 'SD' device:
  Hardware ID: 'SD\GenDisk'
Found card reader device 'Samsung MDGAGC'
Processing 'SD' device:
  Hardware ID: 'SD\GenDisk'
Found non-USB removable device 'Generic USDU1 SD Card'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3918, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xA6202AF7
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 56 MB (58720256 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 3.0 GB (3213885440 bytes)
  Start Sector: 122880, Boot: No, Recognized: Yes
```

pbatard, 2016-02-16:
Great! Thanks for the reports.

Your latest logs allowed me to find one final small issue (which I have now corrected), where the log says `Found non-USB removable device...` instead of `Found card reader device...`. Hopefully, it should be all good now. Once again, thanks for helping make Rufus better!

I'm going to close this issue now, as I pushing the code. But if you, or anybody else interested in card reader support for the upcoming version, want to run some more tests, feel free to use the latest from [here](https://rufus.akeo.ie/testing/) (currently TEST13) and report if you find any issues.

closed 2016-02-16 by pbatard

Herst, 2016-02-18:
Device № 1
```
Rufus version: 2.7.858 (Test13)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
System locale ID: 0x0409
Will use default UI locale 0x0409
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_2109&pid_8110#5&8498a73&0&17#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[067]: USB\VID_0B95&PID_1790\00000000000001
  Found ID[034]: USB\VID_0B95&PID_1790\6&5A74B14&0&2
Processing Hub 2:
Processing Hub 3:
Processing Hub 4:
Processing Hub 5:
  Hub[2] = '\\?\usb#root_hub20#4&1cc8b60&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[189]: USB\VID_8087&PID_8009\5&DFA438D&0&1
Processing Hub 6:
  Hub[3] = '\\?\usb#root_hub30#4&291bc6d2&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[109]: USB\VID_0BDA&PID_0327\201312261010
  Found ID[027]: USB\VID_045E&PID_00DB\5&8498A73&0&4
  Found ID[178]: USB\VID_046D&PID_C069\5&8498A73&0&11
  Found ID[068]: USB\VID_2109&PID_8110\5&8498A73&0&17
  Found ID[003]: USB\VID_2109&PID_2811\5&8498A73&0&6
Processing Hub 7:
  Hub[4] = '\\?\usb#root_hub20#4&abbcf3e&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[213]: USB\VID_8087&PID_8001\5&6E0A24C&0&1
Processing IDs belonging to 'RTSUER':
  USB\VID_0BDA&PID_0327\201312261010
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\GenDisk'
  Matched with ID[109]: USB\VID_0BDA&PID_0327\201312261010
  Matched with Hub[3]: '\\?\usb#root_hub30#4&291bc6d2&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 3.0 device 'SDHC Card' (0BDA:0327)
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk_WD_____WD10EZEX-60M2NA001.0'
  Non-USB or non-removable => Eliminated
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3881, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive has an unknown Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 29.7 GB (31925993472 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
```

Herst, 2016-02-23:
An because I just got the chance and just for fun, here the newest testing build on yet another device with a SD card slot, I present you device № 4 where I booted a Windows-to-Go (for some updates):
```
Rufus version: 2.8.861 (Test3)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
System locale ID: 0x0409
Will use default UI locale 0x0409
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub20#4&30b1fc35&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[020]: USB\VID_8087&PID_8001\5&A465610&0&1
Processing Hub 3:
  Hub[2] = '\\?\usb#root_hub30#4&137729c2&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[009]: USB\VID_2537&PID_1066\0123456789ABDFE
  Found ID[170]: USB\VID_1B1A&PID_7001\5&9C972B&0&3
  Found ID[134]: USB\VID_05E3&PID_0608\5&9C972B&0&5
  Found ID[090]: USB\VID_0E8F&PID_0022\5&9C972B&0&1
Processing Hub 4:
  Hub[3] = '\\?\usb#vid_05e3&pid_0608#5&9c972b&0&5#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[100]: USB\VID_05E3&PID_0723\6&2E24097&0&2
Processing IDs belonging to 'USBSTOR':
  USB\VID_2537&PID_1066\0123456789ABDFE
  USB\VID_05E3&PID_0723\6&2e24097&0&2
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskGeneric_STORAGE_DEVICE__9454'
TRYING 'USB\VID_2537&PID_1066\0123456789ABDFE' [Inst: b]
  NO MATCH
TRYING 'USB\VID_05E3&PID_0723\6&2e24097&0&2' [Inst: 9]
  Matched with ID[169]: USB\VID_05E3&PID_0723\6&2e24097&0&2
  Matched with (GP) ID[134]: USB\VID_05E3&PID_0608\5&9C972B&0&5
  Matched with Hub[2]: '\\?\usb#root_hub30#4&137729c2&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0608)
Device eliminated because it appears to contain no media
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSamsung_SSD_850_EVO_120GEMT0'
TRYING 'USB\VID_2537&PID_1066\0123456789ABDFE' [Inst: b]
  NO MATCH
TRYING 'USB\VID_05E3&PID_0723\6&2e24097&0&2' [Inst: 9]
  NO MATCH
  Non-USB or non-removable => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskInateck_________________0012'
TRYING 'USB\VID_2537&PID_1066\0123456789ABDFE' [Inst: b]
  Matched with ID[009]: USB\VID_2537&PID_1066\0123456789ABDFE
  Matched with Hub[2]: '\\?\usb#root_hub30#4&137729c2&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 3.0 device 'Inateck USB Device' (2537:1066)
Using autorun.inf label for drive C: 'J_CCSA_X64FRE_EN-US_DV5'
Device eliminated because it was detected as a Hard Drive (score 3 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub20#4&30b1fc35&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[020]: USB\VID_8087&PID_8001\5&A465610&0&1
Processing Hub 3:
  Hub[2] = '\\?\usb#root_hub30#4&137729c2&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[009]: USB\VID_2537&PID_1066\0123456789ABDFE
  Found ID[170]: USB\VID_1B1A&PID_7001\5&9C972B&0&3
  Found ID[134]: USB\VID_05E3&PID_0608\5&9C972B&0&5
  Found ID[090]: USB\VID_0E8F&PID_0022\5&9C972B&0&1
Processing Hub 4:
  Hub[3] = '\\?\usb#vid_05e3&pid_0608#5&9c972b&0&5#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[100]: USB\VID_05E3&PID_0723\6&2E24097&0&2
Processing IDs belonging to 'USBSTOR':
  USB\VID_2537&PID_1066\0123456789ABDFE
  USB\VID_05E3&PID_0723\6&2e24097&0&2
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskGeneric_STORAGE_DEVICE__9454'
[... 99 lines cut ...]
TRYING 'USB\VID_05E3&PID_0723\6&2e24097&0&2' [Inst: 9]
  NO MATCH
  Non-USB or non-removable => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskInateck_________________0012'
TRYING 'USB\VID_2537&PID_1066\0123456789ABDFE' [Inst: b]
  Matched with ID[009]: USB\VID_2537&PID_1066\0123456789ABDFE
  Matched with Hub[2]: '\\?\usb#root_hub30#4&137729c2&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 3.0 device 'Inateck USB Device' (2537:1066)
Using autorun.inf label for drive C: 'J_CCSA_X64FRE_EN-US_DV5'
Device eliminated because it was detected as a Hard Drive (score 3 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7783, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Size: 59.6 GB (64005079040 bytes)
  Start Sector: 32768, Boot: No, Recognized: Yes
```
SD card was available for formatting and the Windows-to-Go-drive not (since I didn't check the checkbox) so seems OK to me but maybe you find again something in the log that could be improved.

pbatard, 2016-02-24:
This looks OK to me. Thanks for the report!

commit f88faf1a (2016-02-16): [usb] enable listing of non USB card readers

* Also fix the VID:PID population of USB card readers
* Also improve enumeration debugging
* Also add an unofficial cheat mode to list non USB *REMOVABLE* drives
* Closes #693