pbatard/rufus #696: v2.8 b859 (x86 and AMD64 build, console/GUI run) - don't intialise GUI at startup and freeze...

VictorVG, 2016-02-19:
Win7 SP1 x64 wait on startup external evet(?). Treads list is:

9408, , , ntdll.dll!RtlDecodeSystemPointer+0xadf, Normal
8568, , , ntdll.dll!TpCallbackIndependent+0x238, Normal
12240, , , ntdll.dll!TpCallbackIndependent+0x238, Normal
12476, 6,77, 1 847 956 011, rufus.exe+0x4e3eb, Normal

At the same hardware access for the purpose of a survey of its characteristics, creating a window or another activity besides the expectations of the unknown event is not observed, the download Core i7-2600 objective of the program is around 6.7% - 7.1%. Such a condition can be observed definitely not a long time ...

VictorVG, 2016-02-19:
Then reboot OS and start Rufus.exe using Explorer.exe I see thread list:

4884, , , ntdll.dll!TpIsTimerSet+0x7b0, Normal
6380, 6,09, 1 660 616 588, rufus.exe+0x4eadc, Normal

after 3 minit's and see debug log (MS VC++ 2015 call on build server (NFS)):

[log.txt](https://github.com/pbatard/rufus/files/137728/log.txt)

pbatard, 2016-02-19:
Thanks for the report. I can't see the issue on any Windows 7 (or other machine) I tested with so far.

Having the thread list doesn't help me much, but having the log does. Since b858 modified device enumeration, the problem is probably there, so I need you to do the following:
1. Using the release version of Rufus 2.7 (or any other recent version of Rufus that starts on your machine), press <kbd>Alt</kbd>-</kbd>.</kbd> to enable enumeration logging. You should see `Debug mode enabled` message on the status bar, and the Rufus log will contain info about the enumeration process
2. Still using the version of Rufus that starts, post your full log with enumeration enabled, so that I can see your devices
3. Now try to run b859, and post the log from Visual Studio. Debug enumeration is persisted in the registry, so you should have a bit more information in the log. Hopefully that will help find out what is preventing devices to be listed on your machine.

> At the same hardware access for the purpose of a survey of its characteristics, creating a window or another activity besides the expectations of the unknown event is not observed

I'm sorry but I don't understand what you are saying here. What does it mean?

VictorVG, 2016-02-19:
Please, see log:
[rufus.log.txt](https://github.com/pbatard/rufus/files/137900/rufus.log.txt) and HWiNFO64 hardware log 
[hardware.LOG.txt](https://github.com/pbatard/rufus/files/137901/hardware.LOG.txt)

I realized hang occurs when the USB polling devices such as USB 3.0 controllers, or BT, keyboard and mouse. Display window b8558 - b859 to the front I could not plan, as well as their attempt to fine-tune failed ...

pbatard, 2016-02-19:
Thanks. Can I get the log from Visual Studio when launching Rufus 2.8.859 again, please? Now that you have turned USB Debug on in 2.7.855, it should contain more information, and I am very interested in the new data.

VictorVG, 2016-02-19:
I'll try later, as he is working under BSD UNIX, and Win7 run from an NFS server in VirtualBOX. We'll have to reconfigure part of the .etc/*.conf through which compilers are called, and they are the guys servers installed - in my UNIX gcc enough, but to be honest and not so often have to update packages - bugs fortunately very few ...

VictorVG, 2016-02-20:
**b860** - show GUI, enumerate device and debug is failure, but MS VC++ Static Code Analaiser write 42 warning in to log:

> Серьезность	Код	Описание	Проект	Файл	Строка	Состояние подавления
> Предупреждение	C28252	Несогласованная аннотация для "strcpy": _Param_(1) содержит "SAL_writableTo(elementCount(_String_length_(__formal(1,parameter1))+1))" в предыдущем экземпляре. См. <no file>(0). 	bled	c:\program files (x86)\windows kits\10\include\10.0.10240.0\ucrt\string.h	130	Активно
> Предупреждение	C28253	Несогласованная аннотация для "strcpy": _Param_(1) содержит "SAL_writableTo(elementCount(_String_length_(_Source)+1))" в данном экземпляре. См. <no file>(0). 	bled	c:\program files (x86)\windows kits\10\include\10.0.10240.0\ucrt\string.h	130	Активно
> Предупреждение	C6011	Разыменование пустого указателя "bd". 	bled	rufus\src\bled\decompress_bunzip2.c	685	Активно
> Предупреждение	C6011	Разыменование пустого указателя "outbuf". Самое раннее расположение, где это могло произойти, см. в строке 791	bled	rufus\src\bled\decompress_bunzip2.c	797	Активно
> Предупреждение	C6011	Разыменование пустого указателя "state". 	bled	rufus\src\bled\decompress_gunzip.c	1045	Активно
> Предупреждение	C6011	Разыменование пустого указателя "state". 	bled	rufus\src\bled\decompress_gunzip.c	1217	Активно
> Предупреждение	C6011	Разыменование пустого указателя "state->bytebuffer". 	bled	rufus\src\bled\decompress_gunzip.c	1261	Активно
> Предупреждение	C6011	Разыменование пустого указателя "rc". 	bled	rufus\src\bled\decompress_unlzma.c	83	Активно
> Предупреждение	C6385	Чтение недопустимых данных из "buffer":  доступный для чтения объем равен "(((header.dst_size))<((header.dict_size)))?(header.dst_size):(header.dict_size)" байт, однако считать можно только "3" байт.	bled	rufus\src\bled\decompress_unlzma.c	439	Активно
> Предупреждение	C6011	Разыменование пустого указателя "archive_handle". 	bled	rufus\src\bled\init_handle.c	15	Активно
> Предупреждение	C28252	Несогласованная аннотация для "strcpy": _Param_(1) содержит "SAL_writableTo(elementCount(_String_length_(__formal(1,parameter1))+1))" в предыдущем экземпляре. См. <no file>(0). 	libcdio-driver	c:\program files (x86)\windows kits\10\include\10.0.10240.0\ucrt\string.h	130	Активно
> Предупреждение	C28253	Несогласованная аннотация для "strcpy": _Param_(1) содержит "SAL_writableTo(elementCount(_String_length_(_Source)+1))" в данном экземпляре. См. <no file>(0). 	libcdio-driver	c:\program files (x86)\windows kits\10\include\10.0.10240.0\ucrt\string.h	130	Активно
> Предупреждение	C6011	Разыменование пустого указателя "p_new_node". 	libcdio-driver	rufus\src\libcdio\driver\ds.c	91	Активно
> Предупреждение	C6011	Разыменование пустого указателя "p_new_node". 	libcdio-driver	rufus\src\libcdio\driver\ds.c	115	Активно
> Предупреждение	C6031	Возвращаемое значение пропущено: "snprintf".	libcdio-driver	rufus\src\libcdio\driver\sector.c	171	Активно
> Предупреждение	C6011	Разыменование пустого указателя "*dst". 	libcdio-driver	rufus\src\libcdio\driver\utf8.c	304	Активно
> Предупреждение	C6011	Разыменование пустого указателя "le_src". 	libcdio-driver	rufus\src\libcdio\driver\utf8.c	337	Активно
> Предупреждение	C6001	Использование неинициализированной памяти "*strv".	libcdio-driver	rufus\src\libcdio\driver\util.c	68	Активно
> Предупреждение	C6011	Разыменование пустого указателя "strv". 	libcdio-driver	rufus\src\libcdio\driver\util.c	99	Активно
> Предупреждение	C6387	"new_mem" может быть равен "0":  это не соответствует спецификации функции "memcpy". 	libcdio-driver	rufus\src\libcdio\driver\util.c	114	Активно
> Предупреждение	C6001	Использование неинициализированной памяти "*ud".	libcdio-driver	rufus\src\libcdio\driver\_cdio_stdio.c	140	Активно
> Предупреждение	C6011	Разыменование пустого указателя "ud". 	libcdio-driver	rufus\src\libcdio\driver\_cdio_stdio.c	271	Активно
> Предупреждение	C6011	Разыменование пустого указателя "new_obj". 	libcdio-driver	rufus\src\libcdio\driver\_cdio_stream.c	105	Активно
> Предупреждение	C28252	Несогласованная аннотация для "strcpy": _Param_(1) содержит "SAL_writableTo(elementCount(_String_length_(__formal(1,parameter1))+1))" в предыдущем экземпляре. См. <no file>(0). 	libcdio-iso9660	c:\program files (x86)\windows kits\10\include\10.0.10240.0\ucrt\string.h	130	Активно
> Предупреждение	C28253	Несогласованная аннотация для "strcpy": _Param_(1) содержит "SAL_writableTo(elementCount(_String_length_(_Source)+1))" в данном экземпляре. См. <no file>(0). 	libcdio-iso9660	c:\program files (x86)\windows kits\10\include\10.0.10240.0\ucrt\string.h	130	Активно
> Предупреждение	C6031	Возвращаемое значение пропущено: "snprintf".	libcdio-iso9660	rufus\src\libcdio\iso9660\iso9660.c	369	Активно
> Предупреждение	C6031	Возвращаемое значение пропущено: "snprintf".	libcdio-iso9660	rufus\src\libcdio\iso9660\iso9660.c	1117	Активно
> Предупреждение	C6387	"p_stat" может быть равен "0":  это не соответствует спецификации функции "memcpy". 	libcdio-iso9660	rufus\src\libcdio\iso9660\iso9660_fs.c	953	Активно
> Предупреждение	C6011	Разыменование пустого указателя "p_stat". Самое раннее расположение, где это могло произойти, см. в строке 953	libcdio-iso9660	rufus\src\libcdio\iso9660\iso9660_fs.c	954	Активно
> Предупреждение	C6387	"p_stat->rr.psz_symlink" может быть равен "0":  это не соответствует спецификации функции "memcpy". 	libcdio-iso9660	rufus\src\libcdio\iso9660\iso9660_fs.c	955	Активно
[... cut ...]
> Предупреждение	C6385	Чтение недопустимых данных из "advtmp":  доступный для чтения объем равен "_Old_5`left-plen" байт, однако считать можно только "500" байт.	syslinux-libinstaller	rufus\src\syslinux\libinstaller\setadv.c	123	Активно
> Предупреждение	C28253	Несогласованная аннотация для "strcpy": _Param_(1) содержит "SAL_writableTo(elementCount(_String_length_(_Source)+1))" в данном экземпляре. См. <no file>(0). 	syslinux-libinstaller	c:\program files (x86)\windows kits\10\include\10.0.10240.0\ucrt\string.h	130	Активно

Currently I have and send to Your only this  information... :(

VictorVG, 2016-02-20:
By the way, by the way, but because we have lost sight of a very insidious device - built-in USB Card-Reader - I have these glands made Acorp already so long ago cost that simply forgot about them, but they had run UNIX are minutes twenty BTX hold and I am disabled at their ./etc/fstab - if that is where supplied, give the command "mount ... sda ..." I do not for a long time, and Bastik does not expect a reply from them for three days, checks and everything else and starts X-s ...

Skaendo, 2016-02-20:
I'm having issues with 2.7 as well. It seems to write the ISO to a USB stick fine, and when I load the USB into a running computer everything shows up normal, but when I try to boot from the device, I get nothing. 

I am attempting to use the MBR for BIOS/UEFI, but it ends up writing as GPT for UEFI. I am using it on older hardware that doesn't have UEFI support, and Windows 7 x64 Pro.

pbatard, 2016-02-20:
> b860 - show GUI, enumerate device and debug is failure

I don't understand what you write. Does this mean that b860 does work?

>  but MS VC++ Static Code Analaiser write 42 warning in to log

As long as Rufus compiles, this is irrelevant to the issue. One problem at a time, __please__!
If the GUI doesn't show with b860, then what I am interested in the output from Visual Studio from when you launch it. Basically, I want the same thing from your `log.txt` from the second post.

> I'm having issues with 2.7 as well.

One problem at a time, __please__!

The __only__ thing I am interested with this issue is Rufus b858 or later not displaying a GUI (most likely due to an infinite loop during enumeration, which is why I __really__ want to see your VS output when launching b858 or later with Visual Studio, and with Enumeration Debug turned on).
Anything that is a different problem than the GUI not displaying should go into a separate issue.

pbatard, 2016-02-20:
Actually, @Skaendo, you shouldn't have hijacked this post, as your problem is unrelated. I will therefore delete your comment. Please do not hijack an existing issue unless you are facing the exact same problem (which you don't - you are not compiling the latest Rufus!).

VictorVG, 2016-02-20:
Well, in this case, what happens is you have other reasons, I do not program displays its own window on startup, and maybe I'm wrong, but intuition tells me that the perpetrators of these "miracles" is a card reader Acorp CRIPx00. I've got a pair of slightly different brands - one Acorp CRIP100, and the second Acorp CRIP300, and the fact that in one of their slots no media you will never know for the device in this case simply does not respond when accessing this slot, and whether there is a map it or not before mounting the date for the OS puzzle.

And if so, I think I know how this very disease we treat - should they poll in two phases - the first uvidile that this card reader, remember it, and not to chudil limit the time waiting for his answer, and check there is in him that either no only when you select it from the GUI that was controllable by the user program.

pbatard, 2016-02-20:
Please provide the log I requested with Debug enabled. That's the only way I can __SEE__ what happens and how I can fix the issue. ANYTHING else is guesswork.

VictorVG, 2016-02-20:
Here the trouble is that I can not give the program a command to create it. Although, although, as with the command-line we can run rufus in logging mode?

VictorVG, 2016-02-20:
Yes, I found the offending device - reader. If it is turned off once the program starts. I gave the command to stop OS CF card-reader b860 and immediately got out of the standby state. This is his response awaited program. And I just did not realize it ... :(

Log then this device is stopped:

`Rufus version: 2.8.860
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
System locale ID: 0x0419
Will use default UI locale 0x0419
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub20#4&30aeabc7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[188]: USB\VID_8087&PID_0024\5&8ECB065&0&1
Processing Hub 3:
  Hub[2] = '\\?\usb#root_hub20#4&35bb290d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[145]: USB\VID_8087&PID_0024\5&1DCCD487&0&1
Processing Hub 4:
  Hub[3] = '\\?\usb#vid_040e&pid_0100&tiusbd_hub#5&81b646d&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[246]: USB\VID_1058&PID_10B8\57584B3145363341504E444B
  Found ID[225]: USB\VID_04E8&PID_61C8\00000000011E0DA9
  Found ID[113]: USB\VID_05E3&PID_0612&TIUSBD_HUB\6&1F05B2FF&0&4
  Found ID[148]: USB\VID_1A40&PID_0101&TIUSBD_HUB\6&1F05B2FF&0&2
Processing Hub 5:
Processing Hub 6:
  Hub[4] = '\\?\usb#vid_1a40&pid_0101&tiusbd_hub#6&1f05b2ff&0&2#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[071]: USB\VID_058F&PID_6387\MIMWSDHB
Processing Hub 7:
  Hub[5] = '\\?\usb#vid_8087&pid_0024#5&1dccd487&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[093]: USB\VID_0951&PID_1642\001CC0EC350DBBC0A70B0030
  Found ID[200]: USB\VID_058F&PID_6362\058F312D81B
  Found ID[109]: USB\VID_0A12&PID_0001\6&944C330&0&5
  Found ID[084]: USB\VID_090C&PID_1000\AA04012700007434
Processing Hub 8:
  Hub[6] = '\\?\usb#vid_8087&pid_0024#5&8ecb065&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[003]: USB\VID_09DA&PID_054F\6&12E3C9ED&0&4
  Found ID[058]: USB\VID_0AC8&PID_3450\6&12E3C9ED&0&6
Processing IDs belonging to 'USBSTOR':
  USB\VID_1058&PID_10B8\57584B3145363341504E444B
  USB\VID_04E8&PID_61C8\00000000011E0DA9
  USB\VID_0951&PID_1642\001CC0EC350DBBC0A70B0030
  USB\VID_058F&PID_6387\MIMWSDHB
  USB\VID_058F&PID_6362\058F312D81B
  USB\VID_090C&PID_1000\AA04012700007434
Processing 'IDE' device:
  Disabled by policy
Processing 'IDE' device:
  Disabled by policy
Processing 'IDE' device:
  Disabled by policy
Processing 'IDE' device:
  Disabled by policy
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskEasy____Disk____________1100'
  Matched with ID[084]: USB\VID_090C&PID_1000\AA04012700007434
  Matched with Hub[5]: '\\?\usb#vid_8087&pid_0024#5&1dccd487&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'Easy Disk USB Device' (090C:1000)
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskGeneric_USB_CF_Reader___1.01'
Could not get children of 'USB\VID_058F&PID_6362\058F312D81B'
Found USB device 'Generic USB CF Reader USB Device' (????:????)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_cf_reader&rev_1.01#058f312d81b&1#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000002] Не удается найти указанный файл.
A device was eliminated because it didn't report itself as a disk
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskGeneric_USB_MS_Reader___1.03'
Could not get children of 'USB\VID_058F&PID_6362\058F312D81B'
Found USB device 'Generic USB MS Reader USB Device' (????:????)
Could not open '\\?\usbstor#disk&ven_generic&prod_usb_ms_reader&rev_1.03#058f312d81b&3#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000002] Не удается найти указанный файл.
A device was eliminated because it didn't report itself as a disk
Processing 'USBSTOR' device:
[... 217 lines cut ...]
  Matched with Hub[3]: '\\?\usb#vid_040e&pid_0100&tiusbd_hub#5&81b646d&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'Samsung P3 Portable USB Device' (1A40:0101)
Device eliminated because it was detected as a Hard Drive (score 18 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskWD______Elements_10B8___1007'
  Matched with ID[246]: USB\VID_1058&PID_10B8\57584B3145363341504E444B
  Matched with Hub[3]: '\\?\usb#vid_040e&pid_0100&tiusbd_hub#5&81b646d&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 3.0 device 'WD Elements 10B8 USB Device' (1058:10B8)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
3 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 493, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.8 GB (4059545600 bytes)
  Start Sector: 32, Boot: Yes, Recognized: Yes
Saved H:\Temp\4\rufus.log
`

pbatard, 2016-02-20:
>  I can not give the program a command to create it

But what I want is NOT the log from Rufus, but the log from __VISUAL STUDIO__, which you have here:
[image]

Just copy/paste the content from that window when Rufus freezes, please.

VictorVG, 2016-02-20:
Ok! Checked and external cardreader Transcend TS-RDF8K (USB 3.0) is connected to the controller on Etron ET-168A (integrated on the motherboard) - With this device, there is no problem, the problem is only with Acorp CRIP100 / CRIP200 / CRIP300 (newer I do not have ) due to their control ICs that are designed or illiterate or illiterate are included. In any case, as the hardware developer I can say for sure about the incorrectness of their circuit design and it is the fault of the developer. Here it is necessary to limit the waiting time for their response as to the instruments, I see that in the absence of CF-card, this unit is in an uncertain state on the bus.

pbatard, 2016-02-20:
OK. I'll trust that your SD card device is defective and freezes when being queried using Windows standard APIs, which it shouldn't do.

I still wouldn't mind seeing the content of the window I circled when Rufus freezes, but if you can't provide it, I'll just close this issue.

VictorVG, 2016-02-20:
Thank you! I just have one of the devices Acorp CRIP 300 is, well, how can we not see and that the chip is used there? In my devices with markings erased chip, and there I found her (apparently forgot to delete :)) - ALCOR micro AU6730, but it has long been removed from production. The peculiarity of the chip - being in an uncertain state on the bus with no card slot in my interlocutor has confirmed, well, at least what we know about him.

pbatard, 2016-02-20:
I still would like to see if there's anything I can do in Rufus (such as removing the listing for these specific card readers), which I can __ONLY__ do if you copy/paste the content from the Output window of Visual Studio from when it freezes:

[image]

Can you __PLEASE_ do that?

VictorVG, 2016-02-21:
Please, debug out, build 860:
> MS VC2010, debug out log:
> 
> out:
> 
> "rufus.exe": Загружено: "H:\Temp\4\x86_64\Release\rufus.exe", Двоичный код не был построен с отладочной информацией.
> "rufus.exe": Загружено: "C:\Windows\System32\ntdll.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\kernel32.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\KernelBase.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\setupapi.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\cfgmgr32.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\msvcrt.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\rpcrt4.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\advapi32.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\sechost.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\gdi32.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\user32.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\lpk.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\usp10.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\oleaut32.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\ole32.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\devobj.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\wininet.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\api-ms-win-downlevel-user32-l1-1-0.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\api-ms-win-downlevel-shlwapi-l1-1-0.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\shlwapi.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\api-ms-win-downlevel-version-l1-1-0.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\version.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\api-ms-win-downlevel-normaliz-l1-1-0.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\normaliz.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\iertutil.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\api-ms-win-downlevel-advapi32-l1-1-0.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\userenv.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\profapi.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\crypt32.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\msasn1.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\wintrust.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\comdlg32.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\winsxs\amd64_microsoft.windows.common-controls_6595b64144ccf1df_6.0.7601.18837_none_fa3b1e3d17594757\comctl32.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\shell32.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\imm32.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\msctf.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\api-ms-win-core-synch-l1-2-0.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\cryptbase.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\uxtheme.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Program Files (x86)\Dexpot\hooxpot64.dll", Двоичный код не был построен с отладочной информацией.
> "rufus.exe": Загружено: "C:\Windows\System32\riched20.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\clbcatq.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\dwmapi.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\ExplorerFrame.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\duser.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\dui70.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\inetcpl.cpl", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\ieframe.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\api-ms-win-downlevel-shell32-l1-1-0.dll", Невозможно найти или открыть файл PDB
> "rufus.exe": Загружено: "C:\Windows\System32\cryptui.dll", Невозможно найти или открыть файл PDB
[... cut ...]
> 	msimg32.dll	C:\Windows\System32\msimg32.dll	N/A	N/A	Невозможно найти или открыть файл PDB		56	6.1.7600.16385 (win7_rtm.090713-1255)	14.07.2009 4:31	000007FEFA270000-000007FEFA277000	[8820] rufus.exe: Машинный код
> 	slc.dll	C:\Windows\System32\slc.dll	N/A	N/A	Невозможно найти или открыть файл PDB		57	6.1.7600.16385 (win7_rtm.090713-1255)	14.07.2009 4:33	000007FEFB290000-000007FEFB29B000	[8820] rufus.exe: Машинный код
> 	mlang.dll	C:\Windows\System32\mlang.dll	N/A	N/A	Невозможно найти или открыть файл PDB		58	6.1.7600.16385 (win7_rtm.090713-1255)	14.07.2009 4:29	000007FEEFB20000-000007FEEFB5B000	[8820] rufus.exe: Машинный код

.

pbatard, 2016-02-21:
I don't understand: Why are you suddenly using VS2010 when you were using VS2015 before? Why did the location of your project change (`G:\dev\rufus\x86_64\Debug\rufus.exe` &rarr; `H:\Temp\4\x86_64\Release\rufus.exe`) and why you don't get to `*** Rufus init ***` in the output window.

According to your second entry for this issue (the `log.txt` your provided), this is what you were getting:
```
"rufus.exe" (Win32). Загружено "G:\dev\rufus\x86_64\Debug\rufus.exe". Символы загружены.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\ntdll.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\kernel32.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\KernelBase.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\setupapi.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\cfgmgr32.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\msvcrt.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\rpcrt4.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\advapi32.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\sechost.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\gdi32.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\user32.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\lpk.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\usp10.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\oleaut32.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\ole32.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\devobj.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\wininet.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\api-ms-win-downlevel-user32-l1-1-0.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\api-ms-win-downlevel-shlwapi-l1-1-0.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\shlwapi.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\api-ms-win-downlevel-version-l1-1-0.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\version.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\api-ms-win-downlevel-normaliz-l1-1-0.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\normaliz.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\iertutil.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\api-ms-win-downlevel-advapi32-l1-1-0.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\userenv.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\profapi.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\crypt32.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\msasn1.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\wintrust.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\comdlg32.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\winsxs\amd64_microsoft.windows.common-controls_6595b64144ccf1df_6.0.7601.18837_none_fa3b1e3d17594757\comctl32.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\shell32.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\imm32.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\msctf.dll". Невозможно найти или открыть PDB-файл.
"rufus.exe" (Win32). Загружено "C:\Windows\System32\api-ms-win-core-synch-l1-2-0.dll". Невозможно найти или открыть PDB-файл.
*** Rufus init ***
Will use settings from registry
found locale 'ru-RU'
loc file not found in current directory - embedded one will be used
localization: extracted data to 'C:\Users\VictorVG\AppData\Local\Temp\Ruf86FB.tmp'
localization: found locale 'en-US'
localization: found locale 'ar-SA'
(...)
```

And all I have been asking is for you to run the exact same test you ran to produce `log.txt`, on the same machine, but after running 2.7.855 with <kbd>Alt</kbd>-<kbd>.</kbd> so that the output contains more information.

Either that, or please install and run [DebugView](https://technet.microsoft.com/en-us/sysinternals/debugview.aspx) on the machine that has the card reader, and after making sure you ran 2.7.855 with <kbd>Alt</kbd>-<kbd>.</kbd> on that machine, run Rufus 2.8.858 or later and copy/paste the output from DebugView.

If you don't see `*** Rufus init ***` in the output, even if the application freezes, then you are doing something wrong...

VictorVG, 2016-02-21:
I just call it a network, and then limit the number of connections to the machine with VS2015, so we had to call it that at the moment it was free. :) With DebugView do. I then gently broken cables, chinyus and in between I look for the experiment. :)

VictorVG, 2016-02-21:
Pleas, DebugView log:

> 00000001	0.00000000	[356] *** Rufus init *** 	
> 00000002	0.00019729	[356] Will use settings from registry 	
> 00000003	0.00029729	[356] found locale 'ru-RU' 	
> 00000004	0.00061709	[356] loc file not found in current directory - embedded one will be used 	
> 00000005	0.00280829	[356] localization: extracted data to 'C:\Users\VictorVG\AppData\Local\Temp\Ruf5C4D.tmp' 	
> 00000006	0.00291009	[356] localization: found locale 'en-US' 	
> 00000007	0.00321548	[356] localization: found locale 'ar-SA' 	
> 00000008	0.00326954	[356] embedded.loc(558): the version of this translation is older than the base one and may result in some messages not being properly translated.	
> 00000009	0.00326954	[356] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
> 00000010	0.00326954	[356] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
> 00000011	0.00363018	[356] localization: found locale 'az-AZ' 	
> 00000012	0.00398362	[356] localization: found locale 'bg-BG' 	
> 00000013	0.00459561	[356] localization: found locale 'zh-CN' 	
> 00000014	0.00513493	[356] localization: found locale 'zh-TW' 	
> 00000015	0.00518988	[356] localization: found locale 'hr-HR' 	
> 00000016	0.00552290	[356] localization: found locale 'cs-CZ' 	
> 00000017	0.00585592	[356] localization: found locale 'da-DK' 	
> 00000018	0.00617302	[356] localization: found locale 'nl-NL' 	
> 00000019	0.00651055	[356] localization: found locale 'fi-FI' 	
> 00000020	0.00683756	[356] localization: found locale 'fr-FR' 	
> 00000021	0.00715376	[356] localization: found locale 'de-DE' 	
> 00000022	0.00748048	[356] localization: found locale 'el-GR' 	
> 00000023	0.00795523	[356] localization: found locale 'hu-HU' 	
> 00000024	0.00835071	[356] localization: found locale 'id-ID' 	
> 00000025	0.00870175	[356] localization: found locale 'it-IT' 	
> 00000026	0.00901585	[356] localization: found locale 'ja-JP' 	
> 00000027	0.00940742	[356] localization: found locale 'ko-KR' 	
> 00000028	0.00942364	[356] embedded.loc(8233): the version of this translation is older than the base one and may result in some messages not being properly translated.	
> 00000029	0.00942364	[356] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
> 00000030	0.00942364	[356] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
> 00000031	0.00975215	[356] localization: found locale 'lv-LV' 	
> 00000032	0.01008397	[356] localization: found locale 'lt-LT' 	
> 00000033	0.01042360	[356] localization: found locale 'ms-MY' 	
> 00000034	0.01043951	[356] embedded.loc(9594): the version of this translation is older than the base one and may result in some messages not being properly translated.	
> 00000035	0.01043951	[356] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
> 00000036	0.01043951	[356] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
> 00000037	0.01077013	[356] localization: found locale 'nb-NO' 	
> 00000038	0.01110495	[356] localization: found locale 'fa-IR' 	
> 00000039	0.01161244	[356] localization: found locale 'pl-PL' 	
> 00000040	0.01193975	[356] localization: found locale 'pt-BR' 	
> 00000041	0.01226046	[356] localization: found locale 'pt-PT' 	
> 00000042	0.01260129	[356] localization: found locale 'ro-RO' 	
> 00000043	0.01290728	[356] localization: found locale 'ru-RU' 	
> 00000044	0.01328264	[356] localization: found locale 'sr-SP' 	
> 00000045	0.01363998	[356] localization: found locale 'sk-SK' 	
> 00000046	0.01397210	[356] localization: found locale 'sl-SI' 	
> 00000047	0.01429251	[356] localization: found locale 'es-ES' 	
> 00000048	0.01463904	[356] localization: found locale 'sv-SE' 	
> 00000049	0.01497446	[356] localization: found locale 'th-TH' 	
> 00000050	0.01541078	[356] localization: found locale 'tr-TR' 	
> 00000051	0.01575821	[356] localization: found locale 'uk-UA' 	
> 00000052	0.01614409	[356] localization: found locale 'vi-VN' 	
> 00000053	0.02878561	[356] SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E 	
> 00000054	0.02965044	[356] localization: using locale 'ru-RU' 	
> 00000055	0.02966576	[356] localization: initializing default message table 	
> 00000056	0.03932512	[356] localization: using locale base 'en-US' 	
> 00000057	0.08347017	[356] Rufus version: 2.8.862 	
> 00000058	0.08422209	[356] Windows version: Windows 7 SP1 64-bit 	
> 00000059	0.08437163	[356] Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06 	
> 00000060	0.08448874	[356] Grub versions: 0.4.6a, 2.02~beta2 	
> 00000061	0.08460075	[356] System locale ID: 0x0419 	
> 00000062	0.08470254	[356] Will use default UI locale 0x0419

VictorVG, 2016-02-21:
@pbatard 

Dump memory tasks can help you? He made it with the help of Process Hacker and added to the archive and logs. [**rufus_b862_dmp_and_log.7z**](http://www49.zippyshare.com/v/SWM7MQfJ/file.html)

pbatard, 2016-02-21:
I'm pretty sure you didn't have USB Debug enabled when you ran your test. :disappointed:
 
Can you please run Rufus 2.7.855, press <kbd>Alt</kbd>-<kbd>.</kbd> until you see `USB Debug enabled` on the status bar (make sure you see `Режим USB отладки выключено` in Russian - make sure it does say `выключено` in 2.7 __before__ you try to run 2.8), and then run Rufus 2.8.858+ with DebugView.

> Dump memory tasks can help you?

No. Only a log with USB Debug enabled can.

pbatard, 2016-02-21:
Or, can you confirm that the `rufus.log` and `DEBUG.LOG` from `rufus_b862_dmp_and_log.7z` were created on the same machine, and that you didn't change the <kbd>Alt</kbd>-<kbd>.</kbd> setting?

If that is the case, that is very puzzling, because it would mean that, since there is no `Processing Hub` line, [this API call](https://github.com/pbatard/rufus/blob/master/src/usb.c#L221) fails, which I don't really see as possible, even with a bad card reader.

VictorVG, 2016-02-21:
Yes, b862 is my build error. I rebuild rufus and rewrite full .log:

```
00000001	4:29:48.641	[5556] *** Rufus init *** 	
00000002	4:29:48.641	[5556] Will use settings from registry 	
00000003	4:29:48.641	[5556] found locale 'ru-RU' 	
00000004	4:29:48.642	[5556] loc file not found in current directory - embedded one will be used 	
00000005	4:29:48.643	[5556] localization: extracted data to 'C:\Users\VictorVG\AppData\Local\Temp\Ruf14E2.tmp' 	
00000006	4:29:48.643	[5556] localization: found locale 'en-US' 	
00000007	4:29:48.644	[5556] localization: found locale 'ar-SA' 	
00000008	4:29:48.644	[5556] embedded.loc(558): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000009	4:29:48.644	[5556] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
00000010	4:29:48.644	[5556] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000011	4:29:48.645	[5556] localization: found locale 'az-AZ' 	
00000012	4:29:48.645	[5556] localization: found locale 'bg-BG' 	
00000013	4:29:48.645	[5556] localization: found locale 'zh-CN' 	
00000014	4:29:48.646	[5556] localization: found locale 'zh-TW' 	
00000015	4:29:48.646	[5556] localization: found locale 'hr-HR' 	
00000016	4:29:48.647	[5556] localization: found locale 'cs-CZ' 	
00000017	4:29:48.647	[5556] localization: found locale 'da-DK' 	
00000018	4:29:48.647	[5556] localization: found locale 'nl-NL' 	
00000019	4:29:48.648	[5556] localization: found locale 'fi-FI' 	
00000020	4:29:48.648	[5556] localization: found locale 'fr-FR' 	
00000021	4:29:48.648	[5556] localization: found locale 'de-DE' 	
00000022	4:29:48.649	[5556] localization: found locale 'el-GR' 	
00000023	4:29:48.649	[5556] localization: found locale 'hu-HU' 	
00000024	4:29:48.649	[5556] localization: found locale 'id-ID' 	
00000025	4:29:48.650	[5556] localization: found locale 'it-IT' 	
00000026	4:29:48.650	[5556] localization: found locale 'ja-JP' 	
00000027	4:29:48.651	[5556] localization: found locale 'ko-KR' 	
00000028	4:29:48.651	[5556] embedded.loc(8233): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000029	4:29:48.651	[5556] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
00000030	4:29:48.651	[5556] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000031	4:29:48.651	[5556] localization: found locale 'lv-LV' 	
00000032	4:29:48.651	[5556] localization: found locale 'lt-LT' 	
00000033	4:29:48.652	[5556] localization: found locale 'ms-MY' 	
00000034	4:29:48.652	[5556] embedded.loc(9594): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000035	4:29:48.652	[5556] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
00000036	4:29:48.652	[5556] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000037	4:29:48.652	[5556] localization: found locale 'nb-NO' 	
00000038	4:29:48.653	[5556] localization: found locale 'fa-IR' 	
00000039	4:29:48.653	[5556] localization: found locale 'pl-PL' 	
00000040	4:29:48.654	[5556] localization: found locale 'pt-BR' 	
00000041	4:29:48.654	[5556] localization: found locale 'pt-PT' 	
00000042	4:29:48.654	[5556] localization: found locale 'ro-RO' 	
00000043	4:29:48.655	[5556] localization: found locale 'ru-RU' 	
00000044	4:29:48.655	[5556] localization: found locale 'sr-SP' 	
00000045	4:29:48.655	[5556] localization: found locale 'sk-SK' 	
00000046	4:29:48.656	[5556] localization: found locale 'sl-SI' 	
00000047	4:29:48.656	[5556] localization: found locale 'es-ES' 	
00000048	4:29:48.657	[5556] localization: found locale 'sv-SE' 	
00000049	4:29:48.657	[5556] localization: found locale 'th-TH' 	
00000050	4:29:48.657	[5556] localization: found locale 'tr-TR' 	
00000051	4:29:48.658	[5556] localization: found locale 'uk-UA' 	
00000052	4:29:48.658	[5556] localization: found locale 'vi-VN' 	
00000053	4:29:48.659	[5556] localization: will use default UI language 0x0419 	
00000054	4:29:48.670	[5556] SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E 	
00000055	4:29:48.671	[5556] localization: using locale 'ru-RU' 	
00000056	4:29:48.671	[5556] localization: initializing default message table 	
00000057	4:29:48.681	[5556] localization: using locale base 'en-US' 	
00000058	4:29:48.732	[5556] Rufus version: 2.7.855 	
00000059	4:29:48.736	[5556] Windows version: Windows 7 SP1 64-bit 	
00000060	4:29:48.737	[5556] Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06 	
00000061	4:29:48.737	[5556] Grub versions: 0.4.6a, 2.02~beta2 	
00000062	4:29:48.737	[5556] Locale ID: 0x0419 	
00000063	4:29:48.764	[5556] Found USB 2.0 device 'Easy Disk USB Device' (090C:1000) 	
00000064	4:29:48.787	[5556] Found USB device 'Generic USB CF Reader USB Device' (????:????) 	
00000065	4:29:48.789	[5556] Device eliminated because it appears to contain no media 	
00000066	4:29:48.790	[5556] Found USB device 'Generic USB MS Reader USB Device' (????:????) 	
00000067	4:29:48.793	[5556] Device eliminated because it appears to contain no media 	
[... 219 lines cut ...]
00000287	4:30:33.533	[7904] Processing Hub 8: 	
[... cut ...]
00000302	4:30:33.590	[7904]   Disabled by policy 	
00000303	4:30:33.592	[7904] Processing 'IDE' device: 	
00000304	4:30:33.593	[7904]   Disabled by policy 	
00000305	4:30:33.595	[7904] Processing 'IDE' device: 	
00000306	4:30:33.596	[7904]   Disabled by policy 	
00000307	4:30:33.598	[7904] Processing 'IDE' device: 	
00000308	4:30:33.599	[7904]   Disabled by policy 	
00000309	4:30:33.601	[7904] Processing 'USBSTOR' device: 	
00000310	4:30:33.602	[7904]   Hardware ID: 'USBSTOR\DiskEasy____Disk____________1100' 	
```

VictorVG, 2016-02-21:
and with b862 - I can see his elbow accidentally touched the keyboard is soldered cliffs. He saw to remove your working copy, clone it again and rebuilt - better we do without unnecessary mistakes.

pbatard, 2016-02-21:
OK. I think I will also prefer you to run a version of Rufus I build, so that I can control debug output.

Can I please ask you to download and run [rufus-2.8_TEST1.exe](https://rufus.akeo.ie/testing/rufus-2.8_TEST1.exe) and post the DebugView output? Thanks.

VictorVG, 2016-02-21:
Please, new debug log:
```
00000001	4:53:05.583	[5500] *** Rufus init *** 	
00000002	4:53:05.583	[5500] Will use settings from registry 	
00000003	4:53:05.584	[5500] found locale 'ru-RU' 	
00000004	4:53:05.584	[5500] loc file not found in current directory - embedded one will be used 	
00000005	4:53:05.585	[5500] localization: extracted data to 'C:\Users\VictorVG\AppData\Local\Temp\Ruf65B0.tmp' 	
00000006	4:53:05.585	[5500] localization: found locale 'en-US' 	
00000007	4:53:05.585	[5500] localization: found locale 'ar-SA' 	
00000008	4:53:05.585	[5500] embedded.loc(329): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000009	4:53:05.585	[5500] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
00000010	4:53:05.585	[5500] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000011	4:53:05.586	[5500] localization: found locale 'az-AZ' 	
00000012	4:53:05.586	[5500] localization: found locale 'bg-BG' 	
00000013	4:53:05.587	[5500] localization: found locale 'zh-CN' 	
00000014	4:53:05.587	[5500] localization: found locale 'zh-TW' 	
00000015	4:53:05.587	[5500] localization: found locale 'hr-HR' 	
00000016	4:53:05.588	[5500] localization: found locale 'cs-CZ' 	
00000017	4:53:05.588	[5500] localization: found locale 'da-DK' 	
00000018	4:53:05.588	[5500] localization: found locale 'nl-NL' 	
00000019	4:53:05.589	[5500] localization: found locale 'fi-FI' 	
00000020	4:53:05.589	[5500] localization: found locale 'fr-FR' 	
00000021	4:53:05.589	[5500] localization: found locale 'de-DE' 	
00000022	4:53:05.590	[5500] localization: found locale 'el-GR' 	
00000023	4:53:05.590	[5500] localization: found locale 'hu-HU' 	
00000024	4:53:05.591	[5500] localization: found locale 'id-ID' 	
00000025	4:53:05.591	[5500] localization: found locale 'it-IT' 	
00000026	4:53:05.591	[5500] localization: found locale 'ja-JP' 	
00000027	4:53:05.592	[5500] localization: found locale 'ko-KR' 	
00000028	4:53:05.592	[5500] embedded.loc(7134): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000029	4:53:05.592	[5500] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
00000030	4:53:05.592	[5500] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000031	4:53:05.592	[5500] localization: found locale 'lv-LV' 	
00000032	4:53:05.592	[5500] localization: found locale 'lt-LT' 	
00000033	4:53:05.593	[5500] localization: found locale 'ms-MY' 	
00000034	4:53:05.593	[5500] embedded.loc(8323): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000035	4:53:05.593	[5500] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
00000036	4:53:05.593	[5500] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000037	4:53:05.593	[5500] localization: found locale 'nb-NO' 	
00000038	4:53:05.594	[5500] localization: found locale 'fa-IR' 	
00000039	4:53:05.594	[5500] localization: found locale 'pl-PL' 	
00000040	4:53:05.594	[5500] localization: found locale 'pt-BR' 	
00000041	4:53:05.595	[5500] localization: found locale 'pt-PT' 	
00000042	4:53:05.595	[5500] localization: found locale 'ro-RO' 	
00000043	4:53:05.596	[5500] localization: found locale 'ru-RU' 	
00000044	4:53:05.596	[5500] localization: found locale 'sr-SP' 	
00000045	4:53:05.596	[5500] localization: found locale 'sk-SK' 	
00000046	4:53:05.597	[5500] localization: found locale 'sl-SI' 	
00000047	4:53:05.597	[5500] localization: found locale 'es-ES' 	
00000048	4:53:05.597	[5500] localization: found locale 'sv-SE' 	
00000049	4:53:05.598	[5500] localization: found locale 'th-TH' 	
00000050	4:53:05.598	[5500] localization: found locale 'tr-TR' 	
00000051	4:53:05.599	[5500] localization: found locale 'uk-UA' 	
00000052	4:53:05.599	[5500] localization: found locale 'vi-VN' 	
00000053	4:53:05.611	[5500] SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E 	
00000054	4:53:05.612	[5500] localization: using locale 'ru-RU' 	
00000055	4:53:05.612	[5500] localization: initializing default message table 	
00000056	4:53:05.624	[5500] localization: using locale base 'en-US' 	
00000057	4:53:05.675	[5500] Rufus version: 2.8.861 (Test1) 	
00000058	4:53:05.676	[5500] Windows version: Windows 7 SP1 64-bit 	
00000059	4:53:05.677	[5500] Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06 	
00000060	4:53:05.677	[5500] Grub versions: 0.4.6a, 2.02~beta2 	
00000061	4:53:05.677	[5500] System locale ID: 0x0419 	
00000062	4:53:05.677	[5500] Will use default UI locale 0x0419 	
00000063	4:53:05.702	[5500] IN GetUSBDevices() 	
00000064	4:53:05.702	[5500] Calling SetupDiGetClassDevsA(&_GUID_DEVINTERFACE_USB_HUB) 	
00000065	4:53:05.703	[5500] dev_info = 009600F8 	
00000066	4:53:05.703	[5500] Processing Hub 1: 	
00000067	4:53:05.704	[5500] Processing Hub 2: 	
00000068	4:53:05.704	[5500]   Hub[1] = '\\?\usb#root_hub20#4&30aeabc7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}' 	
[... 24 lines cut ...]
00000093	4:53:05.709	[5500]   Found ID[003]: USB\VID_09DA&PID_054F\6&12E3C9ED&0&4 	
00000094	4:53:05.709	[5500]   Found ID[057]: USB\VID_0AC8&PID_3450\6&12E3C9ED&0&5 	
[... cut ...]
00000108	4:53:05.713	[5500]   Disabled by policy 	
00000109	4:53:05.713	[5500] Processing 'IDE' device: 	
00000110	4:53:05.713	[5500]   Disabled by policy 	
00000111	4:53:05.714	[5500] Processing 'IDE' device: 	
00000112	4:53:05.714	[5500]   Disabled by policy 	
00000113	4:53:05.714	[5500] Processing 'IDE' device: 	
00000114	4:53:05.714	[5500]   Disabled by policy 	
00000115	4:53:05.714	[5500] Processing 'USBSTOR' device: 	
00000116	4:53:05.714	[5500]   Hardware ID: 'USBSTOR\DiskEasy____Disk____________1100' 	
```

pbatard, 2016-02-21:
Ah, now I realize that your earlier log contained the output for both 2.7.855 and 2.8.861.
I have split your output to make that clearer.

Now I can get a better idea of where the enumeration freezes...

pbatard, 2016-02-21:
Okay, I have a possible idea of where the problem might be, and I added some more debug as well. Can you try [rufus-2.8_TEST2.exe](https://rufus.akeo.ie/testing/rufus-2.8_TEST2.exe) and post the output again?

Feel free to take your time. It'll probably be another 24 hours before I can look into this again...

VictorVG, 2016-02-21:
Heaven and earth! Please see the warning notice

[image]

, then the normal program starts and displays the main GUI window and can save rufus.log:

```
Rufus version: 2.8.861 (Test2)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
System locale ID: 0x0419
Will use default UI locale 0x0419
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub20#4&30aeabc7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[188]: USB\VID_8087&PID_0024\5&8ECB065&0&1
Processing Hub 3:
  Hub[2] = '\\?\usb#root_hub20#4&35bb290d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[145]: USB\VID_8087&PID_0024\5&1DCCD487&0&1
Processing Hub 4:
  Hub[3] = '\\?\usb#vid_040e&pid_0100&tiusbd_hub#5&81b646d&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[246]: USB\VID_1058&PID_10B8\57584B3145363341504E444B
  Found ID[225]: USB\VID_04E8&PID_61C8\00000000011E0DA9
  Found ID[113]: USB\VID_05E3&PID_0612&TIUSBD_HUB\6&1F05B2FF&0&4
  Found ID[148]: USB\VID_1A40&PID_0101&TIUSBD_HUB\6&1F05B2FF&0&2
Processing Hub 5:
  Hub[4] = '\\?\usb#vid_05e3&pid_0612&tiusbd_hub#6&1f05b2ff&0&4#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[001]: USB\VID_8564&PID_1000\14JY3PGT8EWQPDZE
  Found ID[192]: USB\VID_8564&PID_1000\12JVUH987CT26O4L
Processing Hub 6:
  Hub[5] = '\\?\usb#vid_1a40&pid_0101&tiusbd_hub#6&1f05b2ff&0&2#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[071]: USB\VID_058F&PID_6387\MIMWSDHB
Processing Hub 7:
  Hub[6] = '\\?\usb#vid_8087&pid_0024#5&1dccd487&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[200]: USB\VID_058F&PID_6362\058F312D81B
  Found ID[109]: USB\VID_0A12&PID_0001\6&944C330&0&5
  Found ID[093]: USB\VID_0951&PID_1642\001CC0EC350DBBC0A70B0030
Processing Hub 8:
  Hub[7] = '\\?\usb#vid_8087&pid_0024#5&8ecb065&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[003]: USB\VID_09DA&PID_054F\6&12E3C9ED&0&4
  Found ID[057]: USB\VID_0AC8&PID_3450\6&12E3C9ED&0&5
  Found ID[084]: USB\VID_090C&PID_1000\AA04012700007434
Processing IDs belonging to 'USBSTOR':
  USB\VID_0951&PID_1642\001CC0EC350DBBC0A70B0030
  USB\VID_058F&PID_6387\MIMWSDHB
  USB\VID_058F&PID_6362\058F312D81B
  USB\VID_090C&PID_1000\AA04012700007434
  USB\VID_1058&PID_10B8\57584B3145363341504E444B
  USB\VID_04E8&PID_61C8\00000000011E0DA9
  USB\VID_8564&PID_1000\14JY3PGT8EWQPDZE
  USB\VID_8564&PID_1000\12JVUH987CT26O4L
SetupDiEnumDeviceInfo(0)
AFTER SPDRP_ENUMERATOR_NAME
Processing 'IDE' device:
  Disabled by policy
SetupDiEnumDeviceInfo(1)
AFTER SPDRP_ENUMERATOR_NAME
Processing 'IDE' device:
  Disabled by policy
SetupDiEnumDeviceInfo(2)
AFTER SPDRP_ENUMERATOR_NAME
Processing 'IDE' device:
  Disabled by policy
SetupDiEnumDeviceInfo(3)
AFTER SPDRP_ENUMERATOR_NAME
Processing 'IDE' device:
  Disabled by policy
SetupDiEnumDeviceInfo(4)
AFTER SPDRP_ENUMERATOR_NAME
[... 214 lines cut ...]
TRYING 'USB\VID_058F&PID_6387\MIMWSDHB'
  NO MATCH
TRYING 'USB\VID_058F&PID_6362\058F312D81B'
  NO MATCH
TRYING 'USB\VID_090C&PID_1000\AA04012700007434'
  NO MATCH
TRYING 'USB\VID_1058&PID_10B8\57584B3145363341504E444B'
  Matched with ID[246]: USB\VID_1058&PID_10B8\57584B3145363341504E444B
  Matched with Hub[3]: '\\?\usb#vid_040e&pid_0100&tiusbd_hub#5&81b646d&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 3.0 device 'WD Elements 10B8 USB Device' (1058:10B8)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
5 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 493, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.8 GB (4059545600 bytes)
  Start Sector: 32, Boot: Yes, Recognized: Yes
```
Big thanks!

pbatard, 2016-02-21:
I'm very pleased to hear that! Thanks for the report. :+1: 

Looks like the issue is actually due to a bug with how I call [CM_Get_Sibling()](https://github.com/pbatard/rufus/blob/master/src/usb.c#L399). In TEST2 I simply commented out this part of the code.

Can I ask you to run one more test with the new [rufus-2.8_TEST3.exe](https://rufus.akeo.ie/testing/rufus-2.8_TEST3.exe)? This version contains the fix that I am planning to use, so I'd like to confirm that it works before I check it in.

VictorVG, 2016-02-21:
Yes, please my test report is here:

1. v2.7 b855 STABLE
```
Rufus version: 2.7.855
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0419
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub20#4&30aeabc7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[188]: USB\VID_8087&PID_0024\5&8ECB065&0&1
Processing Hub 3:
  Hub[2] = '\\?\usb#root_hub20#4&35bb290d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[145]: USB\VID_8087&PID_0024\5&1DCCD487&0&1
Processing Hub 4:
  Hub[3] = '\\?\usb#vid_040e&pid_0100&tiusbd_hub#5&81b646d&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[246]: USB\VID_1058&PID_10B8\57584B3145363341504E444B
  Found ID[225]: USB\VID_04E8&PID_61C8\00000000011E0DA9
  Found ID[113]: USB\VID_05E3&PID_0612&TIUSBD_HUB\6&1F05B2FF&0&4
  Found ID[148]: USB\VID_1A40&PID_0101&TIUSBD_HUB\6&1F05B2FF&0&2
Processing Hub 5:
Processing Hub 6:
  Hub[4] = '\\?\usb#vid_1a40&pid_0101&tiusbd_hub#6&1f05b2ff&0&2#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[071]: USB\VID_058F&PID_6387\MIMWSDHB
Processing Hub 7:
  Hub[5] = '\\?\usb#vid_8087&pid_0024#5&1dccd487&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[093]: USB\VID_0951&PID_1642\001CC0EC350DBBC0A70B0030
  Found ID[200]: USB\VID_058F&PID_6362\058F312D81B
  Found ID[109]: USB\VID_0A12&PID_0001\6&944C330&0&5
Processing Hub 8:
  Hub[6] = '\\?\usb#vid_8087&pid_0024#5&8ecb065&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[084]: USB\VID_090C&PID_1000\AA04012700007434
  Found ID[057]: USB\VID_0AC8&PID_3450\6&12E3C9ED&0&5
  Found ID[046]: USB\VID_03F0&PID_0605\6&12E3C9ED&0&6
  Found ID[003]: USB\VID_09DA&PID_054F\6&12E3C9ED&0&4
Processing IDs belonging to USBSTOR:
  USB\VID_1058&PID_10B8\57584B3145363341504E444B
  USB\VID_090C&PID_1000\AA04012700007434
  USB\VID_04E8&PID_61C8\00000000011E0DA9
  USB\VID_0951&PID_1642\001CC0EC350DBBC0A70B0030
  USB\VID_058F&PID_6362\058F312D81B
  USB\VID_058F&PID_6387\MIMWSDHB
Processing Device: 'USBSTOR\DiskEasy____Disk____________1100'
  Matched with ID[084]: USB\VID_090C&PID_1000\AA04012700007434
  Matched with Hub[6]: '\\?\usb#vid_8087&pid_0024#5&8ecb065&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'Easy Disk USB Device' (090C:1000)
Processing Device: 'USBSTOR\DiskGeneric_USB_CF_Reader___1.01'
Found USB device 'Generic USB CF Reader USB Device' (????:????)
Device eliminated because it appears to contain no media
Processing Device: 'USBSTOR\DiskGeneric_USB_MS_Reader___1.03'
Found USB device 'Generic USB MS Reader USB Device' (????:????)
Device eliminated because it appears to contain no media
Processing Device: 'USBSTOR\DiskGeneric_USB_SD_Reader___1.00'
  Matched with ID[200]: USB\VID_058F&PID_6362\058F312D81B
  Matched with Hub[5]: '\\?\usb#vid_8087&pid_0024#5&1dccd487&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'Generic USB SD Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Processing Device: 'USBSTOR\DiskGeneric_USB_SM_Reader___1.02'
Found USB device 'Generic USB SM Reader USB Device' (????:????)
Device eliminated because it appears to contain no media
Processing Device: 'USBSTOR\DiskJetFlashTranscend_2GB___8.07'
  Matched with ID[071]: USB\VID_058F&PID_6387\MIMWSDHB
  Matched with Hub[4]: '\\?\usb#vid_1a40&pid_0101&tiusbd_hub#6&1f05b2ff&0&2#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Could not get node connection information for 'USB\VID_058F&PID_6387\MIMWSDHB': [0x0000001F] Присоединенное к системе устройство не работает.
Found USB device 'JetFlash Transcend 2GB USB Device' (058F:6387) [ID]
Processing Device: 'USBSTOR\DiskKingstonDT_101_G2_______PMAP'
  Matched with ID[093]: USB\VID_0951&PID_1642\001CC0EC350DBBC0A70B0030
  Matched with Hub[5]: '\\?\usb#vid_8087&pid_0024#5&1dccd487&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
[... 410 lines cut ...]
Device eliminated because it was detected as a Hard Drive (score 18 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskWD______Elements_10B8___1007'
TRYING 'USB\VID_1058&PID_10B8\57584B3145363341504E444B' [Inst: 1494]
  Matched with ID[246]: USB\VID_1058&PID_10B8\57584B3145363341504E444B
  Matched with Hub[3]: '\\?\usb#vid_040e&pid_0100&tiusbd_hub#5&81b646d&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 3.0 device 'WD Elements 10B8 USB Device' (1058:10B8)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
3 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 493, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.8 GB (4059545600 bytes)
  Start Sector: 32, Boot: Yes, Recognized: Yes
```

Today, I again looked at devices such as behaving ALCOR micro AU6730 if the card reader is not CF-card, but in the slot set regular the SD - the bus with him talking trash, maybe it's the fault of a particular instance - on an oscilloscope seen small distortion of the pulse edges but not so strong that the bus conditioners is not corrected, and the main chip is clearly not responding to treatment OS. Okay, when unscrewed the case and I've got an idea -. To bring a pair of ordinary USB sockets in his place :) Just need to come up with a mounting block for RJ45 + 2xUSB sealed off from the burning of the motherboard - there it is soldered in her petals edge of the screen, and then have it some mount bracket. :)

pbatard, 2016-02-21:
Great! At this stage, I don't believe the Acorp card reader had anything to do with this issue as there was definitely a bug in Rufus that needed to be fixed. That's not to say that Acorp devices may not be problematic, but in this case, the code had an issue.

I will be closing the issue now as I push the fix. Thank you very much for running these tests and reporting - it really helped!

closed 2016-02-21 by pbatard

VictorVG, 2016-02-21:
And I personally think with this you lucky - I have for many years faced their products to work, and they told me to put it mildly not much like a lot. Developers do not just save on the little things, for cheap mass products is familiar, and sometimes thrown out of the circuit the elements that just can not be touched or it stops working. (???) It seems that the leader of the development of first-year student, and his colleagues and all students - many of their mistakes just this type of engineer who does not, as a student of the first - the second course will make sure the necessary knowledge as he has not yet acquired, and the enthusiasm he has in abundance.

commit 64828934 (2016-02-21): [usb] fix processing of sibling device instances

* Issue introduced in f88faf1a4f74dca1c88e2324acb041b094b0f999
* Closes #696