pbatard/rufus #700: Rufus cannot install ISO on more than 32GB USB

francesco1119, 2016-02-28:
Because of this [reason ](http://macrorit.com/partition-magic-manager/2tb-hard-drive-fat32-formatter.html)Rufus can't install a Windows7 ISO on a fresh new 64GB USB.

Any idea how to solve this?

Thank you

pbatard, 2016-02-28:
Have you actually tried.

Rufus uses Large FAT32 formatting to format drives that are larger than 32GB (up to 2TB), so there actually is no issue with using FAT32 for a 64GB USB drive.

Now, one issue you may have is if you have a file on your Windows 7 ISO that is larger than 4GB (because FAT32 and Large FAT32 cannot have files that are larger than this). But if you use such an ISO, then it's not an official Microsoft one, because none of the official Microsoft ISOs have a >4GB file, and I will strongly encourage you to use an official one. But then you can use NTFS instead of FAT32. And __yes__ this will __ALSO__ work for UEFI boot (provided that you use a Windows 7 64-bit ISO, as Windows 7 32-bit does not support UEFI booting at all).

In other words, all of the problems you think you are potentially facing have been solved for a long time in Rufus. So, I will ask again, have you tried? And if you have tried and are facing an issue, could you please post your __FULL__ Rufus log (just click the 'Log' button in Rufus), so that we can determine what the real issue is?

pbatard, 2016-02-28:
Addon: One thing I should mention is that, because it is an unnecessary choice, unless you select UEFI as the target system, Rufus will force the use of NTFS for BIOS boot (or UEFI-CSM which is the same thing). So you won't be able to select FAT32. But that's only because there's really no good reason to want to use FAT32 when NTFS will work just as well (if not better, since it avoids the 4GB file issue).

francesco1119, 2016-02-28:
Here the screenshot and the relatives logs, the Windows7 ISO is an original one:

[image]
Rufus version: 2.7.855
Windows version: Windows 8.1 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'Generic Flash Disk USB Device' (0000:7777)
Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4200)
2 devices found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 8158, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x3E00AA58
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 62.5 GB (67107815424 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'GRMCENVOL_EN_DVD'
  Size: 2433157120 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: No
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
  Uses Install.wim: Yes (version 0.13.1)
Using image: WIN_7_ENT_x86_SP1.ISO

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume does not have an x86 Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'Generic Flash Disk USB Device' (0000:7777)
Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4200)
2 devices found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 8158, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x3E07C0AC
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 62.5 GB (67107815424 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB 2.0 device 'Generic Flash Disk USB Device' (0000:7777)
Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4200)
2 devices found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 8158, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x3E07C0AC
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 62.5 GB (67107815424 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Closed Windows format prompt

pbatard, 2016-02-28:
Thanks. This looks similar to #632 and, as you confirm that other flash drives seem to work, seems to be related to this specific 64GB flash drive you use.

`Undetermined error while formatting.` comes from the Windows formatting API (In most circumstances, Rufus does not use its own formatting code, but simply "asks" Windows to format the drive for us), so obviously, the first thing I'd like to ask you is to try to format your drive using Windows Explorer and report what happens.

I would also strongly encourage you to run a bad blocks check (which Rufus can do - just check the relevant checkbox). You may also use <kbd>Alt</kbd>-<kbd>Z</kbd> in Rufus to zero a drive before trying to format again, but this pretty much does the same thing as the bad blocks check, except it fills the drives with zeroes.

Now, one thing that mighy be significant is that your `Generic Flash Disk USB Device` has an USB VID:PID of  `0000:7777`. `0000` as a Vendor ID (VID) is __illegal__ for consumer products (you may of course create your own device with such a VID, but if you are USB compliant and sell it, you __must__ use your own Vendor ID, and it will never be `0000`). So it doesn't look to me like your `0000:7777` device is _"street legal"_ in the first place, which of course, if this is the device you are actually encountering an issue with, make me wanna say that I'm not surprised...

On the other hand, your other device `13FE:4200` appears to be a legitimate Kingston flash drive, for which I am not aware of issues. However, even reputable hardware can fail or, worse, unscrupulous vendors may fake their cheap hardware using the IDs of well known vendors, so you never know...

All in all though, especially as a fair estimate is that probably hundred of thousands of people have been using Rufus with 64GB drives or larger, without issue, I'm afraid to say that there's probably not much I can do in Rufus if the problem is confined to your specific drive...

pbatard, 2016-02-28:
One last thing I can think of is to try to change the cluster size (increase or decrease it) and see what happens.

francesco1119, 2016-02-29:
I agree with you. 
The [64GB USB](http://www.ebay.com/itm/131644546909?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT) I bought on eBay is probably manufactured badly.
In the last 10 minutes I was able to install on it Linux Slax but not Ubuntu.
This could means that some sector could be damaged.
Thanks for letting me know that the vendor actually, is illegal. 
Is there any place I could report this? 

Thank you

pbatard, 2016-02-29:
> Is there any place I could report this? 

I guess you should contact the e-bay seller to let them know that their device is not USB compliant, due to the use of VID `0000`, so that they can report it to their supplier, and hopefully get compliant hardware in the future.

There's not much else that I think can be done for something that was purchased off e-bay...

pbatard, 2016-03-01:
Considering that there isn't anything else I can do on my side, I will close this issue.

closed 2016-03-01 by pbatard