pbatard/rufus #701: Downloading an Update

Fredderic, 2016-02-29:
Just a tiny issue...  When the portable version checks for and downloads an update, it grabs the non-portable update, not the portable one.

pbatard, 2016-02-29:
Which doesn't matter, since the non-portable version will check if there exists a `rufus.ini` present in the same directory and run in portable mode if that is the case. So, here's what will happen:

1. User had downloaded and runs `rufus-2.6p.exe`
2. Because it has a `p` in its name (that's the only difference between the portable and non-portabl;e version by the way - they are actually binary identical and only the name changes) `rufus-2.6p.exe` starts in portable mode and creates a `rufus.ini` in the same directory.
3. Rufus notifies the user that the 2.7 update is available and prompts the user to download it.
4. User downloads `rufus-2.7.exe` (which is binary identical to  `rufus-2.7p.exe` , except for the name) in the same directory.
5. When `rufus-2.7.exe` starts, it checks whether a `rufus.ini` exists in the same directory, and starts in portable mode if it does.

In other words, there are no differences between the portable and non portable version - they are exactly the same application. It's only when the application starts that it will checks for the following 2 conditions:
- Does the executable contain a `p` in its name?
- Does a `rufus.ini` exist in the same directory?
To decide whether it should start in portable mode or not.

This means that, unless someone saves the updated version in a different directory from the one they have the `.ini` file (which I consider unlikely), they will continue to run in portable mode no matter what version they download.

If that doesn't answer your question, feel free to reopen this issue.

closed 2016-02-29 by pbatard

Fredderic, 2016-03-03:
> This means that, unless someone saves the updated version in a different directory from the one they have the .ini file (which I consider unlikely), they will continue to run in portable mode no matter what version they download.

Actually, that IS the issue...  I downloaded the portable version for a reason - It'll live in my Dropbox folder along with a few other little gems like Putty, I probably won't use it for a couple years, and the next time I **do** want this program, I likely won't remember what it's called, let alone that I can change a `p` in its name to flip it between portable or not.

An appropriately pre-checked option you can change before clicking the Download button might be better, but just keeping the same portability would be nice.

In any case, perhaps mentioning this a little more prominently on the homepage, might be in order - would have saved me the bother going back to the site to download the newer portable version, if I'd known I could just rename it.

But...  Still.  Not a big issue.  Just makes it look a little, amateurish, and less trust-worthy, I guess is how I'd describe it.

pbatard, 2016-03-03:
>  let alone that I can change a p in its name to flip it between portable or not.

I don't think you quite understood what I said above. You don't need to change the name if you save it in the same folder where you have a `rufus.ini` file. If Rufus sees a `rufus.ini` in the same directory (even an empty one), it starts in portable mode, regardless of whether the application name contains a `p` or not.

> and the next time I do want this program, I likely won't remember what it's called

That doesn't make sense. The update check only runs when Rufus runs. So the __only__ way you will get an update for Rufus is if you are already running Rufus, in which case you should have no trouble figuring it's name, since it's written on the title bar...

>  Just makes it look a little, amateurish, and less trust-worthy

Nice try, but, no, I don't seem much of a reason to do anything about this. The only users it will affect is people who, despite having just launched the app from a specific folder (which they should be able to remember), decide that they want to save the new Rufus in a different folder, especially as Rufus should prompt to save in the original folder by default... So I fail to see how it will affect that many users, and it looks like you had to put forward a convoluted reason like use with dropbox, (even as the `rufus.ini` should live in the same dropbox folder, which should be the default place the update check should ask you to download the new file). Moreover, there are __very few__ settings that Rufus saves, the most important of which being the language... which, in 99% of the case, will be the same as the one you are running Windows in.

Fredderic, 2016-03-04:
Okay...  One last shot at getting my perspective across.  Feel free not to respond...

Clarification: It will be in Dropbox/public/utils, where I keep a whole bunch of portable (or at least minimally invasive) little tidbits that I've stumbled across and like.  And the most common use the collection of stuff in that folder, is to do something while on a machine other than my own - where I really would like portability, and most certainly won't have my Dropbox account actively mirrored.  And when done, I'll typically leave the folder I downloaded them into, for whomever to keep or delete as they choose.  Plus there's the reference to my public Dropbox folder in their browser history, which they can use to go remind themselves what tool it was that I'd used.  So if I tell them it's a portable program, I'd like it to stay that way.

My issue, is with programs that change the rules like this behind the scenes.  Kind of gotten used to the Browsers doing it, but a small utility like rufus, really shouldn't.  And while yes, as much as phrasing it that way does come across as a 'try', it's also not far from the truth: it really does give me a somewhat uncomfortable feeling, made significantly worse by how trivial I perceive it should be to fix and be done with -- considering the presence of that `p` is already being checked for if no `rufus.ini` file is found -- and I really don't comprehend the rationale for rejecting what I personally view as trivial and obvious expected behaviour.  I'll admit, when I first opened this topic, the response I expected was more like, "oh, yeah, not sure how we missed that one...", and what I got back sounded to me more like, "who cares, I can't be bothered", which sort of caught me off-guard.

But you're probably right, it won't affect many users at all.  Just brand new ones who see there's a portable version, and don't expect it to magically become non-portable when they do something with it that you consider to be 'unlikely' -- which somehow tends to end up being precisely how any given product WILL be used, far more often than we'd ever have imagined (fwiw, my background is in Embedded Systems firmware, where updates are problematic, bootable USB drives irrelevant, and expected and reliable behaviour pretty much mandatory).

pbatard, 2016-03-04:
> where I really would like portability, and most certainly won't have my Dropbox account actively mirrored

You __do__ realize that if you work in this fashion, even if Rufus was downloading the `p` version, you'd lose the whole point of portability (because you don't keep your settings when switching computer). Despite what many people believe, the __ONE__ goal of portability is NOT that an application should be prevented to write in the registry (which is a __ridiculous__ reason to want a portable app) but that, if you switch from one machine to another, the application settings are preserved.

Therefore, if you don't actively mirror your dropbox account then, congratulations, you have just defeated the whole point of using a portable app in the first place, and I have to ask: Why then do you want to use the portable version of Rufus?

But of course, if you do mirror your dropbox account to work in actual portable mode, then you will also mirror the `rufus.ini` (a.k.a. the place where Rufus stores its portable settings), which means that, as I tried to explain previously, __ANY__ version of Rufus that you run from that folder, including versions that don't have `p` in their name, __will still run in portable mode__. The only way you will end up with an updated version of Rufus that does not start in portable mode is if, for some weird reason, you choose not to save that update in your `Dropbox/public/utils/` directory (or `Dropbox/public/utils/rufus/` directory if you decide to have Rufus in its own directory), which I see unlikely to achieve unintentionally, as Rufus should prompt you to download an update in the same directory you launched the app from. In other words, you really need to go out of your way to end up in a situation where Rufus will not start in portable mode, and I'd really like to hear a rational scenario where you think you will actually end up in such a situation.

Oh, and you also didn't answer my question about which settings you feel are that important to preserve in Rufus...

> My issue, is with programs that change the rules like this behind the scenes.

And now, since you still haven't been able to prove your point with a relevant example of how exactly the very corner case where Rufus will not start in portable mode will affect you, you are trying to use philosophical arguments, and "feelings". You do realize that the only consideration I take into account when trying to fix an issue is not how a user "feels" about it, but a real-life, reproducible example, of how the issue is actually affecting them.

> what I got back sounded to me more like, "who cares, I can't be bothered"

That's what you got. What you got is: "I still don't see how exactly this affects you, and I surmise that this is a preemptive request (i.e. one that is not attached to a real-life scenario), because you probably erroneously thought that a version of Rufus that doesn't contain a `p` in it's name cannot run in portable mode, whereas it only takes a `rufus.ini` in the same directory for __any__ version of Rufus to start in portable mode. So I will explain this to you, and, if you are still convinced that you will end up in a situation where the updated version will not start in portable mode, wait for you to provide a real-life scenario that clarifies why you would download the updated version in a directory that doesn't contain the existing `rufus.ini` (which again is the __only__ way Rufus will not start in portable mode)".

> by how trivial I perceive it should be to fix and be done with

Ah, the good old _[Why can't you just add the feature I want? I mean, you're a developer - it shouldn't be that hard!](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want_I_mean_youre_a_developer__it_shouldnt_be_that_hard)_

I'm afraid it's not that trivial, because one thing you don't realize is that the download link is [hardcoded on the update notice on the server](http://rufus.akeo.ie/Rufus_win.ver), so now, either I have to create different update notices for the portable and non portable versions, or, more realistically, modify the download URL and insert a `p`. But then what if someone is using the non p version of Rufus, but with an `ini` file in the same directory, and relying on that feature to be able to share the regular version (because while that person knows they have a need for portable, whereas most of the users they'll be sharing Rufus with won't). For what is worth, while unlikely, this is still a more probable scenario than anything I've heard from you on why you think you MUST have a `p` version downloaded. So now, I have to do add more fine grained description of the level of portability I currently set, to figure out when I should really change the name. Still not that big a deal. However, and this is actually the important part I consider when deciding to add a feature, which is also the one most people who request such features tend to overlook, any feature needs to be regularly tested for breakage. So now that's another minor feature I need to keep track of, and, ideally, on every new version of Rufus I release, I need to test that the portable and non-portable downloads work. Well, considering that I still haven't seen any good justification as to why that feature should be added, and I'm already busy as it is on release days, I'd rather go for more stability with one less feature (especially as I already carefully considered why there wasn't much poi [... cut ...]
Please provide me a valid scenario of that then, which is what I've been asking all along (a vague _"that will happen"_ is hard to take seriously). I have no problems reconsidering the inclusion of a feature when I see hard evidence for it. But as long as you, or anyone else, is unable to provide a good reason as to why they will be likely to save the updated Rufus in a directory that is different from the one they are running the existing one from, I really don't see any reason to change the current application behaviour for update downloads.

Fredderic, 2016-03-09:
It's a nice little utility. So much so, that I don't see much in the way of settings that _need_ preserving - that's why I was talking about a scenario in which settings explicitly **wouldn't** be preserved, including, surprisingly, its portability!  Having what settings it does decide to save, saved into an `.ini` file, means delete two files and it's gone without a trace.  That's exactly what I want, and why I prefer portable apps for things like this - it really is that simple.

I started this as a first-time user of the product, with a circumstance in which it behaved surprisingly and contrary to expectation - as a first time user, not as a developer or as someone already familiar with the product.  It's that insight which is why developers get other people to test their products, and why companies have entire departments devoted to user interaction.  The perspective of a first-time user is an all too rare thing, so when I came across a situation which I felt goes aɡainst fair expected behaviour, and seemed to me like something particularly readily overlooked by established users and potentially especially the developer, I made it a point to mention it.  Again, it's that simple.

This is probably a matter of personal opinion to some degree, but while my use case which brought this to light might be odd and unusual, I can guarantee I won't be the **only** person doing this - there's always an order of magnitude more people who put up with an issue silently, than who actually say something.  And as developers, it's often easy to loose sight of the regular person using a product, who just wants it to work in a nice obvious and consistent way.  I put in the time to sort through three or four applications that seem to do the same thing, and so I don't forget about the one I ended up settling on, I'll be storing it away for future reuse.  Part of that process, was deciding whether to download the portable or non-portable version.  When I go to use it again, likely a good while from now, and decide to accept the upgrade suggestion, it'll have **_silently_ changed an aspect of it's default behaviour**, and an aspect that was specifically part of my choice to download this version (and to some degree, this application) to begin with.  It's that _silently_ part with which I have the greatest issue - if you can't preserve the default behaviour over an upgrade, then I'd rather it just send me to the change notes page with the download links so I can decide for myself.

To be clear, I'll explain my expectation of a portable application: The portable version of the application (especially being a single-file application), and without any further direction (as in from an `.ini` file the application itself generates), will run portably.

And to re-iterate my issue: The rules of portability, from the perspective of someone **not** familiar with the application, **are** changed by upgrading the application, **using the built in upgrade functionality**.  At it's most basic, if I download the portable version, I rather expect it to remain the portable version - your opinion of how or why I use it is mostly irrelevant.  I'm not trying to be rude there, it merely goes back to fair expectation of someone using your product, having chosen which version to download for reasons you're not expected (and perhaps even wanted) to know.

You are right, I wasn't aware how the update is driven, but I did have a suspicion it would be something like that.  Personally, I feel that hacking in that `p` is a little...  unclean.  It would be preferable if the name or url of the portable version could be included in the notice, or even that a second notice channel exist in parallel (either generated with the original, or from it), but if neither of those are a possibility...  But in either case, the upgrade should be consistent with the version that was running, or a means provided to change that.

And for the record, due to the infrequency with which I deal with USB drives, it took me a few moments to figure out which settings I needed to change.  For my wife, children, and a good number of friends I know, many of whom call those devices "giggi sticks", they'd be simply doing what they were told or read, with no idea what many of those options do.  But even many of them would recognise that the ini file means it saves it's setting there, and would fairly expect that copying the application to another directory (or even several other directories) would allow for different configurations.  Most such people do it regularly with games, and I've done it myself with programs in the past.  Your upgrade policy breaks that, and worse still, it'll work a few times, and then mysteriously stop working, and even I would have been hard-pressed to figure out why without at least a couple minutes of frowning and frustration.

In terms of solutions...  I would recommend, starting with most preferred:
* A small check-box next to the download button (even hidden within an 'options' drop-down), pre-loaded with the current portability of the application.
* A message pop-up warning that the version being downloaded will be the non-portable version, and offering a link to the download (or update notes) page instead.
* Or a pop up a quick dialog with radio buttons for the two versions, and the appropriate one pre-selected (though personally I'd only go that route if there were more than two options, and you have the means to provide a list of available options through the upgrade notice).

pbatard, 2016-03-10:
Again, I fix real-life issues, when someone provides me with a real-life scenario, and not philosophical aspects such as "but the app silently changed an aspect of it's default behaviour!". Despite what you'd like to believe, your perspective on that issue is not representative, and, unless there's a real-life scenario with verifiable drawback involved, software companies do not modify features according to a single person's specific likes or dislikes. Considering that about 2 million people download Rufus each month, that the update process has been here for some time, and that people who care about portability are VERY VOCAL about it (I got asked to provide a portable version of Rufus a lot before there was one), the fact that yours is the first report to complain about it, moreover, with a complaint that was based on an erroneous assertion, provides me with better information with regards to the importance of this non-issue than your "I can guarantee I won't be the only one".

Also, you confirmed my initial assertion that you erroneously thought that the update process would break portability, and went into reporting this issue with the idea that you were reporting a glaring bug, so I also have to consider that your insisting might simply be trying to save face after being proved wrong.

Finally, FYI, Rufus does have a "remove all traces" mechanism even when not running portable (<kbd>Alt</kbd>-<kbd>R</kbd>). Granted, it may not be as convenient as removing 2 files, but it is there.

Fredderic, 2016-03-10:
Oh please...  Download counts are meaningless here - unless you have the data to isolate a relevant sample for counting, and you haven't even hinted that that is the case, plus the very issue we're discussing will skew such data anyhow.  As you have confirmed, **only** people copying the executable without the ini file, **since** upgrading, will have even had a chance to notice.  Can you tell me how many of them are among that 2 million downloads?  And how about how many of them just shrugged and either decided not being portable any more didn't matter, or like me, straight away went back to the site to get the updated portable version?  That's going to skew your download count either way, particularly with relevance to this issue.

And no, this thread was always about that one thing: **transferring the post-upgrade executable alone to another location, will cause it to run non-portably!**.  And that fact **has been confirmed by you!**

I'll repeat that, because if anyone is making erroneous assertions here, and has reason to be concerned with being proven wrong, it's You.  I asserted, and you've confirmed, that the version downloaded by the upgrade feature has the default state of being non-portable, regardless of which version was originally downloaded.  And I also never said it was a "glaring bug", I said it was a small issue, and my increased insistence isn't a result of any concern about saving face, I've already pointed out I considered your response rather, undue (trying to keep things civil).

At the very least, I'm sure you can see the appropriateness of showing a quick warning dialog indicating the upgrade version will not default to being portable, and provide a clickable link to the downloads webpage.  And perhaps while you're at it, have a think if maybe there's some way you can find out for certain if, just perhaps, it IS a bigger issue than you realise - I know I would.

Now, please stop trying to distort my intentions.  I believe I made my issues very clear in the last post, and you have done nothing at all to address them, beyond once again stating - using seriously flawed statistics and an "erroneous" attack on my quite accurate assertions - that _you don't think it's worth the effort._  The application **did** "silently change an aspect of it's default behaviour", and I **do** consider that to be a significant issue for reasons I've already stated both practically and personally.

Further, since you appear to have made your mind up regardless at least one reply ago, and considering you are the all-knowing developer, I am merely a user who went to the effort of posting a bug report, yet despite that you've gone to such an awful lot to trouble to argue your point while repeatedly ignoring and attacking both me and mine...  I really do believe that in fact it is YOU who are trying to "save face" - quite honestly, I really don't have any face to save.  It's up to you in the end, as the developer, but do at least try to be honest about it - I have a minor but valid issue, and you simply don't think it's worth bothering about.

pbatard, 2016-03-10:
Sorry, but you posted a flawed bug report for an issue that didn't exist, and then you had to completely change course about what the "actual" issue was in an attempt to try to make it relevant. You tried (and failed) to bring in dropbox in the equation and then tried to bring more irrelevant elements in an attempt to prove a point that was never there.

So, please consider that 2 very specific conditions need to happen for this to qualify as an __inconvenience__ for a Rufus user ("inconvenience" because it will __NEVER__ prevent Rufus from performing its job equally well in either case):
1. Someone, who knows about portability (since they were using the portable version in the first place), has to forget to copy the obvious `.ini` file that resides in the same directory, when they copy the executable.
2. Even if the above was intentional, the receiving person (which is likely to be different from the first one, as it makes little sense to me for the originator to want to copy their portable settings if they plan to use Rufus elsewhere) has to care very strongly about portability.

Then, and only then, might the person who got the copy get slightly inconvenienced...

If you still want to try to pass a mild inconvenience which, as I stated in my second reply _"I fail to see how it will affect that many users"_ (since I have never shied away from mentioning that, in some corner cases, it is possible that it will indeed affect some users), as some kind of critical bug that needs fixing and that I have no excuse for not wanting to address, be my guest.

But don't be surprised if I no longer bother trying to explain to you why this minor __inconvenience__ doesn't actually deserve a fix, because it'll only affect users who are paranoid about portability (and those people will probably be annoyed either way anyway, when they realize that, no matter if you run it in portable mode, [Rufus still indirectly modifies registry keys](https://github.com/pbatard/rufus/wiki/FAQ#Does_Rufus_createmodify_registry_keys)).

Fredderic, 2016-03-12:
Mate, this is getting rediculous.  First up, about erroneous assertions:

Seems to me the very first mistake, was in your reading of my very first post.  Although I didn't say it in such precise terms initially, my assertion was then and is now, that "the upgrade function _silently_ changes the _default_ portability of the application".

So where exactly did I "completely change course"?  I'm STILL saying that the default portability of the application changes on upgrade, and it was what I was saying from the very first post.  The only thing that has changed, is that I'm saying it **BETTER**, now, than I was then.  And I never said it would inconvenience current users, or users taking their configuration with them (because quite frankly I didn't know, and assumed that would be a little too obvious to go unnoticed).

And what's this "some kind of critical bug" business?  My very first words were, "Just a tiny issue".  We've no doubt spend longer debating the issue, than it would have taken you to check whether the executable had a `p` in it, and download the same version to replace it, _or_ as I suggested more recently, put up a small notice dialog indicating that the version being downloaded isn't the [defaulting] portable version.

Now on with the meat of the real issue:

Again, my issue is the surprise factor: it should **never** surprise its users by silently changing an initial **choice that they made** when choosing which version to download - put really simply, if you offer a choice of version, you should respect that choice through upgrades, not decide to just change the users mind for them!  That fact right there is why I made the crack about having doubts as to the trustability of the development - I chose to download the portable version, it should be upgrading to the portable version in keeping with that choice, regardless of what you think of it, or at least _inform me when the transition happens!_  **It's the silent changing of MY choices that is so very wrong** - and yes, I get that that is a little philosophical, but it's also very much practical since you have no way (nor need to) of knowing WHY I made those choices.

And on the note of Rufus installing registry keys regardless of its portability - which I do actually consider to be "some kind of critical bug" - makes me wonder if you shouldn't just ditch the Portable version anyhow.  (You appear to have a measure of contempt for it.)  Of the two points made there, #2 is fine, though one has to wonder about #1.  These would seem to be entirely your keys, in your control, for your use - what exactly prevents you from storing those keys in the ini flle?

pbatard, 2016-03-12:
> the very first mistake, was in your reading of my very first post

You mean, the one where you stated that "it grabs the non-portable update", which is __100% wrong__ (since __any__ Rufus executable is both portable and non portable)?

> So where exactly did I "completely change course"?

Obviously, your second post, and then once more, after you indicate that you will keep Rufus in a dropbox folder, and do care about portability, only to retcon your usage scenario into one that doesn't seem to make much sense, and where you are not going to copy the ini file...

>  The only thing that has changed, is that I'm saying it BETTER, now

No, the only thing that has changed is that I helped you latch on the only scenario where someone might be slightly inconvenienced, but now you're too far in this to acknowledge that, at worst, this will be a minor inconvenience for the handful of people who may copy an updated version of Rufus with an ini, yet care that much about portability, and are still trying to make a __HUGE DEAL__ about what I have correctly anticipated (due to the zero __actual__ reports of people complaining about that corner case) is essentially a non-issue.

> And what's this "some kind of critical bug" business?

The criticality is with how you are still trying to push for this non-issue to be fixed. All your recent posts have been with trying to convince people that, unlike what you might have been saying earlier, this is a big deal. You can't really go around stating that something is "very wrong" and then pretend that it's minor, can you?

> my issue is the surprise factor

And here we go again, in trying to change the angle and morph the issue into something it's not.

>  It's the silent changing of MY choices that is so very wrong

Yes, it's very clear to me at this stage that the issue is how you believe that __YOU__ may be affected in the future, by something that, as I stated many times, is actually unlikely to affect you if you do care about portability (because then you will keep an app's ini file), and are therefore trying to make it look like it's one of the worst thing that a Rufus user could experience (again: "so very wrong"...).

> you have no way (nor need to) of knowing WHY I made those choices

Precisely why I try not to let a single user report decide whether something warrants a fix, but will instead consider the likelihood that __others__ will be affected by a similar "issue", especially when that user's report morphed into something quite different from their original entry along the way, and they tried to justify the importance of the "issue" with scenarios that either were unrealistic, made little sense, or tried to go with such arguments as "Well, I'm sure if you had user panels, they'd think just like me, because I'm representative!" or "Your 2 million downloads/month, with nobody complaining about this except me, don't mean anything".

> what exactly prevents you from storing those keys in the ini flle?

Point 1. is about what happens when using the default version (i.e. "non-portable" though this is actually a misnomer -- but one I don't have a problem using), because even people who use the non portable version might want to know whether a small application, that they probably think doesn't really need to do anything with the registry, may still end up modifying the registry. Please do realize that not everything is aimed at discussing the portable version. So, of course, the settings highlighted in 1 are written to the ini and not the registry if you use the portable version. Besides, I was hoping that the "even if you use the portable version" from point 2. would give a hint that point 1. did not apply to the portable version.

So, once again, you are __wrongly assuming__ that a problem exists where there isn't one, which, in essence, is the one major issue I've had with your entries all along, even more so as you also indicated in one of your posts above (I quote) "You are right, I wasn't aware how the update is driven" (even if, in that same sentence you try to reduce the weight of this statement with "but I did have a suspicion that...").

Maybe then you'll finally start to understand then that, because you have repeatedly provided hints that you didn't really understand how Rufus actually works, and are trying to twist a minor potential corner case inconvenience into something that it isn't, I am having trouble taking your insisting that, what is a very deliberate design decision "is so very wrong", seriously

At the very least, I expect it to be exceedingly obvious that, NO, I'm not going to do anything to address a corner case behaviour that I have been fully aware from the day I added portability, and that I consider a complete non-issue, mostly because portability in Rufus was designed in such a way that there is actually no such thing as a "portable" or "non-portable" version in the first place.

Fredderic, 2016-03-12:
> You mean, the one where you stated that "it grabs the non-portable update", which is 100% wrong (since any Rufus executable is both portable and non portable)?

It is the non-portable update in that if I run it without any configuration in the system, it will behave non-portably.  How it is written, is irrelevant - how it behaves, is everything.

> Obviously, your second post, and then once more, after you indicate that you will keep Rufus in a dropbox folder, and do care about portability, only to retcon your usage scenario into one that doesn't seem to make much sense, and where you are not going to copy the ini file...

It was in my Dropbox folder before I made my first post, and I have used that folder exactly the way I described on several occasions.  I've been doing basically what I described since 1997, with a webpage on my University account, then a USB drive for a few years, now Dropbox.  One of the best things about using Dropbox, is that it's non-writable when accessed over the webpage from a foreign machine.  My typical usage, was sitting at a friends computer fetching and running Putty to "call home" (even had PAM configured for one-time use passwords).  You can't know that, but I did point out that also in that folder is Putty, which should have indicated this is something I've been doing for a while, not just a scenario I'd dreamt up on the spot as you seem to be implying.

However, when running it on my home system, it will have its ini file, and I will have to remember to portabalize it...  not that it apparently actually matters...  Perhaps another idea, would be simply to not save ANY settings, registery or ini, unless told to do so.  I think in your shoes, with a tool like this, I'd probably go that way - choose good defaults and don't save anything unless there were settings already present at start-up.

> The criticality is with how you are still trying to push for this non-issue to be fixed. All your recent posts have been with trying to convince people that, unlike what you might have been saying [...]

The interesting thing, is you've convinced me that there's more of a big deal going on here than I'd originally thought, and most of this conversation has been trying to get you to understand my point of view - it's your product, and you will answer it as you deem appropriate, but independent of whether you agree and actually make any changes, you simply can't adequately answer an issue you don't understand.  And your accusations of erroneous assertions and assumptions, and use of dodgy statistics and other obvious tactics, tell me that you don't understand what it is I've been trying to express.

Some things you have confirmed, are also rather, troubling.  The contempt you seem to have towards the portable mode, and those of your users who wish to run it that way.  This reeks of a "I don't want to do portable, but I was badgered into it, so I'll do it, but I'm going to make it a little broken and inconvenient, fob it off as how things are in the fine print, and just wait for them to get over it and stop bothering me about portability".

> And here we go again, in trying to change the angle and morph the issue into something it's not.

Yes, there has been a shift in focus, I agree...  But the issue is still the same as specified in the original post - I expect that, if I download the portable version, for it to STAY the portable version. And yes, so long as it retains it's existing configuration, it will do so.  But the updated application will not set itself up portably like the original did, and **I was not given ANY notification of this fact.**  That IS an issue.

I don't care if you fix it to download the portable version, or just display a notice that this is the portable version, but it's about to upgrade to a non-portable one (and provide a clickable link to the download page), as long as you don't simply silently change the users choices.  Again, that IS an issue!

> Yes, it's very clear to me at this stage that the issue is how you believe that YOU may be affected in the future, by something that, as I stated many times, is actually unlikely to affect you if [...]

Well of course it's about how it effects me...  I don't know the actual number of people who do the same thing as I do, any more than you do.  It just so happens that I can guarantee there'll be at least a few of your users doing the same thing, who've probably either moved on to other products, or just shrugged and always go back to the download page to get the portable one instead of using the built-in upgrade functionality, just like I did, and would have continued to do had you not mentioned it's simply a naming convention.  THAT is why I consider it a bug that needs to be addressed.

> Precisely why I try not to let a single user report decide whether something warrants a fix, but will instead consider the likelihood that others will be affected by a similar "issue", especially [...]

I'm sorry if I somehow confused you, but the only morphing of the issue has been a direct result of you trying to corrupt and debunk my issue as frivolous and not worthy of consideration.  My scenario is two decades old, maybe closer to three if you count floppy disks before that...  I view my folder in Dropbox, no different to the multi-tool I keep in my pocket.  Here, it's for when I need to do something on someone else's computer, and wish to leave a minimal amount of residue behind when I'm done.  If I run rufus o [... cut ...] er the portable version simply because it leaves less crud behind, particularly when one of my friends asks me to help them do something or other.

And I'd like one last time to re-iterate, even if you don't wish to actually change anything, then simply present a dialog if you're upgrading a `p` version, as a reminder that the new executable won't be portable by default - it'll only come up once an update, and will be beneficial to people who DO want to keep their version defaulting to portable.

pbatard, 2016-03-13:
> How it is written, is irrelevant - how it behaves, is everything.

And how it is likely to be used is the most crucial. I can go about using superlatives too.

> And yes, so long as it retains it's existing configuration, it will do so.

Which, no matter how you want to turn it, pretty much equates to "And yes, so long as someone cares about portability it will remain portable", which has always been my point. For the inconvenience to arise, one has to stop caring about portability by choosing not to copy their portable settings.

> One of the best things about using Dropbox, is that it's non-writable when accessed over the webpage from a foreign machine.

Still don't see how that prevents you from accessing the ini. You do use dropbox in RW mode when you're not at a friend's, in which case the ini will be there.

Once again, you're trying to justify your usage scenario with another "Okay, you may have debunked the usage scenario I described earlier, but one very important thing I failed to mention then, that actually makes it relevant, is that...".

I'm sorry, I can only let that kind of reframing of a scenario slide once or twice before I call bullshit. If you use Rufus on a friend's computer and care about portability, you will of course download your ini file from dropbox. If you don't, then __clearly__ you don't care about whether the app is portable or not, in which case you're clearly trying to pull everybody's leg with regards to this pseudo issue.

Oh, and spare me another trying to refine of your scenario with "But it's still possible that I might get the non-p version from dropbox, and no ini, because it got upgraded there", since:
1. If there is a non p version in your read-only dropbox folder, then __clearly__ you had write access to that folder when the auto-update was running, which means that there __must__ also be an ini file (as the autoupdate only runs while rufus runs, which means that a ini would have been created before the update could have been written, unless, and regardless of how many updates intervened, the very first version you had was actually the non-p version... which is contradictory to what we established was the premise for this whole thing)
2. You'd have to __deliberately__ delete the ini from one of your home computer(s) even though you indicate that you care about portability (and are computer literate enough to know where that this is the place where a portable app will most likely store its settings).

If, as you state, you wish to leave the minimal residue on someone's computer, then you __will__ download the app and it's ini file, which __will__ reside in your dropbox folder, __even if__ an update was downloaded, and it won't matter whether it contains a "p" in its name or not. Or, if you don't download the ini, then I will assume that you either don't care about leaving residuals, or are explicitly trying to create a non-realistic scenario for the sole purpose of trying to prove a point that has ceased to have much to stand on a long time ago.

> tell me that you don't understand what it is I've been trying to express.

You're free to __believe__ what you want (even if it's wrong).

However, please allow me to allow state that I have __actually__ called you out, with proof, on actual elements which you did not understand, the last of which was your belief that point 1 from the description of registry changes applied to the portable version (again, if you are assuming that a __general__ FAQ is going to specialise only in dealing with a portable version, you don't understand what the purpose of a FAQ is, or failed to try to get the context of what you are reading, and you therefore failed to understand that the reason I pointed to the specific FAQ entry was only because of point 2).

>  I don't know the actual number of people who do the same thing as I do, any more than you do.

I'm afraid that, as the developer of an application used by millions of people, with plenty of means of receiving input from users of said application, I can actually get a pretty good estimate of the number of people who do the same thing as you pretend you are going to do and are inconvenienced by it. 

Regardless of what you want to __believe__, my inbox does show that there are some serious portability zealots out there (like the ones who complain that even in portable mode, Rufus may still modify the registry), so, if your best argument is "trust me on that", then I'll retort "trust me on this: if these portability zealots that I know do exist were doing what you are asserting you are planning to do, or if they were inconvenienced by it, I'd have had quite a few vocal complains by now"...

> the core issue, once again, is that I downloaded the portable version, and I want it to stay that way - but in the context within which I am most likely to use it, it will not do so without extra work and inconvenience.

Bullshit. Even in your read-only dropbox scenario, it will stay that way, unless you actually stop to care about portability by deliberately choosing not to download the ini (which __will__ definitely exist if you also have a non "p" version there). 

Please stop embarrassing yourself by trying to come up with more implausible scenarios &mdash; you're not helping your "cause"...

> The contempt you seem to have towards the portable mode

[... cut ...] ay get annoyed when a developer chooses not to address the issue you __feel__ might affect you some day, and therefore are trying to champion. But when the only elements you have to show for it are philosophical considerations, after every single scenario you put forward has been demonstrated to be either flawed from the onset, or so limited in its scope that it is unlikely to apply to anybody else but you, you might want to use your common sense and realise that it's probably time to let it go.

Fredderic, 2016-03-15:
> Which, no matter how you want to turn it, pretty much equates to "And yes, so long as someone cares about portability it will remain portable", which has always been my point. For the inconvenience to arise, one has to stop caring about portability by choosing not to copy their portable settings.

Seemingly contrary to your oppinion, minimal cruft left behind **is** also a valid use of portable applications - and the only one I'm particularly interested in where rufus is concerned.

> Still don't see how that prevents you from accessing the ini. You do use dropbox in RW mode when you're not at a friend's, in which case the ini will be there.

That is my point - I **don't want** to access the ini.  There's nothing in it that I need or want to take with me in such circumstances, I want a nice fresh instance, with all its usual defaults, and downloading both just to keep it in portable mode, means I'd actually want to keep a clean ini separate from my own, if for no reason than simply to be certain you haven't tossed some cached information into the ini that I'd rather wasn't being shared.  Unlikely in the case of rufus, but you think this situation is irrelevant - who knows what issue you'll decide is irrelevant in the future.

Is it really so hard to comprehend that respecting the choice we make when we originally chose which version to download, is important to some people?

> I'm sorry, I can only let that kind of reframing of a scenario slide once or twice before I call bullshit. If you use Rufus on a friend's computer and care about portability, you will of course [...]

You're talking about portability of configuration, I'm talking about portability of the application (with containment of configuration).  Different objectives, but the exact same mechanism - store any configuration in an ini file, if at all.  And why?  Because I have two basic uses for tools like these - one personal, one otherwise.  There's no reframing going on, other than to expand on my usage scenario as you raised specific arguments.  Simply put, I've needed to go hunt for tools to help out a friend often enough over the decades, that I've been keeping around handy little tools I've found and liked for ages now.  I usually don't know when or what form that will take in advance, and the promise that I won't leave any undue crap behind is typically even more important to them - having a portable app makes it much easier because I can point at the ini, and say delete that and the application, and it's all gone.  No `Alt-R` fuss to try and remember.  I also make a point of explaining what I'm doing as I go, so having a fresh config also means there's no "and magic happens" section in between, which means there's a better chance that next time they won't bug me to help.

So again, as for reframing...  No.  My reasons were omitted initially to keep the bug report brief - it was never meant to expand into a 100 page essay.  But the more we go through this, the more issues you drag up to explain why it's not worth bothering, and where that relates to my usage of rufus, I expand my explanation appropriately.  I'm not going to write the story of my life as seen from portable apps, and I'm pretty sure you don't want me to either.  But the base issue that prompted this whole thing, was to start with, and has continued to be, the fact that I want to be able to bring up Dropbox (currently) in a browser, download the executable (and _ONLY_ the executable), and have it run in as minimal-residual mode as possible.  The rest, has mostly been responding to your counter-arguments with discussion on how those issues would effect me and others using portability in the same way.  As I said, I consider that Dropbox folder to be no different to the multi-tool I carry in my pocket - you take out the tool you want, do what you need to do, and try not to leave a mess behind when you're done.

In hindsight, the more optimal solution would likely be a dialog up font if it's operating in a mode contrary to its default (as specified by its naming), with one of those "click this to hide this message" check boxes, backed with a command-line option to force the portability one way or another.  I've never needed to do portability, but I spend some time thinking it through earlier, and I'm pretty certain that's how I'd do it.  It's a common enough pattern, and with good reason.

> Oh, and spare me another trying to refine of your scenario with "But it's still possible that I might get the non-p version from dropbox, and no ini, because it got upgraded there", since:
> 1. If [...]

What "read-only drobbox folder"?  You do realise Dropbox lets you access your files through a standard web browser, right?  And that your folder can be accessed thus without any need of an account.  I can also (and have done) simply email someone a link to a file there.  I can't imagine using rufus that way, but it's an available option.  (I _have_ told friends to "go to my public toolbox folder, get such-and-such, and just use that" before, when I've been speaking to them on the phone while I'mout - even from the waiting room of a medical clinic, for example.  There's a good chance I won't even remember what the saved configuration is, let along want them to use it.

None of this has anything to do with how *I* use it on my own machine.  As it is, I can take advantage of the auto-update functionality because I can't imagine there being anything in the ini that I wouldn't want sha [... cut ...] as state it correctly.  Now, on the way other side of the field, if you'd instead made a closest-to-acceptable suggestion of your own, and then provided reasons not to go ahead with it, I'd have tried to reason through those issues as I have done, and/or find an alternative without those issues, but then I'd have been inclined to let it go at that - at least I'd know I've been heard, and feeling a heck of a lot more comfortable with you as a developer - with or without a resolution in my favour.

pbatard, 2016-03-15:
> That is my point - I don't want to access the ini.

I know.

And I've been trying to tell you that, even if you seem to think that it's your God given right to __DEMAND__ that, should you use Rufus in that very specific scenario, it should somehow try to twist the meaning of "portability" into something that the term was never meant to convey (I'll come to that), it doesn't make much sense for anyone who actually understands what the term "portable" means to do what you will be doing. THEREFORE, because I have to consider that the majority of Rufus users will actually be more sensible than that when it comes to the only contract of "portability" implies in an application, between the developer and its users, I do not think you have much if a case, if the only basis for your case is "I should be able to skip the ini".

Let me repeat that again, so that it's abundantly clear: Considering that, as we have previously established, there should always be an ini displayed at the same location where one user gets the updated version of Rufus, I do not see ANY reason for anyone to not want to pick that ini file unless:
- They __wrongfully__ believe that the term "portable" implies something different than _"the ability to have both the application and its settings in a location that is portable (so that things can be preserved when switching computers)"_
- They understand the semantic meaning of the term "portable" and are interested in using the application in a portable fashion (since that's the one they originally picked in this scenario), but fail to grasp that one needs to copy the readily available ini to do so.

Well, none of these reasons are anything close to something I would ever consider as warranting developer action, because, as I stated many many times before, and regardless of whether you disagree with my assessment or not (but if you do, then you'd better have something better than "I'm sure you're wrong" to prove it), they are VERY UNLIKELY to negatively affect many users, if any at all.

Now, since the first item seems to be the basis of your point for __deciding__ not to copy the ini, I'm sure you'll find reputable sources that may state that, as opposed to its actual dictionary definition, "portable" __is meant to imply__ not touching the registry ever on Windows, or leaving the computer in the exact same state as you found it once you're done (after deleting the app/config file). However, as you should realize, if that's the definition you want to go with, then even the portable version of Rufus is most certainly __NOT__ "portable", because of the whole Local Group Policy business, which by the way can have very undesirable side effects (#711 that, as opposed to this issue, I actually looked into finding a non detrimental way to fix, but decided against when it turned out the fix would likely negatively affect more people than it would help).

So I guess what's left to do then is redefine one of these many "portable" definition to mean something like _"minimal cruft left behind"_ as you stated (instead of "no cruft left behind"). But now what you're essentially saying is _"My definition of "portable", even if it contradicts others, is the one everybody should go with, and especially developers"_. To which you may understand me replying _"Or, how about we go with the definition of portable the developer settles on, especially if they tried to base it purely on semantics?"_

In other words, I don't know where you got the idea that Rufus made any promise that its portability feature meant more than being able to move settings from one computer to another, but if your assertion is that it implies not reverting to non portable mode, should you __deliberately choose__ not to duplicate the settings files, when you copy the app from a directory where the settings file most definitely exists (which I went the trouble of establishing), __then your assertion is wrong__.

This is all the more concerning if you say you are computer-savvy or concerned about app quality, as I would venture there exists a lot of apps that don't even bother checking their name for something like a 'p' but simply check the presence of a settings file in the same directory to decide whether they should run in portable or non-portable mode.

Ergo, by choosing not to the copy the ini file, you are actually taking a gamble based purely on a flawed definition of portability. Don't expect/assume others to follow you there.

> There's nothing in it that I need or want to take with me in such circumstances

Oh really? That's quite a bold statement to make.

How can you assert that you won't need something if either:
1. You don't __know__ what's in there?
2. You know what's in there, but you can't tell how it's used (because some settings, such as `Commcheck` are likely to be obtuse for anybody but the developer)?

For instance, if you're going to use Rufus on multiple computers in a row, you're probably going to start getting annoyed by being prompted for the dialog that asks you whether you want to enable the check for updates or not everytime you switch machine... which you'll only be able to avoid if you copy an ini file. And if you are maniac about the registry not being written into, I can certainly see a bunch of settings that, if you are aware of them, you might also want to be maniac about carrying over, such as display units, dual BIOS+UEFI boot, and other stuff that Rufus will store in the ini/registry.

[... cut ...]
I have also been wondering for some time how leaving a handful of registry settings behind would even remotely qualify as worthy of consideration (since you need to run Rufus with admin rights anyway, so it's not like you're going to be running in a hardened environment that may prevent registry writes... and apart from that, I don't see ANY scenario where this will have any potential negative impact on the computer, its app, or its user...).

But what do I know, me and my apparently flawed definition of portability...