pbatard/rufus #702: HDD splitted in two partition read as unique source

hhiru, 2016-03-01:
If I try to Alt + F to enable HDDs detection, Rufus doesn't recognize my partition inside the HDD as a separate drive: http://i.imgur.com/SDtxdrk.png

pbatard, 2016-03-01:
As your screenshot shows, Rufus does in fact recognize that your USB has multiple partitions: `Paritzioni multiple (E:) (F:) [512GB]` clearly indicates that Rufus is aware that your target drive is a 512 GB one, with 2 partitions.

__HOWEVER__, Rufus is __NOT__ designed to preserve partitions (and NO, I have no plans to add such a feature, for good reasons), so if you format the drive in Rufus, you will end up with a 512GB drive with a __single__ partition.

You will get an extra warning about the fact that your drive has more than one partition before formatting.

If you want to preserve one of the partitions on your drive, please use a different application.

If that doesn't answer your question, feel free to reply.

closed 2016-03-01 by pbatard

hhiru, 2016-03-02:
I was just curious. Guess I'll have to buy a dedicated usb drive to do so :) Thanks for the explanation, love your work!