pbatard/rufus #703: Information: b875 commit 00ffbae61f9e4e96f5aece3192def1c27bd77d66 have warning only on x86 VC++ 2015 build

VictorVG, 2016-03-03:
b875 b875 commit 00ffbae61f9e4e96f5aece3192def1c27bd77d66 have warning only on x86 VC++ 2015 build, but x64 no warning or errors

ms-sys warning:

original (Russian):

cl : командная строка warning D9025: переопределение "/analyze-" на "/analyze:stacksize32850"

translate (English):

cl : command line warning D9025: override "/analyze-" on "/analyze:stacksize32850"

rufus warning:

original (Russian):

..\checksum.c(496): warning C4244: -=: преобразование "uint64_t" в "size_t", возможна потеря данных
..\checksum.c(497): warning C4244: функция: преобразование "uint64_t" в "size_t", возможна потеря данных

translate (English):

..\Checksum.c(496): warning C4244: - =: transformation "uint64_t" in the "size_t", possible loss of data
..\Checksum.c(497): warning C4244: function: convert "uint64_t" in the "size_t", possible loss of data

pbatard, 2016-03-03:
Yes. That's normal.

closed 2016-03-03 by pbatard

VictorVG, 2016-03-03:
We both understand their nature and I just told you that he had seen - it is just observation and I am sure that now, when I gather Far Manager 3 SVN r13889 over there will start "fun" :) - refactoring. After he often caught amusing mistakes. :) But I'm not immune from them and today has caught a couple of "dinosaurs" who were able to! exist in the scheme watered twenty years - they are used to at this point did not come out? In theory they are testing just were bound is found's, here they are lucky for so many years nobody noticed :) But now their luck ended - caught the eye - removed, nothing :).

pbatard, 2016-03-03:
No, I meant I was aware of these warnings __before__ your reported them.

As the last series of commits should indicate, I'm still working on the checksum refactoring, so, when I noticed these warnings, I didn't want to just create a 2 line commit that fixed them, when I still have some stuff coming up the pipeline.

If you start pointing every single leftover warning that happen when I commit new code, and that usually get fixed with the next commit, we'll never see the end of it, and I don't really want to go around closing issue reports with "I know"...

So, while I am grateful that you are looking at the new code, can I ask you to do the following:
If you notice warnings, __please__ wait for the next couple of commits to have happened __before__ opening a new issue to report those.

This will avoid both of us wasting time on stuff that is going to be fixed.

VictorVG, 2016-03-03:
Ok! No question! :+1:

VictorVG, 2016-03-04:
I compile new build 877 commit 71520ba  and check warnings. My report: warning ms-sys is no more, but they fled to other places, but this is an obvious change - laughed and ignored them. It seems they like to play hide and seek with us, Well if so then let pampered child he does not cry. :)

VictorVG, 2016-03-04:
**b878** Git commit 71520baf31

There is only one - to see the most "The Steadfast Tin Soldier":

1> ------ Build started: Project: ms-sys, Configuration: Release Win32 ------
2> ------ Build started: Project: syslinux-libfat, Configuration: Release Win32 ------
2> cache.c
1> cl: Command line warning D9025: override "/analyze-" on "/analyze:stacksize32850"

Looks like what they are interested because they frightened that they are wild panic in all directions screaming like cockroaches on the slots are blocked. :)

pbatard, 2016-03-04:
As I said 2 posts above:

> Oh and the stacksize warning is normal - I manually increased the stack size with the option reported as warning.

I will __NOT__ do anything about that warning, because it's not really a warning, it just says "Hey, I see that you __deliberately__ increased the maximum stacksize for analysis", which I did.

VictorVG, 2016-03-04:
I understand it perfectly, and just showed that the compiler (the robot) as a good fool raised panic where for it and no reason. I thought it was no less ridiculous than the "manifestation of" machine "intelligence" in the chip AU6730 about which we recently spoke - one leg is not connected to ground, and the chip does not behave predictably though like against that all the measures in the scheme adopted - but a piece of silicon I decided that he was more intelligent than man. :)