pbatard/rufus #704: usb stick not formatting to ntfs

vzr1800, 2016-03-03:
I have a USB stick that will not be formatted for NTFS .. and when this happens, I can not make it to an boot stick with windows 10th
I know that there are other programs to remedy this and make it to ntfs ..
but the programs have the possibility to make the stick bootable and download an iso file to it.

pbatard, 2016-03-03:
Please provide a __FULL__ log. I'm afraid _"will not be formatted"_ does not tell me anything useful to help (what happens? what is the actual error you see?). Only a FULL log will provide me with enough details to try to help your out.

vzr1800, 2016-03-03:
if i format the usb stick with: MiniTool Partition Wizard Free 9.1 THERE IS NO PROBLEM FORMATTING TO NTFS.

but in rufus 2,7 it wont..

here is the LOG:

-------------------------------------------------------------------------------------------------------
Rufus version: 2.7.855
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0406
Found USB device 'Generic Flash Udisk USB Device' (1000:1320) [GP]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3798, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC89EECBE
Drive has a DOS/NT/95A (F2) Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Size: 29.1 GB (31240681472 bytes)
  Start Sector: 6722, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'J_CCSA_X86FRE_DA-DK_DV5'
  Size: 2890647552 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
  Uses Install.wim: Yes (version 0.13.1)
Using image: Win10_1511_Danish_x32.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE5 for write access
Will use 'J:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a DOS/NT/95A (F2) Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'J:' after error

Found USB device 'Generic Flash Udisk USB Device' (1000:1320) [GP]
1 device found
No volume information for drive 0x85
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3798, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x008382B8
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 29.1 GB (31243074560 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Closed Windows format prompt
Found USB device 'Generic Flash Udisk USB Device' (1000:1320) [GP]
1 device found
No volume information for drive 0x85
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3798, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x008382B8
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 29.1 GB (31243074560 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

________________________ 

From: Pete Batard 

Sent: Thursday, March 03, 2016 1:48 PM
To: pbatard/rufus 
Cc: vzr1800 
Subject: Re: [rufus] usb stick not formatting to ntfs (#704)

Please provide a FULL log. I'm afraid "will not be formatted" does not tell me anything useful to help (what happens? what is the actual error you see?). Only a FULL log will provide me with enough details to try to help your out.

pbatard, 2016-03-03:
Can you try zeroing the drive for a while (press <kbd>Alt</kbd>-<kbd>Z</kbd> in Rufus, then press cancel after a while), and try formatting again?

Also, what happens if you try to format your drive using Windows' file explorer (just right click on your drive and select.). Can you also try changing the cluster size?

vzr1800, 2016-03-03:
i can’t do it in windows file explrer. (error) 

and i rufus.. it did not help changing the cluster size

vh. Jan

________________________ 
Med venlig hilsen: 
Jan A Jensen.
Hjemmeside: www.4xjj.dk 
E-mail: jan@4xjj.dk Online på Skype Call Jan A Jensen 
Tlf: 2164 3501

From: Pete Batard 
Sent: Thursday, March 03, 2016 2:07 PM
To: pbatard/rufus 
Cc: vzr1800 
Subject: Re: [rufus] usb stick not formatting to ntfs (#704)

Can you try zeroing the drive for a while (press Alt-Z in Rufus, then press cancel after a while), and try formatting again?

Also, what happens if you try to format your drive using Windows' file explorer (just right click on your drive and select.). Can you also try changing the cluster size?

pbatard, 2016-03-03:
> i can’t do it in windows file explrer. (error)

Aha!

Then I'm afraid this is not a Rufus issue, as all Rufus does is ask the Windows API to format the drive, so of course, if Explorer has an issue, then Rufus also has an issue. I wouldn't mind finding out why you can't seem to format your drive, but you might have to take this issue to Microsoft.

One thing that seems strange from your log is that there appears to be an extra 'layer' between your USB Flash drive disk and the flash drive itself:
```Found USB device 'Generic Flash Udisk USB Device' (1000:1320) [GP]```

`[GP]` means that we had to go up to the grandparent to locate the USB device that contains the disk, which means that, as far as Windows is concerned, there is an extra device between the USB disk and the USB flash drive. Does your drive use encryption or has added "security" features by any chance? Some of these may make it incompatible with Rufus, as highlighted at the [end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with).

But really, if Windows Explorer fails to format the drive, there isn't much I can do in Rufus (regardless of whether other applications are successful - Minitool Partition Wizard is __closed source__, so I can not get any information on what they do to work around the Microsoft issue), and you will have to take that issue to either Microsoft or your USB Flash Drive manufacturer...

vzr1800, 2016-03-03:
Okay thank anyway

Tmp341, 2016-03-03:
Have you tried HP USB Storage Format Tool? Try formatting with it. If it doesn't show ntfs, than that means your usb stick doesn't support it.

vzr1800, 2016-03-03:
I have trind hps tool.. but I makes  same Error.  Only mini partition tool
works.

Tmp341, 2016-03-04:
That tool formatting? Then it isn't writing error.

pbatard, 2016-03-06:
I don't think there's anything more I can do, since this is a problem between the Microsoft formatting API (which Rufus simply reuses) and this specific USB flash drive. Therefore I will close this issue.

closed 2016-03-06 by pbatard