pbatard/rufus #708: Enabling "dual UEFI/BIOS mode" will make "MBR partition scheme for UEFI" bootable on BIOS systems

ekkolord, 2016-03-07:
I think "dual UEFI/BIOS mode" should only affect "MBR partition scheme for BIOS or UEFI" , right?

BTW UEFI:NTFS is really awesome!

pbatard, 2016-03-07:
> I think "Dual UEFI/BIOS mode" should only affect "MBR partition scheme for BIOS or UEFI" , right?

Of course, since BIOS is unable to boot from GPT, so no BIOS bootable content is installed, and "MBR for UEFI" and is exactly the same as "GPT for UEFI", but for people who may need to access the USB content from Windows XP (since Windows XP can't access anything GPT either).

Therefore the only way you will get dual UEFI/BIOS is if you select "MBR partition scheme for BIOS or UEFI". By the way, when dual UEFI/BIOS is not possible or disabled (i.e. only BIOS), you should see that this first option displays "MBR partition scheme for BIOS or ___UEFI-CSM___", which means that the USB will __NOT__ boot in pure UEFI mode, but it BIOS mode on UEFI systems. On the other hand, when dual UEFI/BIOS is enabled, you will see "MBR partition scheme for BIOS or UEFI", which means that it can boot both in BIOS mode and pure UEFI mode.

By the way, if a Linux ISO is dual UEFI/BIOS, it will usually remain that way after creating a bootable USB in Rufus, and you don't need to enable dual UEFI/BIOS mode using <kbd>Alt</kbd>-<kbd>E</kbd>. It's only Windows ISO that are actively prevented from booting in dual UEFI/BIOS mode by default, because my expectation is that most of the people installing Windows might need some help in making sure that they only install Windows in the mode they want.

I'll close this issue now, as I believe I answered your question.

ekkolord, 2016-03-07:
Thank you.

[image]

When I enabled dual UEFI/BIOS mode, and creating a Windows bootable USB drive with that 2nd partition scheme and target system type shown in the picture, the bootable USB drive is also bootable on BIOS systems, I think this is a bug.

pbatard, 2016-03-07:
Can you provide a __full__ log? I'd like to see what ISO you used, as well as the other details from your USB creation. Thanks.

pbatard, 2016-03-07:
Also, you did activate the Dual UEFI/BIOS mode then? Coz if that only happens with the cheat mode on, I'm tempted not to do anything about it (since cheat-modes are provided with no warranty of meeting any user expectations &mdash; That's why they are cheat modes).

ekkolord, 2016-03-07:
Yes, only happens with the dual UEFI/BIOS mode on.

```
Rufus version: 2.8.879
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found non-USB removable device 'M4-CT064M4SSD2'
Device eliminated because it contains a mounted partition that is set as non-removable
Found non-USB removable device 'Crucial_CT120M500SSD1'
Device eliminated because it contains a mounted partition that is set as non-removable
Found USB 2.0 device 'SanDisk Extreme USB Device' (0781:5580)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x49464555 (UEFI target)
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.8 GB (32014918656 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 256 KB (262144 bytes)
  Start Sector: 62531186, Boot: No, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'J_CCSA_X64FRE_EN-US_DV5'
  Size: 4414115840 bytes
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: en_windows_10_multiple_editions_version_1511_updated_feb_2016_x64_dvd_8379634.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Using Windows 7 MBR
Found volume GUID \\?\Volume{8035cc9e-e39b-11e5-bdc5-806e6f6e6963}\
Successfully remounted Volume{8035cc9e-e39b-11e5-bdc5-806e6f6e6963}\ on F:\
Copying ISO files...
Extracting files...
Image is an UDF image
Extracting: F:\autorun.inf (128 bytes)

......

Extracting: F:\support\logging\winsetupetw.dll (22.7 KB)
Finalizing, please wait...
F:autorun.inf already exists - keeping it
NTFS Fixup (Checkdisk)...
Volume label is J_CCSA_X64FRE_EN-US_DV5.

Stage 1: Examining basic file system structure ...
  1280 file records processed.
File verification completed.
  0 large file records processed.
  0 bad file records processed.

Stage 2: Examining file name linkage ...
  1496 index entries processed.
Index verification completed.

Stage 3: Examining security descriptors ...
Security descriptor verification completed.
  108 data files processed.

Windows has scanned the file system and found no problems.
No further action is required.

  31264568 KB total disk space.
   4342188 KB in 971 files.
       304 KB in 110 indexes.
     45320 KB in use by the system.
     42672 KB occupied by the log file.
  26876756 KB available on disk.
      4096 bytes in each allocation unit.
   7816142 total allocation units on disk.
   6719189 allocation units available on disk.
NTFS Fixup completed.

Found non-USB removable device 'M4-CT064M4SSD2'
Device eliminated because it contains a mounted partition that is set as non-removable
Found non-USB removable device 'Crucial_CT120M500SSD1'
Device eliminated because it contains a mounted partition that is set as non-removable
Found USB 2.0 device 'SanDisk Extreme USB Device' (0781:5580)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x49464555 (UEFI target)
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.8 GB (32014918656 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 256 KB (262144 bytes)
  Start Sector: 62531186, Boot: No, Recognized: Yes

```

pbatard, 2016-03-07:
Aha, now I remember that I thought about disabling dual UEFI/BIOS mode when "MBR for UEFI" was selected, but decided against introducing a modification that could affect drive creation when dual UEFI/BIOS is disabled.

But now that I've looked into it some more, it should be a relatively easy fix, with a low risk of adverse effect (famous last words...), so let me add that.

closed 2016-03-07 by pbatard

VictorVG, 2016-03-07:
Big thanks! I met with this phenomenon in the PassMark MemTest86 + 6.0 - authors have a dual-boot test there - if the package is launched in the UEFI BIOS is available all (, including the full width of the memory bus, and RAM is checked in SMP mode, as if through the old MBR BIOS - you will get a 32-bit test version 4.20 with partial checking RAM subsystem.

And go catch the correct startup mode - usually version 4.20 starts, although in my UEFI BIOS motherboards stitched (normal version GigaByte Touch BIOS runs too long - consistently for 40 seconds waiting for a response from each SATA / SAS / device SCSI and just not getting it goes . :) that's because more and replaced this option based on the code AWARD Software firmware based on the code AMI UEFI BIOS and in addition had the opportunity to switch when you start the computer NumLock :: Off - to Award Touch BIOS it always NumLok :: on, and to switch it it is necessary to patch the firmware code, and AMI UEFI is an option in the settings, and themselves setting modes more equipment and more precise voltage regulation - firmware greatly overestimates their relatively exposed values, even sometimes reduced the place (for example, RAM Kingston HyperX Genesic DDR3-1600 9,0-9-9-27@1600 MHz is have default voltage 1,65V standing there since SK Hynix chips designed for 1333MHz/1,5V 9,0-9-9-24 - factory overclocking, and UEFI BIOS puts tension 1,644V - enough for them, because AWARD put 1,985V - too many chips on the failure limit).

commit 1cc7fca4 (2016-03-07): [ui] disable BIOS boot when MBR for UEFI is selected in dual UEFI/BIOS mode

* Closes #708